Proxmark/proxmark3 #255: Install issue on windows

lewis-elliott, 2017-03-27:
Trying to compile on windows:

`make clean && make all`

output:

`make[1]: *** [obj/util.o] Error 1
make[1]: Leaving directory `/pm3/client'
make: *** [client/all] Error 2`

pwpiwi, 2017-03-27:
Please post the complete output. This excerpt doesn't help too much.

lewis-elliott, 2017-03-27:
```
make[2]: Entering directory `/pm3/liblua'
make "LUA_A=lua52.dll" "LUA_T=lua.exe" \
        "AR=gcc -shared -o" "RANLIB=strip --strip-unneeded" \
        "SYSCFLAGS=-DLUA_BUILD_AS_DLL" "SYSLIBS=" "SYSLDFLAGS=-s" lua.exe
make[3]: Entering directory `/pm3/liblua'
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lua.o lua.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lapi.o lapi.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lcode.o lcode.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lctype.o lctype.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o ldebug.o ldebug.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o ldo.o ldo.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o ldump.o ldump.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lfunc.o lfunc.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lgc.o lgc.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o llex.o llex.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lmem.o lmem.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lobject.o lobject.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lopcodes.o lopcodes.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lparser.o lparser.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lstate.o lstate.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lstring.o lstring.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o ltable.o ltable.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o ltm.o ltm.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lundump.o lundump.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lvm.o lvm.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lzio.o lzio.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lauxlib.o lauxlib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lbaselib.o lbaselib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lbitlib.o lbitlib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lcorolib.o lcorolib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o ldblib.o ldblib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o liolib.o liolib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lmathlib.o lmathlib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o loslib.o loslib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o lstrlib.o lstrlib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o ltablib.o ltablib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o loadlib.o loadlib.c
gcc -O2 -Wall -DLUA_COMPAT_ALL -DLUA_BUILD_AS_DLL    -c -o linit.o linit.c
gcc -shared -o lua52.dll lapi.o lcode.o lctype.o ldebug.o ldo.o ldump.o lfunc.o lgc.o llex.o lmem.o lobject.o lopcodes.o lparser.o lstate.o lstring.o ltable.o ltm.o lundump.o lvm.o lzio.o lauxlib.o lbaselib.o lbitlib.o lcorolib.o ldblib.o liolib.o lmathlib.o loslib.o lstrlib.o ltablib.o loadlib.o linit.o
strip --strip-unneeded lua52.dll
gcc -o lua.exe -s  lua.o lua52.dll -lm
make[3]: Leaving directory `/pm3/liblua'
make "LUAC_T=luac.exe" luac.exe
make[3]: Entering directory `/pm3/liblua'
gcc -O2 -Wall -DLUA_COMPAT_ALL     -c -o luac.o luac.c
ar rcu liblua.a lapi.o lcode.o lctype.o ldebug.o ldo.o ldump.o lfunc.o lgc.o llex.o lmem.o lobject.o lopcodes.o lparser.o lstate.o lstring.o ltable.o ltm.o lundump.o lvm.o lzio.o lauxlib.o lbaselib.o lbitlib.o lcorolib.o ldblib.o liolib.o lmathlib.o loslib.o lstrlib.o ltablib.o loadlib.o linit.o
ranlib liblua.a
gcc -o luac.exe   luac.o liblua.a -lm
make[3]: Leaving directory `/pm3/liblua'
make[2]: Leaving directory `/pm3/liblua'
gcc -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../tools -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O4 -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/proxmark3.o proxmark3.c
gcc -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../tools -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O4 -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/uart.o uart.c
gcc -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../tools -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O4 -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/util.o util.c
util.c: In function 'msclock':
util.c:638: error: storage size of 't' isn't known
util.c:639: warning: implicit declaration of function '_ftime_s'
util.c:638: warning: unused variable 't'
make[1]: *** [obj/util.o] Error 1
make[1]: Leaving directory `/pm3/client'
make: *** [client/all] Error 2
```

marshmellow42, 2017-03-27:
interesting.  it looks like your environment is missing _ftime_s

it appears there has been a bug in some versions of mingw that is missing that definition: https://github.com/tinycthread/tinycthread/pull/30

pwpiwi, 2017-03-27:
Would you please try and add `#include sys/timeb.h` close to line 637 in util.h and try again.

m2otech, 2017-03-28:
Hi @pwpiwi I had the same problem. I kind of fixed it by:
1. `#define _ftime_s _ftime` as suggested in the linked issue
2. _adding_ `#include <sys/timeb.h>` to provide `_timeb`
3. removing the `if (_ftime_s(&t))` check, as the redefined function is void

This seems to compile, but it's not really a clean solution. Do you have anything better?

Note: I also installed the latest version of mingw (before I was using the one provided by ProxSpace). Which version do you use?

marshmellow42, 2017-03-28:
I am able to reproduce this issue in the old proxspace environment.

marshmellow42, 2017-03-28:
neither <sys/types.h> nor <sys/timeb.h> help as they do not contain _ftime_s in them in that old version.  it  appears it was added in a later version.

m2otech, 2017-03-28:
I've just installed the current version of mingw from here https://sourceforge.net/projects/mingw/?source=typ_redirect

Symbols don't appear to be included there either...

marshmellow42, 2017-03-28:
i think it may have something to do with 64 bit vs 32 bit whether it is included or not?

m2otech, 2017-03-28:
There is `_ftime64` and `_ftime32` beside `_ftime` but no `_ftime_s`

pwpiwi, 2017-03-29:
The "secure" variants of many functions (with the trailing `_s`) had been introduced by Microsoft at least 10 years back. Hard to believe that they didn't make it into some mingw versions yet.  For the time being I would vote for @m2otech's solution. In the end we don' t use other secure functions either (like e.g. strcpy_s).

@m2otech: would you raise a Pull Request, please?

marshmellow42, 2017-03-29:
just changing _ftime_s to _ftime yields the error
util.c: In function 'msclock':
util.c:640: error: void value not ignored as it ought to be
as @m2otech indicated.  i'm not 100% on how to fix that and get the desired result of that if else statement.  thoughts @pwpiwi?

marshmellow42, 2017-03-29:
i think this is what @m2otech was suggesting and it does compile.
```// a milliseconds timer for performance measurement
uint64_t msclock() {
#if defined(_WIN32)
#include <sys/types.h>
#include <sys/timeb.h>
	struct _timeb t;
	return 1000 * t.time + t.millitm;
#else
	struct timespec t;
	clock_gettime(CLOCK_MONOTONIC, &t);
	return (t.tv_sec * 1000 + t.tv_nsec / 1000000);
#endif
}

m2otech, 2017-03-29:
Yes like that but you still need to call the function, just without using
its return value (without the if - so I guess we don't know if the call was
successful). I will upload it later.

pwpiwi, 2017-03-29:
Just add `_ftime(&t);` before the return statement. Otherwise it returns 
uninitialized dars.

Mit AquaMail Android
http://www.aqua-mail.com gesendet

m2otech, 2017-03-29:
Exactly. Actually if we use _ftime() directly (and not _ftime_s()) we don't
need the #define

marshmellow42, 2017-03-29:
Yes,. I caught my cut and paste error that missed that line.  I edited my post on github but that doesn't update the emails...

closed 2017-03-30 by pwpiwi