Proxmark/proxmark3 #261: msclock - mingw old env..

iceman1001, 2017-03-30:
Sorry, the old mingw env still has a problem with the _ftime32...

> In file included from util.c:641:0:
> c:\mingw\include\sys\timeb.h: In function 'msclock':
> c:\mingw\include\sys\timeb.h:73:41: error: nested function '_ftime32' declared 'exter
>  _CRTALIAS void __cdecl __MINGW_NOTHROW  _ftime32 (struct __timeb32* _v1) {
>                                          ^
> make: *** [obj/util.o] Error 1
> pm3 ~/client$

pwpiwi, 2017-03-30:
@marshmellow confirmed that it works on old proxspace. (?) Is this with master or with your fork?

iceman1001, 2017-03-30:
its pm3 master with last pull..

marshmellow42, 2017-03-30:
It works fine in the proxspace environment on my PC.  But it didn't before the fix...

marshmellow42, 2017-03-30:
I will test another PC in a bit

pwpiwi, 2017-03-30:
Hmm. In order to replicate your issue I (first time) followed the instructions on https://github.com/Proxmark/proxmark3/wiki/Windows and installed the old Proxspace environment https://storage.googleapis.com/google-code-archive-downloads/v2/code.google.com/proxmark3/ProxSpace-130613.7z. This is based on gcc 4.4.0 (client) and gcc 4.7.1 (devkitARM). I can compile with only a few warnings
```
gcc -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../tools -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O4 -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/cmddata.o cmddata.c
cmddata.c: In function 'getSamples':
cmddata.c:1143: warning: dereferencing pointer 'sc' does break strict-aliasing rules
cmddata.c:1141: warning: dereferencing pointer 'sc' does break strict-aliasing rules
cmddata.c:1142: warning: dereferencing pointer 'sc' does break strict-aliasing rules
cmddata.c:1140: note: initialized from here
```
and
```
gcc -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../tools -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O4 -DHAVE_GUI -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED  -c -o obj/cmdhf14a.o cmdhf14a.c
cmdhf14a.c: In function 'CmdHF14ACUIDs':
cmdhf14a.c:451: warning: dereferencing pointer 'card' does break strict-aliasing rules
cmdhf14a.c:452: warning: dereferencing pointer 'card' does break strict-aliasing rules
cmdhf14a.c:444: note: initialized from here
```
But no issue at all with _ftime. Which mingw environment are you using?

iceman1001, 2017-03-30:
Its on repeat this Q/A.  I still have the old env, but with a QT5 and GCC 4.9.3 update.

marshmellow42, 2017-03-30:
The warnings have been there for years, I made an issue once for them long ago but was told they are fine.

iceman1001, 2017-03-30:
no warnings in gcc4.9.3..
just complains in timeb.h that _ftime32 is declared extern...

iceman1001, 2017-03-30:
The part in timeb.h thats causing it.  

> #if MSVCRT_VERSION >= 800
> _CRTIMP void __cdecl __MINGW_NOTHROW	_ftime32 (struct __timeb32*);
> #else
> _CRTALIAS void __cdecl __MINGW_NOTHROW  _ftime32 (struct __timeb32* _v1) {
>     return _ftime((struct _timeb *)_v1);
> }
> #endif

iceman1001, 2017-03-30:
I'm starting to dislike mingw more and more.

marshmellow42, 2017-03-30:
I'm no expert, but it sounds like environment corruption or mismatched versions if a system header is erroring like it is.

iceman1001, 2017-03-30:
I'm not counting that possiblity out.

pwpiwi, 2017-03-30:
If the macros expand to a nested function declared extern (this is what the error message says) then this is a no go. Either @marshmellow42 is right or you just discovered a bug in your mingw version.

iceman1001, 2017-03-30:
One thing that states against the system error idea, is that I have @pwpiwi  's msclock in my fork and it compiles.

pwpiwi, 2017-03-30:
Where do you have it? Can't find.

iceman1001, 2017-03-30:
Not pushed,  too much changes going on.

iceman1001, 2017-06-08:
If I understand this one correctly,   according to @marshmellow42  this compiles on some older systems.  
But we strongly suggest to run @Gator96100  's new ming env for windows users.

And that old links, text should have been, or have been edited. 

time to close?

pwpiwi, 2017-06-08:
The new LF graphing overlays require uic.exe in the development environment. It is therefore time to bury the old ProxSpace. @Gator96100's new one is a good alternative for beginners who don't want to install the components separately.

And yes, this can be closed.

closed 2017-06-08 by pwpiwi

iceman1001, 2017-06-08:
I will remove the old env /  add a @Gator96100  rarfile on proxmark.org/files section aswell.