Proxmark/proxmark3 #267: Error loop using "lf t55 wakeup" without "p" param

masterix21, 2017-04-03:
Hi dears, 

if you use "lf t55 wakeup 11223344" without p param, software stop working inside a infine loop.

Example, that works:
`proxmark3> lf t55 wakeup p 11223344
Wake up command sent. Try read now
proxmark3>`

Example with infine loop:
`proxmark3> lf t55 wakeup 11223344
Unknown parameter '1'
Unknown parameter '1'
Unknown parameter '1'
Unknown parameter '1'
Unknown parameter '1'
Unknown parameter '1'
Unknown parameter '1'
...
...
LOOP
...
...
Unknown parameter '1'
Unknown parameter '1'
Unknown parameter '1'
Unknown parameter '1'
Unknown parameter '1'
Unknown parameter '1'
Unknown parameter '1'`

marshmellow42, 2017-04-03:
the while loop is missing "&& !errors"

I'll fix it in my next pull.  i'm working on the t55xx commands anyway.

masterix21, 2017-04-03:
Great :) 

i think could be simplified the usage with:
`lf t55xx wakeup [h] <password>`
exactly like dump command, because a wakeup is possible only with password.

Anyway, thanks.

PR #268 (2017-04-05): various lf bug fixes and improvements
fixes #267 
also fixes a bug in noralsy demod.
improves lf sequence terminator(ST) detection
improves lf t55xx detection and reading.
fixes a bug in sprint_bin_break (didn't print last bit)
fixes a bug in lf indala demod corrupting the end of the ID sometimes
improves psk and nrz detection / demodulation
added t55xx page 1 detection and added it to the lf search chip detection. (based on mfg code E015)
added experimental t55xx testmode write command. (careful.  can and will overwrite entire tag no matter the selected block)
closed 2017-04-05 by iceman1001