Proxmark/proxmark3 #269: data rawdemod am = Segmentation fault: 11

masterix21, 2017-04-06:
Hi dears, 

i think that i found a bug (only tested with T55x7): if you use the command "data rawdemod am" proxmark will return a segmentation fault.

**Step 1:**
lf search u

**Step 2:**
data rawdemod am

Segmentation fault: 11

marshmellow42, 2017-04-06:
I'll check it out.  Can you post some system information including ram?

masterix21, 2017-04-06:
iMac (Retina 5K, 27-inch, Late 2015)
CPU: 3,2 GHz Intel Core i5
RAM: 32 GB 1867 MHz DDR3
MacOS Sierra 10.12.4 (16E195)

marshmellow42, 2017-04-06:
i'm not able to reproduce, can you post a trace after the lf search u.

i have a feeling what it is but i'd like to be sure.

pwpiwi, 2017-04-06:
Ah, great, another Mac user. So you can confirm that building with brew finally works?

masterix21, 2017-04-06:
i have installed QT4 from brew than compiled and it works, but sometimes proxmark3 is not responding and you have to click on graph window to unlock.

You can follow it:
brew install cartr/qt4/qt
git clone https://github.com/Proxmark/proxmark3.git
cd proxmark3 && make clean && make

and it works but with the problems that i mentioned above.

pwpiwi, 2017-04-06:
Did you try the procedure described in https://github.com/Proxmark/homebrew-proxmark3/blob/master/README.md (using the second option with HEAD)

masterix21, 2017-04-06:
@marshmellow42 
> i'm not able to reproduce, can you post a trace after the lf search u.

Here what you need :)

[proxmark3.txt](https://github.com/Proxmark/proxmark3/files/902821/proxmark3.txt)
[trace.txt](https://github.com/Proxmark/proxmark3/files/902822/trace.txt)

proxmark3> data rawdemod am

Using Clock:32, Invert:0, Bits Found:513          
# Errors during Demoding (shown as 7 in bit stream): 4          
ASK/Manchester - Clock: 32 - Decoded bitstream:          
0000000000000000
0000000000000000
0000000000177000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000177
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
0000000000000000
1770000000000000
          
Segmentation fault: 11

masterix21, 2017-04-06:
@pwpiwi yes, but with your procedure it compile with guidummy.

marshmellow42, 2017-04-06:
hmmm.  i assume you can repeat the issue and it crashes every time.

can you try changing line 172 in client/util.c to 	
`for (size_t out_index=0; out_index < max_len-1; out_index++) {`

and trying again?

masterix21, 2017-04-06:
same problem

marshmellow42, 2017-04-06:
hmmm...  i cannot seem to reproduce the issue on windows or linux.

can you try changing line 163 to
`max_len = ( len+(len/breaks) > MAX_BIN_BREAK_LENGTH ) ? MAX_BIN_BREAK_LENGTH : len+(len/breaks)-1;`

my guess is we are going beyond the limits of the data[] array.  but we shouldn't be...

masterix21, 2017-04-06:
i found the exact line where you should bypass the error.

if you change the file "client/cmddata.c" at line 250 from:
`uint64_t lo = 0;`

to:
`return 1;`

it works. 

I know that isn't a solution, but the problem seems to be "uint64_t lo = 0;".

marshmellow42, 2017-04-06:
that appears then to point to the AskEm410xDecode function.  (not the defining of uint64_t lo)?  
can't you just comment out the line with `AskEm410xDecode(true, &hi, &lo);` and see if that is working

masterix21, 2017-04-06:
yes, the problem is `AskEm410xDecode(true, &hi, &lo);`: confirmed.

for test, i changed `if (emSearch) {` to `if (false && emSearch) {` and works without errors.

marshmellow42, 2017-04-06:
after the lf search u
please run
`data setdebug 2`
then do the 
`data rawdemod am` 
and post the output

masterix21, 2017-04-06:
[debug.txt](https://github.com/Proxmark/proxmark3/files/903232/debug.txt)

The error is caused by `memcpy(BitStream, DemodBuffer, BitLen);` line 152 file client/cmdlfem4x.c

marshmellow42, 2017-04-06:
also can you make sure your code is up to date with the master, your line numbers appear off...

masterix21, 2017-04-06:
My mistake, sorry. Line line 151 file client/cmdlfem4x.c
`memcpy(BitStream, DemodBuffer, BitLen);`

marshmellow42, 2017-04-06:
how do you know it is that line specifically?

masterix21, 2017-04-06:
Because works with a "return" before, and with a "return" after fails :)

Isn't the best way to debug, but is a way 👯

marshmellow42, 2017-04-06:
thanks for all the debugging work (i still cannot reproduce the error...)
can you add this line before the memcpy call:
`PrintAndLog("DemodBLen: %d, bitstreamsize: %d", BitLen, sizeof(BitStream));`
and let me know what it output.

masterix21, 2017-04-07:
DemodBLen: 513, bitstreamsize: 320000

marshmellow42, 2017-04-07:
@pwpiwi could I be allocating too much memory with the arrays?

pwpiwi, 2017-04-07:
Unlikely. 32Bit programs can access 2GBytes. I didn't have a look at the code yet, but you surely don't allocate that much?

masterix21, 2017-04-07:
i think no, because if you call two times "lf search u" it works, but if you call "lf search u" before, and next "data rawdemod am" it fails. 

The problem could be in the data memory, that you populate after "lf search u". Another difference that i saw is that "lf search u" print data ASK/Manchester only with "0", but if you call "data rawdemod am" you will see "177": seems "data rawdemod am" modify the original "am" printed by "lf search u" because proxmark found errors during demoding.

Please see "proxmark3.txt" file attached upon.

marshmellow42, 2017-04-07:
The rawdemod has errors because you didn't tell it to search for the ST that your tag has.  It is optional in the rawdemod but default in lf search u.

marshmellow42, 2017-04-07:
If you did a 'data rawdemod am s' it would be identical to the askmandemod the lf search calls.

masterix21, 2017-04-07:
@marshmellow42 are you on skype? If you add me i can give you the access to my PC.

marshmellow42, 2017-04-07:
Sorry not at my PC atm.

marshmellow42, 2017-04-07:
let's start over...  can you compile and test my fork https://github.com/marshmellow42/proxmark3.git 
(note that you will have to flash the fullimage.elf before the client will work with the device)
i've been making minor fixes there before they get back to the master.

if the problem still exists there we will go from there.

masterix21, 2017-04-10:
iMac:proxmark3-marshmellow42 masterix21$ ./client/flasher /dev/cu.usbmodem14131 armsrc/obj/fullimage.elf
Loading ELF file 'armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0002b370->0x0002b370) [R X] @0x94
1: V 0x00200000 P 0x0012d370 (0x000019ec->0x000019ec) [RW ] @0x2b404
Note: Extending previous segment from 0x2b370 to 0x2cd5c bytes

Waiting for Proxmark to appear on /dev/cu.usbmodem14131. Found.
Entering bootloader...
(Press and release the button only to abort)
Waiting for Proxmark to reappear on /dev/cu.usbmodem14131.... Found.

Flashing...
Writing segments for file: armsrc/obj/fullimage.elf
 0x00102000..0x0012ed5b [0x2cd5c / 359 blocks]....................................................................................................................................................................................................................................................................................................................................................................... OK

Resetting hardware...
All done.

Have a nice day!

iMac:proxmark3-marshmellow42 masterix21$ ./client/proxmark3 /dev/cu.usbmodem14131
dyld: Library not loaded: /usr/local/opt/qt/lib/QtGui.framework/Versions/4/QtGui
  Referenced from: /Users/luca/proxmark3-marshmellow42/./client/proxmark3
  Reason: image not found
Abort trap: 6

masterix21, 2017-04-10:
Yeah! now it works :) bug solved in your fork

marshmellow42, 2017-04-10:
Thanks for confirming.  I'll push to the master.

PR #274 (2017-04-10): lf read adjustments
`lf read` and `lf snoop` now automatically loads samples to graphbuffer (data samples) 
`lf read` can take argument for # of samples to collect. (it will only sample that many)
cleanup t55x7
fix #269 (buffer lengths)
fix pskclockdetect bug
fix lf t55xx detect bug during ST check
closed 2017-04-10 by pwpiwi