Proxmark/proxmark3 #270: Build fails on Arch Linux

ikarus23, 2017-04-06:
Hi,

building the current master on my arch linux fails with the following error:
```
g++ -MT obj/proxgui.o -MMD -MP -MF obj/proxgui.Td -I/usr/include/qt/QtWidgets -I/usr/include/qt -I/usr/include/qt/QtGui -I/usr/include/qt -I/usr/include/qt/QtCore -I/usr/include/qt -Wall -O4 -c -o obj/proxgui.o proxgui.cpp
In file included from /usr/include/qt/QtGui/qtguiglobal.h:43:0,
                 from /usr/include/qt/QtWidgets/qtwidgetsglobal.h:43,
                 from /usr/include/qt/QtWidgets/qapplication.h:43,
                 from /usr/include/qt/QtWidgets/QApplication:1,
                 from proxguiqt.h:11,
                 from proxgui.cpp:12:
/usr/include/qt/QtCore/qglobal.h:1097:4: error: #error "You must build your code with position independent code if Qt was built with -reduce-relocations. " "Compile your code with -fPIC (-fPIE is not enough)."
 #  error "You must build your code with position independent code if Qt was built with -reduce-relocations. "\
    ^~~~~
make[1]: *** [Makefile:196: obj/proxgui.o] Error 1
make[1]: Leaving directory '/home/gklostermeier/Programs/proxmark3/client'
make: *** [Makefile:33: client/all] Error 2
make  13.98s user 0.56s system 96% cpu 15.096 total
```

I fixed this by adding `-fPIC` [at this line](https://github.com/Proxmark/proxmark3/blob/master/client/Makefile#L37). Is this the correct way to get rid of this? Can someone confirm this. If so, I will patch it in the master.

pwpiwi, 2017-04-06:
In general, the line is correct (plus all the other lines with `CXXFLAGS`). I had the same with Kali Linux and qt5. With `-fPIC` I could compile but it then crashed at runtime. Maybe `CFLAGS` need `-fPIC` as well. I don't know if you can mix object files with and without `-fPIC`.

ikarus23, 2017-04-07:
I only added `-fPIC` to `CXXFLAGS` and so far I have no issues with that, even at run time.

pwpiwi, 2017-04-07:
I will test together with some more changes tonight.

Did you check if this has adverse effects on other OS and/or qt4?

ikarus23, 2017-04-07:
No, I only tested this on my arch linux, which has Qt4 and Qt5 installed.

pwpiwi, 2017-04-08:
Please try Pull Request #271

ikarus23, 2017-04-10:
I tried it and it worked like a charm.

pwpiwi, 2017-04-11:
Changes have been committed.

closed 2017-04-11 by pwpiwi