Proxmark/proxmark3 #272: Script Functionality is Broken as of b3ec0b3d

digitalentropy, 2017-04-10:
The commit on 4/1 appears to have broken listing/running scripts, at least in my macOS build environment.  It appears to be looking in the wrong working directory for the scripts folder.

pwpiwi, 2017-04-10:
Which commands do you use to install?

Where is your scripts directory?

digitalentropy, 2017-04-10:
My usual protocol is to “make clean” and “make all” from the main proxmark3 directory, which compiles both the firmware and the client.  The scripts directory is in the default location inside the client directory.

pwpiwi, 2017-04-10:
Any warnings or errors during compile?

digitalentropy, 2017-04-10:
I mean, sure, tons of warnings but no errors.  I don’t think I’ve ever seen a proxmark3 client compile on a system without any warnings.  Most of it is deprecated stuff though, nothing major.

Just to clarify though, the commit that it started exhibiting this behavior on was 5249ff94.  I thought I updated the issue but I guess it didn’t save the edit.

pwpiwi, 2017-04-10:
Two quite interesting statements:
>  the commit that it started exhibiting this behavior on was 5249ff9

... Which has no changes relating to the script directory

> sure, tons of warnings but no errors.  I don’t think I’ve ever seen a proxmark3 client compile on a system without any warnings.  

I  am used to zero warnings during the build process. Could you please post the log?

digitalentropy, 2017-04-10:
Yes, I also found it interesting for the same reason, but it appears to be fairly reproducible.  This is why I opened the issue, because I couldn't find any obvious changes myself and I was hoping someone else can lend their experience and troubleshooting it.

I've not observed the issue when compiling on Windows, and I haven't tested it on Linux.

I'm attaching two logs.  The first one is compiling 4ab135c0 from source.  You'll see the many clang warnings yelling about deprecated functions here and there, but nothing super scary.  You'll also see me run the client and have it work as expected.

The second, 5249ff94, appears to compile in the same way and interacting with scripts even works correctly the first time, and fails every subsequent attempt.  Even more fun, *something* is taking focus away from the terminal window right after I execute the proxmark3 binary (which doesn't happen on previous builds) AND I get a CoreAnimation error every time I exit.

I'm guessing this is going to be a very stupid issue, but I'm not sure where to focus yet.

[4ab135c0.txt](https://github.com/Proxmark/proxmark3/files/910531/4ab135c0.txt)
[5249ff94.txt](https://github.com/Proxmark/proxmark3/files/910532/5249ff94.txt)

iceman-1, 2017-04-10:
> I'm attaching two logs. The first one is compiling 4ab135c from source. You'll see the many clang warnings yelling about deprecated functions here and there, but nothing super scary. You'll also see me run the client and have it work as expected.
> The second, 5249ff9, appears to compile in the same way and interacting with scripts even works correctly the first time, and fails every subsequent attempt. Even more fun, something is taking focus away from the terminal window right after I execute the proxmark3 binary (which doesn't happen on previous builds) AND I get a CoreAnimation error every time I exit.

Iceman fork got a PR for an null terminal string bug.

pwpiwi, 2017-04-10:
@iceman-1: great! Will fix when I am back at keyboard.

pwpiwi, 2017-04-10:
Fix committed. You may try again, please.

digitalentropy, 2017-04-11:
Thanks @pwpiwi!  Script functionality has resumed but I'm still getting the following error upon exiting the client:

`CoreAnimation: warning, deleted thread with uncommitted CATransaction; set CA_DEBUG_TRANSACTIONS=1 in environment to log backtraces, or set CA_ASSERT_MAIN_THREAD_TRANSACTIONS=1 to abort when an implicit transaction isn't created on a main thread.`

digitalentropy, 2017-04-11:
Perhaps close this issue and open a new one?  I don't know if the two were related, but they popped up at the same time in the same commit for me.

pwpiwi, 2017-04-11:
> Perhaps close this issue and open a new one? I don't know if the two were related, but they popped up at the same time in the same commit for me.

They are not related (have seen it long before). Yes, please open a new issue.

btw: I have also committed some changes which should dramatically reduce the number of warnings during build. I would be interested to see which warnings remain...

closed 2017-04-11 by pwpiwi

digitalentropy, 2017-04-11:
Will do.

Remaining warnings enclosed.

[compiling.txt](https://github.com/Proxmark/proxmark3/files/912157/compiling.txt)