Proxmark/proxmark3 #273: hf legic sim is broken

ikarus23, 2017-04-10:
Hello!
as far as I can tell `hf legic sim` is broken. When I'm using the functionality I get the following output (tested against 3 different readers):

Reader 1 (Printer):
```
#db# Starting Legic emulator, press button to end          
#db# Unexpected: sz:7, Data:049, State:2, Count:3          
#db# IV: 055          
#db# Read Nb: 0, Addr: 62          
#db# Read Nb: 1, Addr: 69          
#db# Read Nb: 2, Addr: 66          
#db# Cycles: 9, Frame Length: 6, Time: 10177          
#db# Cycles: 20, Frame Length: 11, Time: 17365          
#db# Cycles: 56, Frame Length: 11, Time: 38616          
#db# Cycles: 92, Frame Length: 11, Time: 59840
```

Reader 2 (Access Control):
```
#db# Unexpected: sz:7, Data:06d, State:2, Count:6          
#db# IV: 055          
#db# Read Nb: 0, Addr: 0          
#db# Read Nb: 1, Addr: 1          
#db# Read Nb: 2, Addr: 2          
#db# Read Nb: 3, Addr: 3          
#db# Read Nb: 4, Addr: 4          
#db# Read Nb: 5, Addr: 46          
#db# Cycles: 9, Frame Length: 6, Time: 10183          
#db# Cycles: 20, Frame Length: 11, Time: 17721          
#db# Cycles: 52, Frame Length: 11, Time: 36440          
#db# Cycles: 83, Frame Length: 11, Time: 54545          
#db# Cycles: 115, Frame Length: 11, Time: 73005          
#db# Cycles: 147, Frame Length: 11, Time: 92157          
#db# Cycles: 189, Frame Length: 11, Time: 116625 
```

Reader 3 (Another PM3 with `hf legic reader`):
```
#db# Starting Legic emulator, press button to end          
#db# 0x19 - Frame: 019          
#db# Unexpected: sz:10, Data:02b, State:2, Count:0          
#db# IV: 055          
#db# Cycles: 9, Frame Length: 6, Time: 1015
```

On the PM3 reading the simulated tag (Reader 3) it outputs:
```
proxmark3> hf legic reader
#db# setting up legic card          
#db# MIM 1024 card found, reading card ...          
#db# !!! crc mismatch: expected 7 but got e !!!          
#db# operation aborted  
```

Am I doing something wrong? Has someone used `hf legic sim` successfully?

ikarus23, 2017-04-11:
Is there anybody who had used a simulated legic tag successfully? I've gone back to 4961e292aa4d1f2b6f35afa06e574d0f768ccb09 (2013) but sill no luck...

marshmellow42, 2017-04-11:
Not me, sorry.  Never played with legic.

iceman1001, 2017-04-11:
the main problem when rewriting the [b]hf legic *[/b] was lack of valid reader and exact timming with the crypto lsfr shifted.  So when rewriting the sim,  you need to keep track of when the lfsr increases,  without a valid reader to test responses to your simulated cmd I stopped with refactoring.

However when/if the sim will be completed, the whole new cmd structures would allow for some nice easy tools.  logging, eload/esave when sim, etc etc.  

I saw this refactoring as a stepping stone to understand the newer Legic advant.   I wll get some advant tags soon.  which will help.  I might buy a twn4 or something.

ikarus23, 2018-09-09:
I guess this can be closed since https://github.com/Proxmark/proxmark3/commit/1b902aa01afa39ff413b74142cd56d092b4500a1, right?

iceman1001, 2018-09-09:
yes

ikarus23, 2018-09-09:
Well then, big thanks to @drandreas!

closed 2018-09-09 by ikarus23

ikarus23, 2018-09-16:
I tested it on and it worked perfectly! Thank you so much!