Proxmark/proxmark3 #284: Segmentation fault on exit on Linux and osx

marshmellow42, 2017-04-21:
Seems to be an issue with the exit routine and associated with the qt graphing code.
@holiman when you find a little time can you take a look?

holiman, 2017-04-21:
Could do that, if I find some time. Any more info? Only when exiting, or at other times aswell ?

marshmellow42, 2017-04-21:
Seems to only appear when sending the exit cmd  from the Pm3 cli

marshmellow42, 2017-04-23:
ubuntu error example:
```user@ubuntuvm:~/master$ ./client/proxmark3 /dev/ttyACM0
ERROR: invalid serial port
proxmark3> exit
Segmentation fault (core dumped)
user@ubuntuvm:~/master$ 
```

digitalentropy, 2017-05-03:
I can confirm on macOS 10.12.4 that the actual segfault appears only when exiting via the exit command.  When breaking via Ctrl-C the issue does not appear.

I have also observed that I can tell when it's going to happen because when executing the client the focus will be taken away from my command window and I have to manually click back into it.  Whenever things start off that way, exiting will result in a segfault.  If I don't notice the focus change, then I do not get a segfault.

PR #302 (2017-06-04): fixing the crash on exit (issue #284)
Using a `QThread` instead of a `pthread` to run the `main_loop()`. I am still not confident if the cleanup (deleting objects) is OK. However, this happens just before exiting `main()` anyway. Should fix #284.
closed 2017-06-04 by marshmellow42