Proxmark/proxmark3 #285: PM3 under Windows

ikarus23, 2017-04-22:
Hi, I got an mail from an PM3 Windows user who is struggling with the installation.
I do not use Windows, so I'm not much of a help. Could some of you help him out?

>Dear Sir or Madam,
>
>I have recently purchased Proxmark 3 Elechouse V3 RF reading device. After
>several attempts to find software and installation guides I was still not
>able to install it. The installation process is complicated.
>
>I have tried following instructions from this manual
>
>https://github.com/Proxmark/proxmark3/wiki/Windows#getting-started
>
>however, on step 2 the GitHub for windows fails to install. the message
>tells: "application can not be started. contact the vendor"
>
>Could you please give me an advice on how to properly install the device.
>
>Thank you
>
>Best Regards

marshmellow42, 2017-04-23:
the github install error sounds like an issue for the github support.

as far as the windows wiki development install instructions.  yes they need to be updated as the proxspace environment is no longer functional.

pwpiwi, 2017-04-23:
> the proxspace environment is no longer functional

see PR #286. We could make it functional again.

iceman1001, 2017-06-05:
Does this issue still exist after latest fixes?  Time to close?

marshmellow42, 2017-06-05:
the old proxspace environment is not functioning without changes.  uic.exe needs to be added and i'm not sure how the hardnested will affect it.

i suggest we post a new proxspace environment. then close this issue.

pwpiwi, 2017-06-05:
I have updated the Wiki to link to @Gator96100's ProxSpace.

iceman1001, 2017-06-08:
time to close?

digitalentropy, 2017-06-11:
Certainly.  I just validated @Gator96100's ProxSpace on Windows 10. It appears to work for all the builds I tested back a couple months.

closed 2017-06-11 by iceman1001