Proxmark/proxmark3 #288: proxguiqt fails to compile

patrickdk77, 2017-04-28:
Need to include stdint.h somewhere for this function prototype.

g++ -MT obj/proxguiqt.o -MMD -MP -MF obj/proxguiqt.Td -Wall -O3 -DQT_SHARED -I/usr/include/qt4 -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtGui   -c -o obj/proxguiqt.o proxguiqt.cpp
In file included from proxguiqt.cpp:10:0:
proxguiqt.h:33:17: error: ‘uint8_t’ has not been declared
  void PlotDemod(uint8_t *buffer, size_t len, QRect r,QRect r2, QPainter* painter, int graphNum, int plotOffset);
                 ^
proxguiqt.cpp:306:6: error: prototype for ‘void Plot::PlotDemod(uint8_t*, size_t, QRect, QRect, QPainter*, int, int)’ does not match any in class ‘Plot’
 void Plot::PlotDemod(uint8_t *buffer, size_t len, QRect plotRect, QRect annotationRect, QPainter *painter, int graphNum, int plotOffset)
      ^
In file included from proxguiqt.cpp:10:0:
proxguiqt.h:33:7: error: candidate is: void Plot::PlotDemod(int*, size_t, QRect, QRect, QPainter*, int, int)
  void PlotDemod(uint8_t *buffer, size_t len, QRect r,QRect r2, QPainter* painter, int graphNum, int plotOffset);
       ^
proxguiqt.cpp: In member function ‘virtual void Plot::paintEvent(QPaintEvent*)’:
proxguiqt.cpp:510:85: error: no matching function for call to ‘Plot::PlotDemod(uint8_t [131072], size_t&, QRect&, QRect&, QPainter*, int, size_t&)’
   PlotDemod(DemodBuffer, DemodBufferLen,plotRect,infoRect,&painter,2,g_DemodStartIdx);
                                                                                     ^
proxguiqt.cpp:510:85: note: candidate is:
In file included from proxguiqt.cpp:10:0:
proxguiqt.h:33:7: note: void Plot::PlotDemod(int*, size_t, QRect, QRect, QPainter*, int, int)
  void PlotDemod(uint8_t *buffer, size_t len, QRect r,QRect r2, QPainter* painter, int graphNum, int plotOffset);
       ^
proxguiqt.h:33:7: note:   no known conversion for argument 1 from ‘uint8_t [131072] {aka unsigned char [131072]}’ to ‘int*’

marshmellow42, 2017-04-28:
Interesting that I never saw that on Linux...  

It should be placed in the proxguiqt.h file, along with string.h

patrickdk77, 2017-04-28:
Just to see if it worked, I added just stdint.h to that file and it compiled fine, without string.h
Figured it is a case of different qt headers.

marshmellow42, 2017-04-28:
I'm going to be away from a computer for a while so if you could make the change and issue a pull request I'll accept it into the repo.

marshmellow42, 2017-04-28:
string.h should be needed for the size_t, but just like my compiler never noticed the uin8_t yours must already have that defined.  
Best to include it as well.

closed 2017-04-28 by iceman1001