Proxmark/proxmark3 #291: Does flashing from HEAD work?

twitchyliquid64, 2017-05-16:
I had it all working from elechouse.

I just finished building and flashed the bootloader. It said all done, have a nice day etc.

Now, it appears as a USB device:

```shell
[18070.756603] usb 1-1.1: New USB device found, idVendor=9ac4, idProduct=4b8f
[18070.756608] usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[18070.756611] usb 1-1.1: Product: PM3
[18070.756614] usb 1-1.1: Manufacturer: proxmark.org
[18070.757028] cdc_acm 1-1.1:1.0: ttyACM0: USB ACM device
```

But is totally unresponsive otherwise, either by `./proxmark3` or otherwise.

The A & C LEDs are lit, occasionally the B & D flash. No idea what that means.

I think it only enumerates if I press the button.

```shell
[xxx@xxx-MS-7758]:~/projects/crap/proxmark3> sudo client/flasher /dev/ttyACM0 -b bootrom/obj/bootrom.elf armsrc/obj/fullimage.elf 
Loading ELF file 'bootrom/obj/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000ca0->0x00000ca0) [RWX] @0x298

Loading ELF file 'armsrc/obj/fullimage.elf'...
Loading usable ELF segments:
0: V 0x00102000 P 0x00102000 (0x0002b4a0->0x0002b4a0) [R X] @0x94
1: V 0x00200000 P 0x0012d4a0 (0x000019ec->0x000019ec) [RW ] @0x2b534
Note: Extending previous segment from 0x2b4a0 to 0x2ce8c bytes

Waiting for Proxmark to appear on /dev/ttyACM0. Found.
Sending bytes to proxmark failed

```

Have I bricked it? :(

I have a JLink lying around ... somewhere ... would rather not use it if there is any choice.

OS = Ubuntu 16.04 x64. The sourcecode I flashed is at HEAD.

marshmellow42, 2017-05-16:
Try flashing the bootrom and arm separately.  (Arm without the -b)

digitalentropy, 2017-05-17:
Yes, HEAD is quite functional right now.  As this issue appears to be more related to your local system or hardware rather than that source code, I would recommend moving your question to the proxmark3 forum at proxmark.org.

twitchyliquid64, 2017-05-21:
Thanks, ill try all of these when I get home.

Reading through the ./flasher code, its just a tight loop for flashing so my guess is:
1) Hardware borked
2) Flash screwed up in an unlucky way
3) Elechouse made uncompatible changes to the design
4) Flash succeeded but my computer is in a bad state

Worst case, I can probably get my hands on a JTAG flasher or cobble one together.

iceman1001, 2017-05-21:
As @marshmellow42  mentioned,  try flashing bootrom seperate.   If device doesn' work (connects) look at the udev/blacklist since you seens to be on a *nix distro.

iceman1001, 2017-06-08:
With the latest fixes to dev-envs such homebrew,   this one should not be an issue anymore.

Time to close?

twitchyliquid64, 2017-06-08:
gimme a few more days to find the time/effort to try and unbrick it lol. After that we can close :)

iceman1001, 2017-06-13:
try the btn-press trick:  press btn,  flash fullimage. releae btn. 
if not working try;  press btn; flash bootrom;  flash fullimage release btn.

pwpiwi, 2017-06-24:
You may also read http://www.proxmark.org/forum/viewtopic.php?id=4904 for usefull hints.

pwpiwi, 2017-07-04:
Close?

iceman1001, 2017-07-04:
close!

closed 2017-07-04 by pwpiwi