Proxmark/proxmark3 #292: Mifare Darkside attack imp

iceman1001, 2017-05-17:
Its an old attack and works.  However I re-read some old papers to see if I understood it better this time around.  Our darkside attack [3] involves the old one (nack leak) and the new differential one (28b).  The attack involves a online part and a offline part.  Where collecting leaked keystream bits via nack is done online.

In [2]  they say that the online part can be enhanced by the way we generate our nonce_reply. In curr that is the **mf_nr_ar** bytearray.   This is "find a combo nr_ar where the parity is all ok" and trigger the nack response.  The paper [2] (end of page11) suggest that using a random nr_ar to trigger the first nack,  is faster than using a fixed nr_ar all time on average.   

This is based on a claim 1500randon nonce/sec vs 30fixed nonces/sec.   To me it sounds like a trying to syncronise with a fixed nonce (repeating prng cycle)  vs just bluntly testing randon nonces. 

question;  have I understood this correct?   Is there no need to sync with the prng anymore?

**ref:**

1. http://proxmark.org/files/Documents/13.56%20MHz%20-%20MIFARE%20Classic/Implementing_an_RFID_MIFARE_CLASSIC_Attack.pdf
2. http://proxmark.org/files/Documents/13.56%20MHz%20-%20MIFARE%20Classic/Cryptanalytic_Attacks_on_MIFARE_Classic_Protocol-CT_RSA_2013.pdf
3. https://github.com/Proxmark/proxmark3/blob/master/armsrc/iso14443a.c#L2310

pwpiwi, 2017-05-18:
Interesting idea.

Re your question: we still would need to sync, but not for the first NACK, which takes the most time.

iceman1001, 2017-05-18:
Because if we don't sync (lock on to one nonce) we can't use the current offline implementation.  Which I belive is built around using gathered data from one fixed nonce. 

Now, is this randon nonce idea possible to impl?  Since I really don't want to fiddle in Courtois lsrf_common_prefix implementation (the differential 29b crypto attack).  Meaning if we only adapt deviceside impl, we should be able to get a faster Darkside attack?

cjbrigato, 2017-05-24:
Hello.
Would it really be so much of a gain since we have here : 
- no hardware rng on the AT91SAM7S
- very few entropy source I can think of
- an already quite enough constrained environment before even thinking of implementing prng

How much _different_ nonces would be "required" to be sent before the sync is considered either  failure or best ensured by fixed nonce attack ?
This question because precomputed list of nonce doesn't sound like anyhow an alternative here.

Unless I didn't understood something, it seems that context considered, benefits would be outweighted.

iceman1001, 2017-05-24:
The differential nonce alternativ takes about 32nonces on average to get a nack - respone,  whereas the fixed nonce attack takes 128nonces.  This is the speedup.  At most it should take 256nonces to get the first nack since there is only 8bits of parity to fiddle with.  The implemenation also tries to lock on to the prng-cycle.  This is the slow part, since it repeats every 618ms.  We need to gather 4 or 8 nacks, not sure actually.  This sync can take some tries,  for me its 6-10 times to find it,  then it get those nacks fast.

Using piwi's fast auth, 32auth/sec is doable,  I think I've averaged at 56auths/sec with it.  Thats 0.6sec.
6-10 resyncs = 10 * 6.18ms = 6.18sec
4|7 nacks (we already have one now) is  7*6.18 = 4.32 sec,
So data collecting time needed could be on max 11sec.  Now if we resync faster, so will the data collection also be. 
-- compared with
128 / 56 = 2sec  ( gives 12.5 sec tot)
256 / 56 = 4sec  ( gives 14.5 sec tot)
--
On average current darkside is 20-35sec runtime, this however includes the offline part.  Which should only be 1sec normally. The reason for 20-35sec usually for me is the resync part, especially when the nonce-distance is 1 to 8.  
--

The papers doesn't mention if there exists "better" nonces to test for the nack-bug.  Like using these pre-defined 32nonces we will get a nack in 99% of all the time.  256*4=1024bytes at most. Still doable. 

The current imp of darkside attack uses all zeros as the first nonce.

cjbrigato, 2017-05-24:
@iceman1001 I started with one of my well known "long reply is long", but had mercy
Long story short I think either :
* you have misread/misunderstood/mixed some paper parts
* you makes asumption that "averages" as biased from probability is anywhere near statistical average
* you have forgotten the implies of such probs, and what is "random" here (a full-space of ONE_TIME challenged (thus known or generable + tracked fi already challenged) such shorting on strict wording.

Long story short, there a no "256 at most", but "256 in average", which is just another way of saying p=1/256 or to get it better p'=255/256 of not actually getting the parity bits sets in such condition. This makes at least a 177 challenges to get anywhere near p~=1-p'~= .5 (a new way of abusing the "average" word), and a strict 256 (random&&different) challenges leads us to something near 0.63 of actually that result.
In this regard, we absolutely can't limit any random_fixed list.
Remember the argument for the random nonce  is built upon avoiding the need for the "reset" technique or periodic technic.
With nothing to enclose the "randomness" of all this, pm3 just cannot afford such even then random speedup.

At least not at all with the numbers you wrote here.

iceman1001, 2017-05-24:
Nop, not sure about it at all.
However the average 128 or 32, and 256  comes from the paper [2] in first post,  I just added execution times based upon them.  There is also the failure, ie, after 256 tries no nack response, possibility.
I belive the author didn't think in propabilities, more of actual measurements of results, when he wrote those numbers.  The number I have a problem with in the paper is the claim 1500/sec random nonces.  For me we need authentication dialogs to trigger the nackbug, which takes about 4-5ms according to NXP's own papers, not just nonces from tag.  So that claim seems invalid,  1000/5 = 200auths / sec seems more reasonable.   Hence I don't belive in the paper's speedup claims either.    But that doesn't mean their idea is wrong about getting some kind of speedup. 

Even your probability 177auths,  seems about right given current implementation,  which measures for me between 170-370 auths for the online part. 

So the rest of your input about pm3 device not being able to handle the 1500 nonces/s hardwarewise,  may not need to be considered since the claim 1500nonces/s is most likely false.

Of course, I may have totally missunderstood the paper and got my facts wrong.  Which was the main reason for me posting here to begin with.  To see if the new suggested way is actually better and the current one. The author is a accredited researcher, not me.

cjbrigato, 2017-05-24:
I was on paper 2 too and still I think you've misread some of his wording and assumptions.

> However the average 128 or 32, and 256 comes from the paper [2] in first post, I just added execution times based upon them. 

These numbers are quite more than one time cited for quite diverse case. Still your scenario seems nothing near any average use case.

> I belive the author didn't think in propabilities,

But indeed he use the very wording and nomenclature of probabilities all the way, as long as with pure theorical numbers (apart from the aproximation he made on auth speeds).

> here is also the failure, ie, after 256 tries no nack response, possibility.

256+ tries do not justify in any way a failure case in the random_nonce challenge process, for the very reasons of p=1/256 .
A "failure" to get a nack on such case would be to have got _PAST_ 255/256's of the whole nonce space. space without having any NACK at all. 
This equals to 2149055850 fail over a space of 2157483647. Even if that's unlikely, it definitely could happen.
I'm sorry but I still can't make any understanding of some of your assumption, may them be on the speedup gain potential or your understanding of Jovan's wording. If you can make a clear quoting of what makes you think anyhow 256 auth would be trustable upper bound of any of our darkside scenario, I could rest again

iceman1001, 2017-05-30:
Yes, I was unclear. 256 queries limit are for the fixed nonce.

However, I though on this as the process of nack triggering as being either fixed_nonce and changing parity.  or changing_nonce and fixed parity.    The option of changing_nonce and changing parity seems like finding a niddle in the hackstack.  As @cjbrigato  pointed out.  Maybe I missunderstood paper [2].
And now I have to re-read paper 2

This quote from [2], about the best strategy, which has the average claim on 256 tries for random nonce  
and 128 queries for fixed nonces. This sounds so strange to me.

> For random tag nonces, the keystream is also random and cannot be controlled by the fake reader. Therefore, the best strategy for the fake reader to get a successful query is to choose randomly {nR} , {aR} , and {p} until a successful query occurs, i.e., until all 8 parity bits in p are correct. On average, this requires 256 queries with random nT. For fixed tag nonces, if {nR} is kept fixed by the fake reader, then the keystream and nR will also be fixed. Further, if {aR} is also kept fixed, then aR will be fixed and, hence, all 8 parity bits in p will be fixed too. To get a successful query, the best strategy for the fake reader is then to choose different instead of random values of {p}. The average number of queries is thus reduced to 128=256/2.

I still think the better strategy would be having a random nonce with fixed parity. Why would random nonce and randon parity require on average 256 queries?!  Its like @cjbrigato  says,  like a neddle in a hackstack.

What to do?
So I asked Roel Verdult about this. He says he has verified Courtois numbers and claims, not Golic.  Maybe a minor speedup exists.  However, Dr Verdult thinks a correlation attack like the one in Hitag2 maybe be more successful in terms of speedups.  He also ended with a minor challenge,

> Hope you will be able to push the attacks to the limits. Good luck, I'm
> curious how fast you guys can get it ;)

Feeling inspired yet @pwpiwi  ? ;)

**ref**
-  http://www.cs.ru.nl/~rverdult/Introduction_to_Cryptanalysis-Attacking_Stream_Ciphers.pdf
-  https://www.usenix.org/conference/usenixsecurity16/technical-sessions/presentation/garcia

closed 2017-11-24 by iceman1001

iceman1001, 2017-11-25:
Closed due to inactivity.  No posts, no suggestions, no code,  == not interesting for community.

pwpiwi, 2017-11-25:
Well, it is more an academic than a practical question how fast we can make it. I assume this is the reason for the lack of interest. I am still interested though.

iceman1001, 2017-11-25:
I'm gonna focus on a  "nack" detection instead.  It looks like a detection algo would be almost the same as the darkside attack :/

iceman1001, 2017-11-28:
@pwpiwi   
haven't tested the speeds,  but this implementation seems interesting.
https://github.com/emsec/ChameleonMini/blob/master/Firmware/Chameleon-Mini/Application/Crypto1.c

iceman1001, 2017-12-06:
That was funny or interesting news today. Lessoned learn with making the **hf mf nack** command,   we could make a much faster  "darkside" attack for those cloned fudan etc card which ALWAYS leak nack. 

Two runs to collect the needed data,    one auth is down to 5ms.
8 auths * 2 * 5ms  == 80ms in total for online attack part.  Funny.