Proxmark/proxmark3 #294: Compile error in hfsnoop.c

ikarus23, 2017-05-26:
Hi, I get an error compiling the latest master.
```
arm-none-eabi-gcc -c -I../include -I../common -Wall -Werror -pedantic -std=c99 -Os -DWITH_ISO14443a_StandAlone -DWITH_LF -DWITH_ISO15693 -DWITH_ISO14443a -DWITH_ISO14443b -DWITH_ICLASS -DWITH_LEGICRF -DWITH_HITAG  -DWITH_CRC -DON_DEVICE -DWITH_HFSNOOP -fno-strict-aliasing -ffunction-sections -fdata-sections -DZ_SOLO -DZ_PREFIX -DNO_GZIP -DZLIB_PM3_TUNED -I../zlib -mthumb-interwork -o obj/hfsnoop.o hfsnoop.c 
hfsnoop.c: In function 'HfSnoop':
hfsnoop.c:57:9: error: increment of a boolean expression [-Werror=bool-operation]
     if (++trigger_cnt > triggersToSkip) {
         ^~
cc1: all warnings being treated as errors
make[1]: *** [../common/Makefile.common:84: obj/hfsnoop.o] Error 1
make[1]: Leaving directory '/home/ikarus/Documents/hack/rfid/proxmark3/armsrc'
make: *** [Makefile:31: armsrc/all] Error 2
```
Turns out, `trigger_cnt` is defined as `bool`. Should we change it to `uint16_t`?

marshmellow42, 2017-05-26:
It compares against triggersToSkip which is an int.  I'd match that.

marshmellow42, 2017-05-26:
Btw that is a multi year old bug (since hf snoop was first implemented...

pwpiwi, 2017-05-26:
Compilers are getting better still. @ikarus23 : which compiler version is this?

ikarus23, 2017-05-26:
@marshmellow42  Matching it with `int` from `triggersToSkip` sounds reasonable. I will do that.
@pwpiwi I'm using `gcc version 7.1.0`. I'm a Arch user, so this should be the latest stable version.

ikarus23, 2017-05-26:
Fixed by c87c452120aa30e0db9ca8046f00def677365ae6.

closed 2017-05-26 by ikarus23