Proxmark/proxmark3 #295: Can Proxmark3 simulate 4K card?

BlackPaaanda, 2017-05-26:
I dumped my own card and found it is a M 4K card. However, when i use "hf mf sim", the reader can only read 16 sectors while there are 40 sectors data stored in the proxmark3. I can retrieve all data using "hf mf eget <sector>", but how to simulate the 4K card using proxmark3?

iceman1001, 2017-05-31:
Its because the **hf mf sim** command is just made to simulate 1k/s50 cards,  as you might already have guess.
At the present you can't simulate any other size (mini/2k/4k) mifare classic.  The most of the code is implemented for 1k,  so extending it to different sizes wouldn't be too hard to do.  
Feel free to implement it!  

ref:
deviceside:  https://github.com/Proxmark/proxmark3/blob/master/armsrc/iso14443a.c#L2367
client: https://github.com/Proxmark/proxmark3/blob/master/client/cmdhfmf.c#L1122

iceman1001, 2017-07-24:
Question is answered. I'm closing.

closed 2017-07-24 by iceman1001