Proxmark/proxmark3 #296: hf mf dump 2 File reading error.

thesle3p, 2017-05-31:
Hi I get a ```File Reading error.``` bug when ever I try and dump a mifare 2k card. I have no trouble cracking the card or  reading individual blocks 
but when ever I run ```hf mf dump 2``` on the card I get that error. I confirmed that my dumpkeys.bin file is reachable etc. Here is the info on the card
```
proxmark3> hf search
          
 UID : db b4 a8 79           
ATQA : 00 04          
 SAK : 08 [2]          
TYPE : NXP MIFARE CLASSIC 1k | Plus 2k SL1          
proprietary non iso14443-4 card found, RATS not supported          
Answers to chinese magic backdoor commands: NO          

Valid ISO14443A Tag Found - Quiting Search
```

iceman1001, 2017-05-31:
That error message shows when the client fails to read the dumpkeys.bin file.
How did you create the dumpkeys file?

thesle3p, 2017-05-31:
Tried doing manual ```hf mf chk * ? ``` followed by ```hf mf nested 1 0 A <key>   d ```  as well as by running ```script run mfkeys ``` same result both times.

iceman1001, 2017-05-31:
and the results was?

thesle3p, 2017-06-01:
That's what got me here.......

thesle3p, 2017-06-01:
Update: when I do ```hf mf dump``` it is able to dump blocks (but not write to file) and tries to iterate past  the number of blocks on the card, I am guessing because it does not know it's a 2k card. it's only when I run ```hf mf dump 2```  that it fails to read dumpkeys.bin.

pwpiwi, 2017-06-01:
`hf mf dump` withou a size parameter assumes a 1K card, i.e. it shouldn't iterate past the number of blocks on a 2K card. What error messages do you get in this case?

`hf mf chk * d` without a size parameter assumes a 1K card and would write a dumpkeys.bin file with keys for the first half of a 2K card only.

`hf mf nested 1` : The 1 indicates 1K memory. You get only half of the keys of a 2K card.

It would really help if you paste the commands and the responses here.

pwpiwi, 2017-06-01:
BTW: How do you know that you have a 2K card?

iceman1001, 2017-06-13:
@thesle3p   how did it go?  You solved your issue?  Time to close?

iceman1001, 2017-07-24:
Since @thesle3p  has not answered the questions, I will close this one.  Open a new issue with more information if needed

closed 2017-07-24 by iceman1001