Proxmark/proxmark3 #299: New release

iceman1001, 2017-06-01:
With all breaking changes and long since the last one, its time for a new release.
I suggest v3.0.0 since this release has breaking and major changes.

I have started a draft here:
https://github.com/Proxmark/proxmark3/releases

Feel free to add descriptive text,   or let me know here.   When its ready, its just a publish away.

pwpiwi, 2017-06-01:
Agree with the version leap - mainly because of the confusion with asper's binaries.

But let's fix the crash on exit first?

iceman1001, 2017-06-01:
...haven't that crash been there since ages?  
... however, I did get a out-of-mem error in init_bitflup_statelists() in your code on ubuntu 1604...

pwpiwi, 2017-06-01:
The bug is there since ages - but annoying.

Maybe you are out of memory? You should have at least 4 GB. Some VMs have less with default settings.

iceman1001, 2017-06-01:
true, its kind of annoying if you are on a *nix distro.  Is there a solution for it now?
and true, my wm was on 1gb mem setting. will try 4gb now.

pwpiwi, 2017-06-01:
Regarding the bug I have an idea at least. Will try to fix next days.

iceman1001, 2017-06-01:
and yes,  it works with 4gb setting..

marshmellow42, 2017-06-04:
it appears to be time :)

iceman1001, 2017-06-05:
The new tag is published.  We got a v3.0.0 now.
This will be better for the newer users even if the pm3 client has grown in size. It will be better since we'll reduce the confusion of all pre-compile distros and their versionnumbers.   Finally this will be better since the current functionality of PM3 is greatly improved.
There is ofcourse problems still, the issues list both on the offical pm3 repo and on iceman repo tells the story of our battlescars. 

To you,  @pwpiwi  and @marshmellow42 I offer my sincere gratitude for your countless hours poured into this project.  Without ppl like you the RFIDsec scen would have look differently.

closed 2017-06-05 by iceman1001