Proxmark/proxmark3 #300: older compiler error with new hardnested

marshmellow42, 2017-06-01:
gcc 4.9.2 on mingw errors here: (similar to :https://github.com/p12tic/libsimdpp/issues/28 )
```cmdhfmfhard.c: In function 'get_SIMD_instruction_set':
cmdhfmfhard.c:72:2: error: Parameter to builtin not valid: avx512f
  if (__builtin_cpu_supports("avx512f")) strcpy(instruction_set, "AVX512F");
  ^
make[1]: *** [obj/cmdhfmfhard.o] Error 1
make: *** [client/all] Error 2

marshmellow42, 2017-06-01:
full gcc version:
```$ gcc --version
gcc.exe (i686-posix-dwarf-rev1, Built by MinGW-W64 project) 4.9.2
Copyright (C) 2014 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

pwpiwi, 2017-06-01:
Well, the compiler can only know CPU instruction sets which had been known when it came out. We either need to drop support for newer CPUs or for older compilers.

marshmellow42, 2017-06-01:
It appears others adjust based on detected compiler.  Is that not possible here?

Gator96100, 2017-06-01:
The new hardnested update did break my ProxSpace environment as well as my precompiled builds, because I'm using gcc 4.9.2 and get the same error. Adjusting based on compiler would be helpful (although updating gcc would probably a better solution).

marshmellow42, 2017-06-02:
it appears gcc 5.3 added avx512f support....  i suppose i should adjust my "fix" to check sub version...

marshmellow42, 2017-06-02:
https://gcc.gnu.org/gcc-5/changes.html says 5.3 is first support for avx512f
i've adjusted my fix (#301) to account correctly for that.  (unless someone finds a better way)

PR #301 (2017-06-02): make hardnested compile on gcc 4.9.2
plus add Td to gitignore.

unless someone has a better solution ;) this fixes #300 for me. (and probably @Gator96100 )
closed 2017-06-02 by iceman1001