Proxmark/proxmark3 #303: proxmark3 no longer compiles on Kali Nethunter (source hardnested)

thesle3p, 2017-06-06:
It looks like the implementation of the hardnested attack on Mifare Classic broke client compatibility with devices that run Kali Nethunter. As was discussed in #301 Android devices won't be able to run the SIMD code but I am not sure that's why the proxmark3 client no longer compiles. Here is the error I get on ``` make client``` as well as a standard ```make```

```
unrecognized command line option '-mno-sse2'; did you mean '-fno-dse'?
unrecognized command line option '-mno-avx'; did you mean '-Who-a'?
unrecognized command line option 'mno-avx2'; did you mean '-Who-a'?
unrecognized command line option '-mno-avx512f'
```

Would it be possible to add a target to the make file to  compile without hardnested? Assuming that is why the client no longer compiles.

marshmellow42, 2017-06-06:
what gcc version?

thesle3p, 2017-06-06:
4:6.3.0-4

pwpiwi, 2017-06-06:
These are machine specific switches. @thesle3p compiles for ARM, the x86 switches (SSE2, AVX, etc.) are not available. @thesle3p is right, we need some Makefile modifications.

marshmellow42, 2017-06-06:
i have a similar error in travis ci build env.
```gcc -MT obj/hardnested/hardnested_bf_core_NOSIMD.o -MMD -MP -MF obj/hardnested/hardnested_bf_core.Td -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O3 -DHAVE_GUI -mno-mmx -mno-sse2 -mno-avx -mno-avx2 -mno-avx512f -c -o obj/hardnested/hardnested_bf_core_NOSIMD.o hardnested/hardnested_bf_core.c
gcc: error: unrecognized command line option ‘-mno-avx512f’
make[1]: *** [obj/hardnested/hardnested_bf_core_NOSIMD.o] Error 1
make[1]: Leaving directory `/home/travis/build/marshmellow42/proxmark3/client'
make: *** [client/all] Error 2

iceman1001, 2017-06-06:
4.6 shouldn't compile with avc412, if using @marshmellow42  fixes

marshmellow42, 2017-06-06:
The makefile still contains the -mno switches.

iceman1001, 2017-06-06:
bummer...

pwpiwi, 2017-06-06:
@thesle3p: can you please test my pull request? Unfortunately I am lacking the necessary Nexus device...

marshmellow42, 2017-06-06:
i know this is about nethunter, but the new pull did not fix ubuntu 14.04 error:
```
gcc -MT obj/hardnested/hardnested_bf_core_NOSIMD.o -MMD -MP -MF obj/hardnested/hardnested_bf_core.Td -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O3 -DHAVE_GUI -mno-mmx -mno-sse2 -mno-avx -mno-avx2 -mno-avx512f -c -o obj/hardnested/hardnested_bf_core_NOSIMD.o hardnested/hardnested_bf_core.c
gcc: error: unrecognized command line option ‘-mno-avx512f’
make[1]: *** [obj/hardnested/hardnested_bf_core_NOSIMD.o] Error 1
make[1]: Leaving directory `/home/user/master/client'
make: *** [client/all] Error 2

thesle3p, 2017-06-06:
@pwpiwi  it compiles.

marshmellow42, 2017-06-06:
Awesome.  I'll create a different issue for Ubuntu.

marshmellow42, 2017-06-07:
The fix is committed.  closing this issue...

84matte84, 2017-08-15:
Hi guys,

I am having the same problem of " error: unrecognized command line option '-mno-mmx'" while trying to compile proxmark3 in Nougat (nethunter)...
I have followed the above discussion and tried to compile all the suggested forks but none of them worked. 

Somebody so kind to help me solving this issue please?
Thanks in advance...

pwpiwi, 2017-08-15:
Can you please type `uname -m` at a command prompt and post the result?

84matte84, 2017-08-15:
aarch64

pwpiwi, 2017-08-16:
Pushed a fix. Please try again.

84matte84, 2017-08-16:
@pwpiwi thanks it compiles now!
Though...it does not work... I get the ttyACM0 in the dmesg but when I try to connect to it via ./proxmark3 /dev/ttyACM0 it does nothing and even the log does not return anything...

pwpiwi, 2017-08-16:
Please search the forum for similar issues with the USB connection.

closed 2017-08-16 by pwpiwi