Proxmark/proxmark3 #305: compile issues on ubuntu 14.04

marshmellow42, 2017-06-06:
gcc 4.8.2

error:
```
gcc -MT obj/hardnested/hardnested_bf_core_NOSIMD.o -MMD -MP -MF obj/hardnested/hardnested_bf_core.Td -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../zlib -I/opt/local/include -I../liblua -Wall  -g -O3 -DHAVE_GUI -mno-mmx -mno-sse2 -mno-avx -mno-avx2 -mno-avx512f -c -o obj/hardnested/hardnested_bf_core_NOSIMD.o hardnested/hardnested_bf_core.c
gcc: error: unrecognized command line option ‘-mno-avx512f’
make[1]: *** [obj/hardnested/hardnested_bf_core_NOSIMD.o] Error 1
make[1]: Leaving directory `/home/user/master/client'
make: *** [client/all] Error 2

marshmellow42, 2017-06-06:
i believe i have a fix for this i'll publish after piwi's makefile changes are pushed.

pwpiwi, 2017-06-06:
The avx512 compiler options have been added in GCC 4.9

marshmellow42, 2017-06-06:
i found it was odd that my windows gcc 4.9.2 didn't complain about the make file switches but it did complain about the `__builtin_cpu_supports("avx512f"))` which evidently wasn't supported until 5.3

so i'll have some makefile tests coming.  :(

PR #306 (2017-06-07): fix compile error gcc lt 4.9
it isn't pretty, but it gets gcc 4.8.2 on ubuntu 14.04 to work (fixes #305)
tested on:
mingw 4.9.2
Kali 2016
closed 2017-06-07 by pwpiwi