Proxmark/proxmark3 #314: v3.0.1

iceman1001, 2017-06-08:
Since the compilation errors for some envs has been fixed;  I suggest a minor new release.
I've started the draft,  feel free to update it with text.  I also think we can add the "hitag2" PR before we release.

Link to draft
https://github.com/Proxmark/proxmark3/releases/tag/untagged-ed1a9243877b5b0b7434

pwpiwi, 2017-06-08:
I thought as well. This would finally be a working release for the homebrew tap.

CHANGELOG.md should be up to date before releasing.

iceman1001, 2017-06-08:
Something you will sort out now?

marshmellow42, 2017-06-08:
There are some new severe coverity issues with the hardnested and some other code I think should be addressed first.

marshmellow42, 2017-06-08:
The more I think on it fixing all the coverity scan issues will take time, once complete we should do another release, but the recent fixes warrants a release now as well.

pwpiwi, 2017-06-08:
CHANGELOG.md updated. Regarding the hitag2 PR I would prefer to have it tested by someone before merging.

iceman1001, 2017-06-08:
Neither marshmellow or I have access to hitag2 tags.   If you (piwi) don't,  we have to find someone who has.  I believe someone contacted me about it.  I'll check and see if I can get hold of that user to test it out for us.

iceman1001, 2017-06-09:
So, with @marshmellow42  latest PR merge.  Where are we on this one?

marshmellow42, 2017-06-09:
I'm not aware of any other compatibility issues.  I'm going to work on a few coverity issues, but they aren't high impact.

iceman1001, 2017-06-09:
and @pwpiwi ?  

Still no news about Hitag2 write PR.

marshmellow42, 2017-06-09:
before potentially breaking(low probability) the repo with new functionality (hitag2 write) i think we should go ahead with 3.0.1 release and then update the homebrew tap to use that release by default.

then we can go from there.

marshmellow42, 2017-06-09:
unless pwpiwi has something more to add/fix first..

iceman1001, 2017-06-09:
I've some minor fixes to hardnested..   But with my fork, I can't push PRs.  
I mailed @pwpiwi  about it.

marshmellow42, 2017-06-09:
you need a branch that is at the master head.  :p

iceman1001, 2017-06-09:
Despite what ppl think of me, I'm not that good at Git.  So I can make a branch based on pm3 master?!?  I didn't know that.  Third thing I learned today

pwpiwi, 2017-06-09:
Let's go for it. Seems to be quite settled as it is.

iceman1001, 2017-06-09:
Ok.  tag v3.0.1 is created. 
I've also made a PR for the homebrew repo.

closed 2017-06-09 by iceman1001