Proxmark/proxmark3 #316: rewrite wiki - Windowspage

iceman1001, 2017-06-09:
The windows wiki page needs to be updated with references to @Gator96100  's proxspace environment instead.  All the references to code.google.com should be changed into the wiki on github.

ref:
https://github.com/Proxmark/proxmark3/wiki/Windows

How to download:
- A local copy of Gator96100's proxspace env is copied to www.proxmark.org/files  
- Gator96100's github repo.

pwpiwi, 2017-06-09:
I had it updated a few days ago re Gator96100

iceman1001, 2017-06-09:
Yeah, I saw you changed the first paragraph. 
The rest of the text still references code.google, outdated 756-bin release, etc.

pwpiwi, 2017-06-09:
Yes, and we should make CDC driver install the default with HID being the exception instead vice versa.

marshmellow42, 2017-08-22:
i made a few adjustments to the windows wiki page.  it is a little more up to date

closed 2017-11-30 by iceman1001