Proxmark/proxmark3 #318: macOS Brew SHA256 mismatch error

joswr1ght, 2017-06-12:
Attempting to install the proxmark3 tap returns an error `SHA256 mismatch`:

```
Joshuas-MBP:~ jwright$ sw_vers
ProductName:	Mac OS X
ProductVersion:	10.11.6
BuildVersion:	15G1510
Joshuas-MBP:~ jwright$ rm -fr /Users/jwright/Library/Caches/Homebrew
Joshuas-MBP:~ jwright$ brew update
Updated 1 tap (caskroom/cask).
No changes to formulae.
Joshuas-MBP:~ jwright$ brew install proxmark3
==> Installing proxmark3 from proxmark/proxmark3
==> Downloading https://github.com/proxmark/proxmark3/archive/v3.0.1.tar.gz
==> Downloading from https://codeload.github.com/Proxmark/proxmark3/tar.gz/v3.0.1
######################################################################## 100.0%
Error: SHA256 mismatch
Expected: 83eec4277014b5265c68c8055602d793812048d83ff34d1f15009df8dfab381f
Actual: bace0dd34e35923bfd926cf0943e615a00b2588bb958a13afde3fd46cd34a821
Archive: /Users/jwright/Library/Caches/Homebrew/proxmark3-3.0.1.tar.gz
To retry an incomplete download, remove the file above.
```

iceman1001, 2017-06-12:
Nice catch!
Wrong repo to post this issue.  SHould have posted it "proxmark/homebrew-proxmark3"

Anyhow, I've have pushed a fix.  Would you mind testing again?

joswr1ght, 2017-06-12:
Fixed, thank you!

closed 2017-06-12 by joswr1ght