Proxmark/proxmark3 #323: How decode iso14443 protocol ?

maxben14, 2017-06-16:
How to decode the protocol of communication between the reader and the card? In my file recorded this protocol through the radio at the frequency of the mifare. I want to translate this into the format of the command's proxmark. Tell me how to do this?
I think use function `MillerDecoding`
https://github.com/Proxmark/proxmark3/blob/master/armsrc/iso14443a.c

marshmellow42, 2017-06-19:
You have a lot more research to do.  There is no direct answer for you.  ISO14443 has many protocols under it.  And you have not specified any hardware this is for, or if you are working on an open source project as is supported here.

Therefore there is no direct answer for you.  

Ultimately this would be a discussion to have on the forum as it doesn't appear to have anything to do with the further development of the proxmark3.

closed 2017-06-19 by marshmellow42

maxben14, 2017-06-19:
I read this atile http://anyram.net/blog_ru/?p=814 about sniffer mifare protocol.
In aticle there is file is sniff prototcol mifare iso14443a. I want from this file get command for example ataq, reqa. How this do ? Can me write how from file get Miller and Manchester cod ?

File sniff protocol mifare: https://yadi.sk/d/WJ5Wi-4h3KFxHt

marshmellow42, 2017-06-20:
I'm not sure any code there would be helpful to a SDR project.  There are many factors to consider.  But again this discussion is better suited to the proxmark3 forum.  http://www.proxmark.org/forum/index.php

marshmellow42, 2017-06-20:
The main reason I say the code may not help is the proxmark3 has an FPGA (hardware) that does preprocessing of the signal before it gets to the code you are looking at.  You can't just run raw signal through it.

pwpiwi, 2017-06-20:
With a radio you will pick up the reader signals only. The tag doesn't send actively.