Proxmark/proxmark3 #325: indala demod/clone command broken in R301

ntkntk, 2017-06-19:
indalapskdemod indalaclone have been changed. so the old indala code in GUI does not work anymore.

but while updating and test I see there is a fault in R301 either in the indala command report, or clone, 

this is one of the test I had last year

clone fbd87d6a
lf sea
lf se used to detect indala tag straight away previously.

REPORT:
proxmark3> lf indala
help             This help
demod            [clock] [invert<0|1>] -- Demodulate an indala tag (PSK1) from GraphBuffer (args optional)
read             Read an Indala Prox tag from the antenna
clone            <UID> ['l']-- Clone Indala to T55x7 (tag must be on antenna)(UID in HEX)(option 'l' for 224 UID
altdemod         ['224'] -- Alternative method to Demodulate samples for Indala 64 bit UID (option '224' for 224 bit)
proxmark3> lf indala clone 11223344
Cloning 64bit tag with UID 011223344
#db# DONE!
proxmark3> lf indala clone 1122334455667788
Cloning 64bit tag with UID 1122334455667788
#db# DONE!
proxmark3> lf indala clone 1122334455667788 1
Cloning 64bit tag with UID 2233445566778811
#db# DONE!
proxmark3> lf indala clone 1122334455667788 1
Cloning 64bit tag with UID 2233445566778811
#db# DONE!
proxmark3> lf sea
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

#db# Uknown frame length: 2
proxmark3> lf indala clone 1122334455667788
Cloning 64bit tag with UID 1122334455667788
#db# DONE!
proxmark3> lf sea
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

#db# Uknown frame length: 2
proxmark3> lf indala  clone fbd87d6a
Cloning 64bit tag with UID 0fbd87d6a
#db# DONE!
proxmark3> lf sea
NOTE: some demods output possible binary
  if it finds something that looks like a tag
False Positives ARE possible

Checking for known tags:

#db# Uknown frame length: 2
proxmark3>

but now indala failed

ntkntk, 2017-06-19:
Also the new command ares somehow confusing
proxmark3> lf indala
help             This help
demod            [clock] [invert<0|1>] -- Demodulate an indala tag (PSK1) from GraphBuffer (args optional)
read             Read an Indala Prox tag from the antenna
clone            <UID> ['l']-- Clone Indala to T55x7 (tag must be on antenna)(UID in HEX)(option 'l' for 224 UID
altdemod         ['224'] -- Alternative method to Demodulate samples for Indala 64 bit UID (option '224' for 224 bit)

I understand demod is for 64bit indata tag, altdemod offers the option to support 224bit. But seeing the comment "Alternative method to Demodulate samples for Indala 64 bit UID option '224' for 224 bit", i am lost. 

Something is "doppel gemoppert", too quick programming here.

marshmellow42, 2017-06-19:
The data you are cloning is not valid.  Therefore it won't read back a valid indala card.

And the option is a lower case L as it has always been not a 1.  Both demodulation routines off 224 support.   

This is not an issue or bug.

closed 2017-06-19 by marshmellow42