Proxmark/proxmark3 #334: RFC: Splitting proxmark3 into several libraries

micolous, 2017-06-25:
I want to build an Android version of the client. Not just "cross-compile the Linux version to run on rooted devices", but actually supporting a native Android GUI, have it work out of the box on non-rooted devices, and publishable on Play Store.  I've started working on this by having a project that selectively builds parts of the official client, and then calling them as a library.

I successfully pulled across tuning (including displaying the LF frequency graph) and basic 14a card reading and simulation functionality.  I rewrote client-to-PM3 communication code in Java, but hit difficulties with porting crypto-related functionality.  This required having wrappers into the proxmark3 codebase, attaching many client functions into Android frameworks (eg: logging), and stubbing out things that were broken.

Occasionally though, I need to make changes to proxmark3 core client code in order to make my stuff work better.  [The feedback I've received at the end of last year](https://github.com/iceman1001/proxmark3/pull/63#issuecomment-269196176) suggested that my best way forward with this is to rewrite this code into a library.

Between @iceman1001's popular fork and the mainline version there are protocol differences, where each side has conflicting flags and operations.  It would help to centralise this data so that it is easier to merge these trees together.

But the only way this project can be sustainable is if there's a library anyone can use, and that would include pulling proxmark3's official client onto this library as well.  Otherwise the two branches will diverge over time, and I'll end up making a repeat of @iceman1001's client.

I also note that [there is a Windows client](https://github.com/Proxmark/Gaucho-GUI) which has essentially the same challenges that I do.

There would need to be a few pieces to a split:

- `pm3-data`: Contains a registry of constants and flags used to build a PM3 client. Different clients and the `armsrc` code can pull this data and generate header/class files based on it, allowing them to have the same definitions.

   It wouldn't be necessary that each fork implements all of the features, but they could gracefully degrade based on some other functionality which would be out of scope for this project.

   Additionally, this could be used to build protocol documentation (something that doesn't exist today, and that is a major blocker to any reimplementation).

- `pm3-crypto`: Contains all the cryptographic functionality used by proxmark3.

- `pm3-clientlib`: Contains a client library which will emit sequences of hardware communication calls.  Implementers provide different hooks to address different communication methods.

- `pm3-cli`: Uses clientlib, and provides a CLI interface to interface with the PM3. This would be very minimalist, and map user entered commands onto `clientlib` calls.  It would replace `./client/proxmark`.

This would allow a few levels of approaches, based on what an implementor wants:

1. Attaching to `clientlib` would be ideal for those writing automation/scripting.

1. Attaching to `crypto` and `data` would allow a ground-up implementation of the client for those with special needs.

1. Attaching to `crypto` would be good for those using other hardware.

I'm looking for feedback on this proposal. It's not a quick fix and will take quite some time to implement, but my eventual goal is for this to replace the existing proxmark3 client, and make it less monolithic by decoupling concerns in different parts of the stack.

iceman1001, 2017-06-25:
To start off, its an interesting concept and I wish you luck of getting people jump on the project. 

Anectdote, when I started with pm3 , 2013, I though of making a pm3 client in C#. After starting it, I realised that there is so much knowledge coded in the client and it will be a gargantual effort to reproduce this into a new client.    Then after some months came Gaucho with the idea of wrapping it instead. Which enabled windows users to get a window-client and configuable with the settings-xml file.  All change and new commands can be easily added to it.  No need to release code. Perfect.  Just that project,  took some months.  Most of which was to write the settings-xml file.

To clarify, the GauchGUI does **not** try to solve your issues. It is just a plain wrapper around the pm3 client. It opens up the com port and sends same commands as you enter in the client. 
As I understand your idea, its to make a complete new client calling libs which is a different idea.

I've said that I'll not put effort in the remake of GauchoGUI project. Its just that I don't get anything out of it. I also have no need for an android client, nor do I have a need for more clients on other systems.  
That effort is for the ones who gets something out of it.

The PM3 project is over 10years and functionality has just been added and added.  Cleaning it up is a nightmare, trust me. I've tried.  So has @marshmellow42 and @pwpiwi also done.  It still many parts of code which hasnt been fixed up.  One issue with changing existing code is the need to be able to verfiy that the new code produces the same correct results as the old code.  So you need to have alot of different tags and readers to be able to mimic stuff.

micolous, 2017-06-25:
My first goal is to look for easy things that can be split up, then gradually pull all the implementation out of the standard client and put it into the library instead.

The reason why I thought this would be useful for the Gaucho client is that [I saw some reference to serial communication](https://github.com/Proxmark/Gaucho-GUI/blob/master/Proxmark%20Tool/frmMain.cs#L157).  That client has the luxury of being able to use standard serial IO libraries provided by the OS, where as for Android I'll need to wrap that up into the userspace USB-ACM driver I'm using, and bring that all back into Java land.

I acknowledge compatibility is going to be a nightmare. Which is why I think that the way to win this is **not** to entirely re-implement the protocol in another language (which is what I had been attempting until this point), but to push all the standard client's functionality into a library and call into that instead.  It just means that any time someone wants to port to a new platform (to the point of needing new communication routines), they supply some hooks in order to handle USB communication on that platform, and then start telling the library to do stuff for them.

I was mulling over changing the protocol entirely but I think that's asking for more trouble at this point.

micolous, 2017-06-28:
I've been chipping away at this in my `libmk1` branch a little. I'm thinking the best way I'll move forward with this is a series of smaller pull requests.

Next I'm planning to work on eliminating a bunch of global state with client-to-pm3 communications.

iceman1001, 2017-06-28:
It might be a good idea to have a seperate branch for your PM3-libs instead of re-writting the Stable PM3 master...

marshmellow42, 2017-06-28:
@micolous am i correct in saying that if this idea is kept separate from the main pm3 project then it breaks much of the intended outcome?  (one library for all to use so they don't diverge down different paths over time)...

marshmellow42, 2017-06-28:
also, converting all the client code into 4 separate librarys will drastically change all files and folder structure of the current repository.  this is a major change and one that will wreak havoc on all forks out there and all current ongoing development (as @iceman1001 pointed out earlier in another thread)

i believe there needs to be a lot more discussion on this topic from the admins and founders for this.

so if @micolous wants to continue work on it i agree with @iceman1001 that we use a new branch for now until what to do with it is decided upon.  

we should be able to cherrypick updates we definitely want for the master from the branch until then.

marshmellow42, 2017-06-28:
also on the topic of separate branch vs separate repo:
with a branch pulling select updates from the branch to the master should be pretty straight forward.  if we were to create a new repo it would make this more difficult and rely more on human file manipulations.

marshmellow42, 2017-06-28:
the benefits from a project such as this could be significant which is why i don't want to dismiss this quickly.  some brief thoughts i had on pros and cons are below.

Pros:

- the PM3 design is aging and new hardware may be desirable - getting the code into more a compatible framework might reduce the effort to upgrade to newer hardware designs.
- having separate libraries may encourage other uses and generate more developers that contribute to the project
- Segmenting the project may make it easier to test and contain bugs
- many old bugs will likely be found and fixed in the process
- refactoring and proper organizing of the code is needed

Cons:

- many new bugs may be introduced.  and some will be difficult to detect or test since many functions and features may only get used once a year or less for rare tag types...
- the code flow may get more complex for the novice developer.
- 

please add to this list...

micolous, 2017-06-29:
My intent with making a bunch of smaller changes is I want to avoid the position that @iceman1001 is in right now with his fork, where it's quite hard to reconcile the changes.  I could go ahead and deal with a bunch of stuff, and you'd be left with a huge PR that is unreasonable to review.  I'd also miss out on months of patches to the project, which would have to be refactored to some new API.

The fact that @iceman1001 uses `uart_set_speed` when @proxmark doesn't is something that really highlights my point.  To one side of the fork it is dead code that we can remove because it simplifies the API, and to the other it is critical code that is relied on.

We have git history if we ever want to "get stuff back", and at worst, the release tarballs.  You're right that a compiler _should_ delete unreachable code, but this also makes it difficult to understand the project.  It also should be possible to successfully compile the client with `-Werror` (which should resonate with the work you've already done with Coverity).

As an example, that AES `armsrc` code I deleted in #341 is a totally different API to the version which is included in the `client`.  And the version in `client` is only ever used by Lua scripts. So part of me says "why isn't there a library the Lua can call to do Proxmark stuff, and we make implementing crypto for Lua someone else's problem?"

I could make a seperate repo or branch, but the whole point of this exercise is to put the "official" client in a better position to be able to allow it to be used as a library as well, so when someone writes some functionality for the official client, then they also write it for the library.

I'm not greatly attached to the 4-way split, it is merely one option, it may be that we want to put some different splits that make sense for the desired outcomes.

After hacking on this for a bit, in the near term, a two way split seems more achievable:

- `libproxmark3`, which contains everything needed to communicate with the hardware, perform standard operations, and returns data in a machine-readable format (ie: a bunch of structs) and methods which directly map to existing pm3 commands.
- `proxcli`, which is a command line interface that parses machine-readable data from libproxmark3, and calls its API to do various commands.

Longer term, we can look at doing deeper splits in order to decouple crypto code or certain activities from the hardware that it needs to talk to, which would allow porting to other hardware, sharing with libnfc, or different interaction models (eg: completely changing the serial protocol).

The first major thing to be done is eliminating global state.  This can happen before a single library is created, and even while you have a monolithic `./proxmark3`.

The biggest piece of global state I need to handle is device connectivity.  This is would require that `ReceiveCommand` and `SendCommand` gets an extra parameter with a device reference.  While this is used in a lot of different places in the code, it is fairly cheap to fix it up.

This would also allow things like unit testing, that we're interacting with the hardware in the way that we expect, without requiring someone sit there with hundreds of tags and multiple PM3 units to see if they're doing the right thing.  These small steps can enable big wins.

marshmellow42, 2017-06-29:
you miss the point of the "dead functions".  many tags are only partially implemented with the eventual intent to fully support them.  this is a slow process with the few volunteer developers.  so functions are built that will ultimately be needed for full support for the tag type.
example: DESFire EV1.  has full AES support and would require it in at least the client but for simulation would also need it in the ARM.  the implementations may be required to be different as the arm has limitations that the client does not, so i'd expect a full implementation on the client and a simplified one for the arm.  work on desfire cards has extended over at least 5 years with large gaps in between and will continue (hopefully someday soon).

removing "dead functions" that have future application is counter productive.  
should functions have been created with similar design and inputs? yes, but this is difficult to enforce with limited resources.

and as far as the uart_setspeed() utilizing that function in the master repo is on my todo list.  which is why it was added in the first place.  as there are potential communication speed benefits to adjusting it.

marshmellow42, 2017-06-29:
and the -Werror is a preference thing.  we do use -Wall so all (well most) warnings are visible.  but some compiler versions complain about things that are not correct (bugs).  and the users here use just about every conceivable compiler version there is...

marshmellow42, 2017-06-29:
But I do agree with most of your idea's and comments!  and i welcome improvements!  Thanks!

just remember this project has been built mainly by novice volunteer developers (myself included) that don't do this for a living and did not always have an overall big picture of what others were doing and may just barely understand C.  And they built with the intent only to "make it work" for the function they wanted, nothing more.

this has created what we have now.  So yes, there are a lot of things that SHOULD be different.  it is a work in process.

any and all cleanup work will be welcomed, but scrutinized for compatibility with existing and known ongoing work.

iceman1001, 2017-06-30:
Just the idea of making the pm3 functionality available for other devices than pm3 says_new reposity_ to me.  If this is one of your quests then I suggest you to have a new repo under NFC-Tools account.  
This PM3 repo is made for the purose of a Proxmark3 device.  Much of its inner workings in client/device is based upon its a PM3 device its talking with.

The idea of cleaning up PM3 Master to be helpful for other possible clients is nice,  but for that purpose I recommend the structure like CryptoMiniSat, with serveral different branches.

Standalone vs client,  well,  good luck in finding cryptos that works on both sides without adding loads of  #ifdef  defines structures inside them.  Just look at the unified crypto1/crapto1 files now.  Did it become easier to understand?   No, I'm not pro those ones.  It just brings extra confusion to the project.

One major issue with PM3 project is that it runs on too many systems.  Its hard to please everyone with their own limits.  Too much time goes into helping the repo compile on someones favorit OS.  On top of it, we are the ones stuck with supporting it.   Nop, not for me anymore.  When ppl want to have support on icemanfork, they have to add it themself and I'll happily merge it but I ain't supporting it.  
Thanks for Travis CI.  

The enterprise design with interfaces and unittesting is also something hard to add now, since none sits with actual cards / readers to help out defining the exact purpose/actions/consequences on how this specific tag should act/react.   
What this project lacks the most is a somewhat detailed specification on how client/device interacts. As now every newcomer has to spend hours on following what does a specific usbcommand do and how not to call it.  I still get bitten by this.

micolous, 2017-07-01:
Software development isn't what I'm primarily employed for (though I do end up maintaining a lot of other peoples' code), and my motivation for doing this is my personal research projects. In order to achieve that, I need to maximise mobility of the proxmark3 hardware.

I'll steer clear of going on a deleting frenzy for now, that's not really required for my work.  You've made your position clear here. :)

I agree that proxmark3 client "is designed for proxmark3 hardware".  So I'd like to keep using bits of it for my own purposes, while I'm still using proxmark3 hardware.  There is a fair bit of interesting crypto functionality that would make sense in other contexts, and some of it is copied from other projects.

The reason I was suggesting having some tests in there is precisely to benefit you in the long term -- you can prove more than just "it builds".  While it is a significant investment (which I'm making, no you!), it benefits even amateur or single-serve contributors.

I'm not looking to take over maintenance of the project in the long term, which is why I'd like to get everything _which makes sense_ put into the mainline client.  I'm not saying "hey you should have a bunch of JNI bindings for Java", as that doesn't make sense for many people.

iceman1001, 2017-07-04:
...been thinking about the way forward,  when trying to solve a problem with the least amount of intrusion.
like if the problem is global state,   is the solution to shuffle it around everywhere in the code?  Making change on every single function call?!?  because that function migth use it? 

What is the actual problem we are having with global state?  One point would be making sure the state is gathered in one place,  to enable a separate lib.   I think this disussion needs to be taken first before there is a solution presented.     On the top of my head,  I still see no problem with having the struct (as presented now with serial, usbcommand array, debug etc)  in one place.  I think global state isnt evil per se.   Spread global variables everywhere in code is evil.    

_So before I'm even considering that PR, I want to hear what problem we are fixing with global state variables and is there less intrusive solutions to the problem_

pwpiwi, 2017-07-04:
Most of the time global state (global variables) isn't a good idea. It creates side effects where you don't expect them. It makes it hard to understand the code for new maintainers. For this and other reasons modern (object oriented) programming languages don't have global variables.

I personally don't care if functions are collected into libraries and whether we are linking one library or many object files. But getting rid of global variables is a good idea.

marshmellow42, 2017-07-04:
I do agree, that global variables are not ideal, and welcome this change.  (And yes I did contribute a few global variables with the lf demods, but that was just laziness :). )

iceman1001, 2017-07-05:
This project is not oo-based and sprinkled global variables over the code is a curse.  

So if I understand @OP  point of view, He don't mind cluttering up all functions in the current branch. ok, good to know for futher references when the next change comes along.

The lack of good arguments or alternative solutions makes me against this change still.

pwpiwi, 2017-07-05:
> sprinkled global variables over the code is a curse

Exactly. Therefore we try to get rid of them.

> So if I understand @op point of view, He don't mind cluttering up all functions in the current branch. ok

I don't mind too. But it's @micolous (or did you mean @marshmellow42 ?) not @op. Indeed, there are lots of changes but most of them are just adding an additional parameter to a function - not much to go wrong if several independend PRs can be made.

iceman1001, 2017-07-05:
The given solution is to stuff all global_Vars in a struct,   pass that struct around everywhere and the reason given is because globalvars is evil.  Where is the benefits for this?  I can see alternatives like having all global vars in one place and call that struct (?) whenever it is needed.   It solves the same problem, ie not having global_vars sprinkeled everywhere.   So no, I am not convinced at all. The lack of arguments makes me even stronger in my believe. 

When I write @OP , I mean the original poster.  The @OP of this thread is @micolous , you (@pwpiwi) seem confused.

pwpiwi, 2017-07-05:
I am probably not as confused as @op (https://github.com/op) might be.

Recommended reading: http://wiki.c2.com/?GlobalVariablesAreBad (with Pros and Cons):
>Really Bad Reasons to Use Global Variables
>  [...]
>  "I don't want to pass it around all the time."
>  [...]

😃

iceman1001, 2017-07-05:
*rotfl*  now that OP with have many messages..

That link talks about oo-solutions,  which still not complies with this project.   If that was the best argument for this suggested solution, you failed convincing me majorly.

pwpiwi, 2017-07-05:
Then its a tie. You didn't convince me either. :grin:

iceman1001, 2017-07-05:
When I look at it,  it really don't matter to me.  The icemanfork is old anyway.  I'll remove it soon.

pwpiwi, 2017-07-05:
Oh no, please don't do that. I am sure there is a lot more in the iceman fork besides magic tag detection, uart-setspeed, prng-detection, ...

marshmellow42, 2017-07-05:
Yes don't remove it just yet.  Merging many pieces of it, while time consuming, will still have many benefits.  In time I may find time to continue to help merge pieces of it.   But I don't think these changes affect that drastically, as there are already enough differences we can't copy and paste.

iceman1001, 2017-07-05:
this is beside the purpose of this issue,  and will not be discussed here.

cjbrigato, 2017-07-23:
_@iceman1001 I know you love that and you missed them, so here I come with my super-power : "Infinite Wall of Words". I tried but this RFC is in so many ways absurd than I couldn't._

## About the libs:
that would be so much work that it would for sure be faster to start from scratch a new client . But remember what @iceman1001 said about the "knowledge" and other old gods secrets that lie in the flows of old C89 here and there. With this in mind, it's either a no-go or an "up-to-you-to-fork". 
But that would still be strange idea. 
As @iceman1001  said, the Gaucho's work is a wrapper one. And as things are, and that's more true every day and as every new lines of codes spawn : Developing a wrapper is : 
- the most efficient
- the most ideal to concentrate on "your business" : an android gui (which is... a gui frontend to the client. nothing more. Why libraries when you got uart ?)
- the most portable
- the least footprint sized for your project with the least dependancies : I've got all homemade standalone statically compiled client (one binary to rule them all <3) built near automatically every night including the more obvious arm architectures (from armv5 to v7l and now including aarch64) and the fact is the the whole "proxmark" binary statically compiled with musl-libc, stripped, along with embedded lua's libs and scripts from the main tree is always something between 4 and 5MBs. Nothing more. Nothing else needed. Not even a root as long as you make enough native changes to work a little on how the uart is acccessed.
_(Just make it through the Android API if you can't manage to access the cdc device... which is 70% probable not to be even compiled in nor easily obtainable in this jungle of "near-3.X~linux kernels more or less correctly forward-ported to something they versioned 4.X, undocumented and a lot of time even not easily obtainable with full sources" android kernel anyway)_

## About anything else than this libs idea :
What do we do (people *actually* having to work on interoperability as more half of his worktime) when we're not given libs : everywhere we're given API's. Then that's it, just think of the client as your API. Because that's just when and that your best bet.

## But it's messy and undocumented
No it's not messy. And it's in it's king documented in a more efficent ways than all comment-generated (and never updated, full of old errors, bad copy/pasta etc, because it was "cool" at first but heh don't care it works) documentation I see everywhere : It's because the code truly talks for itself. And is his self documentation. It's explicit. It's not as DRY as possible and there definitely are repeated copy/pasted blocks (some of them from very strange but interesting way of thinking) . But it's as efficient with more readability, because it's a codebase that definitely is *taken care* of by the top tiers mainteners.
Don't like to parse some responses from the client ? Would like to specifically for this project handle some requests ? Well do it directly : the protocol is f*cking simple and very easy to go through. It's pure client-server of old days : 
- "UsbCommand" Structure + full protocol constants (thus "commands") may them be implemented client, server, or both sides with explicit names : https://github.com/Proxmark/proxmark3/blob/master/include/usb_cmd.h
- Receiving and sending (whatever the protocol would be. Here it's cdc over usb but it could be the very same "UsbCommand" structure passed over any socket. Want fun ? Fork Master, make it doable over socket (will need a basic as possible IP stack on the device, examples for such MCU are everywhere) Even that would be less work and more understandable than the proxlibgate. : https://github.com/Proxmark/proxmark3/blob/master/common/cmd.c
- Then it's straightfull to understand the commands : This block starting here is the entry point of near anything (in this context of having a documented protocol ) on the device side : https://github.com/Proxmark/proxmark3/blob/master/armsrc/appmain.c#L935 . Just jump to the func of your choice from there this will be your best documentation.
- Ok client side is quite more "messy" in this regard, as the cases are exploded more or less cleanly in their "cmd[Modulation/tag/whatever].c" with arguable choice made but still : that's all.

All is here, working, readable if you need specific.
When you dont, well then don't go low-level and just use the client as an API.
That's what I've done for my personal wild-client over... guess what... http : https://github.com/cjbrigato/proxmark-cjmark-designs. 

## Regarding this whole "RFC"
In general it is a very strange request you made to a well established project with such history (making the actual codebase more decent than a very lot of production systems with all the "legacy-retrocompatibility") and context : You're actually trying to *convict* the top tiers contributors of such project to be as inefficient as possible in making the next awaited implementation of whatever ("dead functions as it was called *cough*) available and that the whole code base should be over-complexily exploded into what I could call when reading this "Object Oriented Cognitive biased"-software architecture proposal ? You remind me of this random guy which get's up one morning and, as he only know C++ and sees something like God in it, decide to came anywhere he could to advocate what should be done the linux kernel : Rewriting it in C++ . That's not exageration if yo [... cut ...]  all the rest, they are out of the question.
But guss what, printf is quite good when quickly producing agnostic formated outpud to standard out. But near anything else, printf is out of the question. May we start a "printf is bad" crusade ? ;)
Very same for globals. And again considering the way history has chosen and kept the memory management deciding of such general policies and alikes seems quite saucy, even at their best I can't think of something else that the Impertinence lexical fields.

cjbrigato, 2017-07-23:
@iceman1001 I know you love that and you missed them, so here I come with my super-power : "Infinite Wall of Words". I tried but this RFC is in so many ways absurd I couldn't.

## About the libs:
that would be so much work that it would for sure be faster to start from scratch a new client . But remember what @iceman1001 said about the "knowledge" and other old gods secrets that lie in the flows of old C89 here and there. With this in mind, it's either a no-go or an "up-to-you-to-fork". 
But that would still be strange idea. 
As @iceman1001  said, the Gaucho's work is a wrapper one. And as things are, and that's more true every day and as every new lines of codes spawn : Developing a wrapper is : 
- the most efficient
- the most ideal to concentrate on "your business" : an android gui (which is... a gui frontend to the client. nothing more. Why libraries when you got uart ?)
- the most portable
- the least footprint sized for your project with the least dependancies : I've got all homemade standalone statically compiled client (one binary to rule them all <3) built near automatically every night including the more obvious arm architectures (from armv5 to v7l and now including aarch64) and the fact is the the whole "proxmark" binary statically compiled with musl-libc, stripped, along with embedded lua's libs and scripts from the main tree is always something between 4 and 5MBs. Nothing more. Nothing else needed. Not even a root as long as you make enough native changes to work a little on how the uart is acccessed.
_(Just make it through the Android API if you can't manage to access the cdc device... which is 70% probable not to be even compiled in nor easily obtainable in this jungle of "near-3.X~linux kernels more or less correctly forward-ported to something they versioned 4.X, undocumented and a lot of time even not easily obtainable with full sources" android kernel anyway)_

## About anything else than this libs idea :
What do we do (people *actually* having to work on interoperability as more half of his worktime) when we're not given libs : everywhere we're given API's. Then that's it, just think of the client as your API. Because that's just when and that your best bet.

## But it's messy and undocumented
No it's not messy. And it's in it's king documented in a more efficent ways than all comment-generated (and never updated, full of old errors, bad copy/pasta etc, because it was "cool" at first but heh don't care it works) documentation I see everywhere : It's because the code truly talks for itself. And is his self documentation. It's explicit. It's not as DRY as possible and there definitely are repeated copy/pasted blocks (some of them from very strange but interesting way of thinking) . But it's as efficient with more readability, because it's a codebase that definitely is *taken care* of by the top tiers mainteners.
Don't like to parse some responses from the client ? Would like to specifically for this project handle some requests ? Well do it directly : the protocol is f*cking simple and very easy to go through. It's pure client-server of old days : 
- "UsbCommand" Structure + full protocol constants (thus "commands") may them be implemented client, server, or both sides with explicit names : https://github.com/Proxmark/proxmark3/blob/master/include/usb_cmd.h
- Receiving and sending (whatever the protocol would be. Here it's cdc over usb but it could be the very same "UsbCommand" structure passed over any socket. Want fun ? Fork Master, make it doable over socket (will need a basic as possible IP stack on the device, examples for such MCU are everywhere) Even that would be less work and more understandable than the proxlibgate. : https://github.com/Proxmark/proxmark3/blob/master/common/cmd.c
- Then it's straightfull to understand the commands : This block starting here is the entry point of near anything (in this context of having a documented protocol ) on the device side : https://github.com/Proxmark/proxmark3/blob/master/armsrc/appmain.c#L935 . Just jump to the func of your choice from there this will be your best documentation.
- Ok client side is quite more "messy" in this regard, as the cases are exploded more or less cleanly in their "cmd[Modulation/tag/whatever].c" with arguable choice made but still : that's all.

All is here, working, readable if you need specific.
When you dont, well then don't go low-level and just use the client as an API.
That's what I've done for my personal wild-client over... guess what... http : https://github.com/cjbrigato/proxmark-cjmark-designs. 

## Regarding this whole "RFC"
In general it is a very strange request you made to a well established project with such history (making the actual codebase more decent than a very lot of production systems with all the "legacy-retrocompatibility") and context : You're actually trying to *convict* the top tiers contributors of such project to be as inefficient as possible in making the next awaited implementation of whatever ("dead functions as it was called *cough*) available and that the whole code base should be over-complexily exploded into what I could call when reading this "Object Oriented Cognitive biased"-software architecture proposal ? You remind me of this random guy which get's up one morning and, as he only know C++ and sees something like God in it, decide to came anywhere he could to advocate what should be done the linux kernel : Rewriting it in C++ . That's not exageration if you scale [... cut ...]
Very same for globals. And again considering the way history has chosen and kept the memory management deciding of such general policies and alikes seems quite saucy, even at their best I can't think of something else that the Impertinence lexical fields.

### PS2 : 
I decline responsabilities for the english quality you've been just reading. Be sure that I'll fight againts any goto and any pervert and mean global to ensure giving our children a better world that the one we had, a world without such behavior and imperfection. 
Also I'm french, so I just have a wine-augmented english.

micolous, 2017-10-09:
@cjbrigato Thanks for your feedback.  I don't mean to offend by anything said here, and I am not on a crusade to belittle past contributions.

I have a goal that I want to achieve.  My concern is the long term, which is why I released an RFC, to start some discussion.  _I recognise that not everyone will agree with my opinions, and that's OK._  Arguing about style and bugs in the existing implementation was probably not the best way for me to start this discussion, and there are parts of the plan I should scale back.

I will skip over addressing each of your points individually here, but I'll provide some historical context, a new (scaled back) plan which should address your major concerns, and background on alternatives I considered.

## Some historical context

You're probably unaware of the history on this, as I've been working on this quietly (and intermittently) for over a year.

I started some proof-of-concept work last year, in my own private repositories, but it is an ugly hackjob that is not maintainable.  I have re-implemented significant parts of the PM3 protocol in Java already, and that's something I really shouldn't do, and I'm planning on deleting that and starting again.

I made some comments on commits I was making, but it became clear to me that my minor patches wouldn't be sufficient to achieve the end goal, and some patches could not be reasonably included in the pm3 client.

Demo screenshots below (from October 2016):

[image]

[image]

[image]

## My new plan

I'm planning on scaling back the scope in order to make the goal easier to achieve.  However, elimination of global variables (and using a context pattern instead) is important for using the software as a library, rather than a standalone binary.  This will allow me to easily reset the state of the proxmark3 client context when destroying or recreating connections with the device if it goes away.

**My plan:**

1. Finish refactoring the communication related functionality. (this PR is pending review)
2. Create a new `pm3_context` struct which gets passed around to all commands, containing the majority of global state, including the flasher. It should be possible to create a new `pm3_context` without calling `main()`. (this needs a little bit of work, but I need to submit the comms refactor first)
3. Build some proof-of-concept (probably not using Java) showing how PM3 could be used as a library. (not started)

I'm not going to look at further major changes or splits to the PM3 codebase, unless they are strictly necessary to achieve my goal.  This means three-way library splits are off the table, and having the client not statically linked is probably also off the table.

I want to get stuck into the Android code during November, so I'd be aiming to complete the proxmark3 side work by then.

I'm not going to implement 100% of features in either the proof-of-concept library or the Android application.  But it should be possible for someone to easily add features without many changes to the PM3 client.

## Alternatives considered

You're right in saying that I could have written some refactored version of the PM3 client that works the way I want it to. I could have it done with a couple of weeks of work, and have avoided this entire discussion.

My concern is that if it wasn't picked up by anyone in the core team, upstream PM3 would diverge, it would require including patches again, I would get frustrated (as the features I really wanted are already there and working), I would lose interest and move on, and we would have another dead piece of code on the internet within a year.

The whole reason for writing this RFC is because I want to involve other PM3 contributors, because this is a significant change, and I've made mistakes on this journey already when I didn't ask questions and didn't talk to people about my plans.

If it gets too hard, then I'll take this route, but I really don't want to drop that on anyone.

W8M2Hg9lLmWqXSGC, 2017-11-01:
The changes proposed by @micolous should not go in to master. If anything this should be a branch.

micolous, 2017-11-27:
@W8M2Hg9lLmWqXSGC Can you please provide rationale?  These changes are being staged in a branch, and are part of code cleanup work I'm doing.

Branching the code and writing ugly hackjobs on the side of the mainline client is _exactly what I'm trying to avoid._  What I'm proposing would also make your GUI project a lot easier to achieve.

closed 2018-02-19 by micolous