Proxmark/proxmark3 #342: cmdhfmfhard.c undefined reference

magic2110, 2017-06-29:
Attempting to build proxmark3 bits on Raspberry Pi 3 (RPi3) Model B running Kali Linux for RPi3 v2017.1 (SHA256Sum: 4976C446802EE16252954453DC577E2001698492E52DDE47B27B8548C018A686)

After Running:

> sudo apt-get install git build-essential libreadline5 libreadline-dev gcc-arm-none-eabi libusb-0.1-4 libusb-dev libqt4-dev ncurses-dev perl pkg-config
> git clone https://github.com/Proxmark/proxmark3.git
> cd proxmark3
> make clean && make

Abbreviated Results:

> obj/cmdhfmfhard.o: In function `get_SIMD_instruction_set':
> /root/proxmark3/client/cmdhfmfhard.c:75: undefined reference to `__builtin_cpu_supports'
> /root/proxmark3/client/cmdhfmfhard.c:76: undefined reference to `__builtin_cpu_supports'
> /root/proxmark3/client/cmdhfmfhard.c:80: undefined reference to `__builtin_cpu_supports'
> /root/proxmark3/client/cmdhfmfhard.c:81: undefined reference to `__builtin_cpu_supports'
> /root/proxmark3/client/cmdhfmfhard.c:82: undefined reference to `__builtin_cpu_supports'
> collect2: error: ld returned 1 exit status
> Makefile:213: recipe for target 'proxmark3' failed
> make[1]: *** [proxmark3] Error 1
> make[1]: Leaving directory '/root/proxmark3/client'
> Makefile:33: recipe for target 'client/all' failed
> make: *** [client/all] Error 2
>

pwpiwi, 2017-06-29:
What is your gcc version? Please run `gcc -v` and post the results here.

magic2110, 2017-06-29:
> Using built-in specs.
> COLLECT_GCC=gcc
> COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/6/lto-wrapper
> Target: arm-linux-gnueabihf
> Configured with: ../src/configure -v --with-pkgversion='Debian 6.3.0-12' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --enable-default-pie --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
> Thread model: posix
> gcc version 6.3.0 20170406 (Debian 6.3.0-12)

marshmellow42, 2017-06-30:
it appears you may be using glibc version < 2.23  
see:
https://gcc.gnu.org/bugzilla/show_bug.cgi?id=71777

magic2110, 2017-06-30:
> root@kali:~# ldd --version
> ldd (Debian GLIBC 2.24-9) 2.24
> Copyright (C) 2016 Free Software Foundation, Inc.
> This is free software; see the source for copying conditions.  There is NO
> warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
> Written by Roland McGrath and Ulrich Drepper.

magic2110, 2017-06-30:
I managed to update the gcc version to 6.3.0 20170516 (Debian 6.3.0-18) but still fails as above.

Output for `gcc -v`:

> Using built-in specs.
> COLLECT_GCC=gcc
> COLLECT_LTO_WRAPPER=/usr/lib/gcc/arm-linux-gnueabihf/6/lto-wrapper
> Target: arm-linux-gnueabihf
> Configured with: ../src/configure -v --with-pkgversion='Debian 6.3.0-18' --with-bugurl=file:///usr/share/doc/gcc-6/README.Bugs --enable-languages=c,ada,c++,java,go,d,fortran,objc,obj-c++ --prefix=/usr --program-suffix=-6 --program-prefix=arm-linux-gnueabihf- --enable-shared --enable-linker-build-id --libexecdir=/usr/lib --without-included-gettext --enable-threads=posix --libdir=/usr/lib --enable-nls --with-sysroot=/ --enable-clocale=gnu --enable-libstdcxx-debug --enable-libstdcxx-time=yes --with-default-libstdcxx-abi=new --enable-gnu-unique-object --disable-libitm --disable-libquadmath --enable-plugin --enable-default-pie --with-system-zlib --disable-browser-plugin --enable-java-awt=gtk --enable-gtk-cairo --with-java-home=/usr/lib/jvm/java-1.5.0-gcj-6-armhf/jre --enable-java-home --with-jvm-root-dir=/usr/lib/jvm/java-1.5.0-gcj-6-armhf --with-jvm-jar-dir=/usr/lib/jvm-exports/java-1.5.0-gcj-6-armhf --with-arch-directory=arm --with-ecj-jar=/usr/share/java/eclipse-ecj.jar --with-target-system-zlib --enable-objc-gc=auto --enable-multiarch --disable-sjlj-exceptions --with-arch=armv7-a --with-fpu=vfpv3-d16 --with-float=hard --with-mode=thumb --enable-checking=release --build=arm-linux-gnueabihf --host=arm-linux-gnueabihf --target=arm-linux-gnueabihf
> Thread model: posix
> gcc version 6.3.0 20170516 (Debian 6.3.0-18)

Output for `apt-cache policy libc6`:

> libc6:
>   Installed: 2.24-12
>   Candidate: 2.24-12
>   Version table:
>  *** 2.24-12 500
>         500 http://http.kali.org/kali kali-rolling/main armhf Packages
>         100 /var/lib/dpkg/status

Output for `apt-cache policy gcc`:

> gcc:
>   Installed: 4:6.3.0-4
>   Candidate: 4:6.3.0-4
>   Version table:
>  *** 4:6.3.0-4 500
>         500 http://http.kali.org/kali kali-rolling/main armhf Packages
>         100 /var/lib/dpkg/status

pwpiwi, 2017-06-30:
I have added a pull request which should fix your issue. Either try it directly or wait until someone approves it and it is merged to master.

magic2110, 2017-06-30:
I tested it directly and this fixed the build error. I am now successfully able to manipulate my proxmark3 in the Kali Linux for RPi3 v2017.1 environment on my RPi3. Thanks so much for the help and quick fix!

closed 2017-06-30 by magic2110

pwpiwi, 2017-06-30:
...but most probably you will get an out of memory error when trying to run hf mf hardnested?

magic2110, 2017-06-30:
I don't have a mifare classic card to play with but I ran one of the examples just for fun. I saw no memory errors for it. I can test whatever command you'd like and report output back here if you'd like though.

> proxmark3> hf mf hardnested 0 A FFFFFFFFFFFF 4 A
> --target block no:  4, target key type:A, known target key: 0x000000000000 (not set), file action: none, Slow: No, Tests: 0           
> 
> 
>           
>  time    | #nonces | Activity                                                | expected to brute force          
>          |         |                                                         | #states         | time           
> ------------------------------------------------------------------------------------------------------          
>        0 |       0 | Start using 4 threads and no SIMD core                  |                 |          
>        0 |       0 | Brute force benchmark: 8 million (2^23.0) keys/s        | 140737488355328 |  200d          
>       42 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |  200d          
> Killed
> root@kali:~/proxmark3/client# 
>

pwpiwi, 2017-07-01:
You can try a simulation with `hf mf hardnested t 1`. Doesn"'t even require a proxmark.

magic2110, 2017-07-01:
proxmark3> hf mf hardnested t 1
--target block no:  0, target key type:A, known target key: 0x000000000000 (not set), file action: none, Slow: No, Tests: 1           

          
 time    | #nonces | Activity                                                | expected to brute force          
         |         |                                                         | #states         | time           
------------------------------------------------------------------------------------------------------          
       0 |       0 | Start using 4 threads and no SIMD core                  |                 |          
       0 |       0 | Brute force benchmark: 8 million (2^22.9) keys/s        | 140737488355328 |  204d          
      43 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |  204d          
Killed

pwpiwi, 2017-07-01:
Somehow this had to be expected (needs some 1,5GB memory) but it would have been nicer with an error message.