Proxmark/proxmark3 #348: HF 15693 tags issues

Fl0-0, 2017-07-04:
hf search does not detect 15693 cards correctly anymore

Bug probably in client code:

- Latest client with latest proxmark firmware:
With a 15693 tags:

```
Prox/RFID mark3 RFID instrument          
bootrom: master/v2.2.0-552-gebe9a3b-suspect 2017-07-04 15:43:52
os: master/v2.2.0-552-gebe9a3b-suspect 2017-07-04 15:43:53
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/05/17 at 17:48:26
          
uC: AT91SAM7S256 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 256K bytes. Used: 193020 bytes (74%). Free: 69124 bytes (26%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> hf search
```
LED A flashes continuously, tag not found.
Reset the proxmark  and **hf 15 cmd inquiry works**

- v3.0.1 client with latest proxmark firmware:
With a 15693 tags:

```
Prox/RFID mark3 RFID instrument          
bootrom: master/v2.2.0-552-gebe9a3b-suspect 2017-07-04 15:43:52
os: master/v2.2.0-552-gebe9a3b-suspect 2017-07-04 15:43:53
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/05/17 at 17:48:26
          
uC: AT91SAM7S256 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 256K bytes. Used: 193020 bytes (74%). Free: 69124 bytes (26%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> hf search
          
Tag UID : E004020052B53DE2          
Tag Info: NXP(Philips); IC SL2 ICS53/ICS54(SLI-S) ICS5302/ICS5402(SLIX-S)          

Valid ISO15693 Tag Found - Quiting Search

proxmark3> hf 15 cmd inquiry
UID=E004020052B53DE2          
Tag Info: NXP(Philips); IC SL2 ICS53/ICS54(SLI-S) ICS5302/ICS5402(SLIX-S)
```

pwpiwi, 2017-07-04:
@marshmellow42: with iclass code being the only relevant part being changed after v3.0.1, can you please have a look?  `hf search` calls `HFiClassReader("", false, false)` in https://github.com/Proxmark/proxmark3/blob/ebe9a3ba04eead5832879355434c20f78b33d7b2/client/cmdhf.c#L671. Is it possible that it never returns? Suspect line: https://github.com/Proxmark/proxmark3/blob/ebe9a3ba04eead5832879355434c20f78b33d7b2/armsrc/iclass.c#L1766

iceman1001, 2017-07-04:
second sample from @Fl0-0  shows a working **hf search** and  **hf 15 cmd inqury**

For me, the tag position is vital.  Making **hf search** miss iso15693 tag quite often.

Fl0-0, 2017-07-05:
Hello, I've commented the iClass part in CmdHFSearch(), it works again and never miss the tag !

```
/*
        ans = HFiClassReader("", false, false);
        if (ans) {
                PrintAndLog("\nValid iClass Tag (or PicoPass Tag) Found - Quiting Search\n");
                return ans;
        }
  */
```
```
Prox/RFID mark3 RFID instrument          
bootrom: master/v2.2.0-552-gebe9a3b-suspect 2017-07-04 15:43:52
os: master/v2.2.0-552-gebe9a3b-suspect 2017-07-04 15:43:53
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/05/17 at 17:48:26
          
uC: AT91SAM7S256 Rev B          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 256K bytes. Used: 193020 bytes (74%). Free: 69124 bytes (26%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> hf search
          
Tag UID : E004020052B53DE2          
Tag Info: NXP(Philips); IC SL2 ICS53/ICS54(SLI-S) ICS5302/ICS5402(SLIX-S)          

Valid ISO15693 Tag Found - Quiting Search
```

pwpiwi, 2017-07-05:
As @Fl0-0 already mentioned that the current firmware is OK, I am now suspecting
https://github.com/Proxmark/proxmark3/blob/ebe9a3ba04eead5832879355434c20f78b33d7b2/client/cmdhficlass.c#L195 as being the culprit. @marshmellow42 ?

pwpiwi, 2017-07-05:
Yeah, but this is the old code which probably had been changed by @marshmellow42 for a good reason. I am not completely sure about the reason for the change in this line but I think that 
```
if (readStatus == 0 && !loop) {
    // Aborted
    if (verbose) PrintAndLog("Quitting...");
    return 0;
}
```
should do the trick.

marshmellow42, 2017-07-05:
Sorry was sleeping. ;).   Yes @pwpiwi s code should do the trick.   I can do a pull request later if no one beats me to it.

marshmellow42, 2017-07-05:
there are other issues here as well.  i'm working on a fix.

closed 2017-07-05 by pwpiwi

Fl0-0, 2017-07-05:
hf search works fine with all my tags now, thanks !

PR #350 (2017-07-05): fix some iclass reader bugs
endless loop in reader if no tag was found
button press on pm3 did not cancel.
led_b was left on in some cases
also moved 14b detection to last in hf search to help speed up the
command for the other tags.  14b is slow (does multiple tests)
Thanks to @Fl0-0 and @pwpiwi for their idenfication of some of the
issues.

fixes #348