Proxmark/proxmark3 #353: Make a "snap" for Ubuntu and other Linux distros

jonpolak, 2017-07-11:
Was wondering if anyone tried to make a [snap](https://snapcraft.io/create/) of this? If not, I've started to make one.

iceman1001, 2017-07-14:
Go for it!  Btw, what is snap and how do you use it?

jonpolak, 2017-07-17:
@iceman1001  [What is a SNAP](https://www.ubuntu.com/desktop/snappy)

pwpiwi, 2017-07-17:
A homebrew for Linux?

iceman1001, 2017-07-17:
...so it distributes binaries?  or?   since pm3 project has a tendency to want users to use the latest source code, which is not always the latest release

And how is it going for you?

jonpolak, 2017-07-17:
I'm not in the mac world...so don't know about homebrew. It is a way to
have the latest source compiled automatically and people on linux can
update all their snaps with a single command. The client wont need to
download any dependencies -- it's a self contained binary that's
automatically compiled for each supported OS.

iceman1001, 2017-09-20:
time to close?

Since this not a issue,  
and the question is asked is "No,  noone has done a snap for pm3".
this might aswell be closed

jonpolak, 2017-09-20:
yes close.

closed 2017-09-20 by iceman1001