Proxmark/proxmark3 #360: Simulate a special 14a card

maozhenyu123, 2017-07-16:
I got a special 14a card. The UID saves in the block 1 

Communications between Reader and Card

Reader WUPA 52
Card Answer 03 00 as  ATQA
Reader 30 01(Read UID)
Card Answer XX XX XX XX CRC CRC
XXXXXXXX stands for UID

So....
How can I simulate this kind of cards? Should I build my own firmware to do do?

iceman1001, 2017-07-16:
On the top of my head,  you can write a lua-script and using **hf 14a raw** commands to simulate this tag.
or you can add it to **hf 14a sim** code on the deviceside,  where you input identify type / uid to be used as simulation.  in armsrc you make this specific behavior.  Quite easy I say

maozhenyu123, 2017-07-16:
Ehh. I edited some codes, recompiled the firmware and flashed it to my PM3. But......
It seems not working well.

Not responding on Mac///// On Windows, I got 0.00V after using "hw tune"

But other functions work

iceman1001, 2017-07-16:
thats strange..  are you sure you have the same firmware and client from the same repo?  
Don't mix them!

And look if the usbmodem### has changed

maozhenyu123, 2017-07-16:
I am pretty sure.....
And I use the new code(####). Failed.

I am gonna do that again.

maozhenyu123, 2017-07-18:
Cool! Done.

iceman1001, 2017-07-18:
close?

closed 2017-07-18 by maozhenyu123