Proxmark/proxmark3 #363: proxmark.org is down?

jsoiluva, 2017-07-25:
Hi Guys,

I know here is not a good place to post this question, but I just noticed proxmark.org or promark.nl are not reachable anymore. seems the site is totally down.

does anyone know what happened to our playground?
It would be a shame to see a lot of excellent topics disappeared like this.

iceman1001, 2017-07-25:
It happens sometimes.  It was up this morning,  down some time,  then up,  and now it seems to be down again.   Could be the webhosting company doing things with the server.
Its not like the site is gone forever,  I'm sure it will be up again soon.

jsoiluva, 2017-07-25:
@iceman1001 great to hear that. thank you for your reply. cheers

closed 2017-07-25 by iceman1001