Proxmark/proxmark3 #364: lf hid sim inserting spaces into ID

ShadyWushu, 2017-07-25:
lf hid sim seems to be inserting blank characters into the output tag ID.  For exampe, using lf hid read, we can get:

#db# TAG ID: 1234567890 (XXXXX) - Format Len: 34bit - FC 123 - Card: XXXXX

Using lf hid sim outputs the following

proxmark3> lf hid sim 1234567890
emulating tag with ID 12        34567890

In some cases, it eliminated the 3rd character:

proxmark3>lf hid sim 12a34b56c7
Emulating tag with ID 12        34b56c7

I have not been able to use lf hid sim to emulate cards against a reader.  For example, while I was able to use this technique is the past to open a HID controlled door by emulating my own card, this is no longer the case.

iceman1001, 2017-07-26:
Nice catch,  
I pushed a fix to my fork,   https://github.com/iceman1001/proxmark3/commit/e4ae456772f2232ae0cd221f27c3083b071f7584

pwpiwi, 2017-07-27:
Fix pushed to this repository, too.

closed 2017-07-27 by pwpiwi