Proxmark/proxmark3 #368: What is Mac OS 10.11.3 unable to read tag?(question)

johnreese2017, 2017-07-31:
[image]
always come up with invalid serial port
why?totally new on rfid 
ask for help

marshmellow42, 2017-07-31:
Because lf search is not the port your device is attached to.  Read the wiki.

johnreese2017, 2017-07-31:
sorry,still don't understand

johnreese2017, 2017-07-31:
sorry,where can I find the wiki of this?

iceman1001, 2017-07-31:
...when you read this thread...  raise your eyes up,  above your title row "why is mac os 10.11.3..",  there you see  **Code Issues Pull Requests Projects Wiki Insights**

Try clicking **Wiki** and start reading it.

johnreese2017, 2017-07-31:
This time it shows[image]

johnreese2017, 2017-07-31:
Thank You@iceman1001
And what I want to ask is which part of the wiki "definitely"

iceman1001, 2017-07-31:
how about we start from the beginning. 
*  hw version
*  hw tune
* lf search

Run those commands, and paste the output here so we can have a baseline of information.

johnreese2017, 2017-08-01:
no problems for now,thank you

closed 2017-08-01 by johnreese2017