Proxmark/proxmark3 #375: FPGA lo_simulate.v purpose / use

iceman1001, 2017-08-12:
FPGA code,   the file lo_simulate.v  indicates that its use is for  low frequency simulation. 
However I fail to see its ever been call inside fpga_lf.v,  not even referenced.  
There is no mode for it on the armside either.  

What has happend it?  Out of date because of izsh's improved edge_detect?   Since on arm,  we use the 
mode FPGA_MAJOR_MODE_LF_EDGE_DETECT when simulating LF.

Not to forget to mention that its not even called in the fpga/makefile

ref:
https://github.com/Proxmark/proxmark3/blob/master/fpga/lo_simulate.v
https://github.com/Proxmark/proxmark3/blob/master/fpga/Makefile

pwpiwi, 2017-08-16:
Correct. It has been replaced by lo_edge_detect.v

iceman1001, 2017-08-16:
ok

closed 2017-08-16 by iceman1001