Proxmark/proxmark3 #379: Unable to run and flash RDV on Nethunter

84matte84, 2017-08-16:
Dear all,

I have a proxmark3 RDV and I am trying to use it in my Android with Nethunter (Nexus 5x device).
Thanks to @pwpiwi , I was able to compile it but now when I execute it, it simply returns to the shell without any message:

```
root@kali:~/proxmark3/client# ./proxmark3 /dev/ttyACM0
root@kali:~/proxmark3/client#
```

Even though the device is correctly recognized:

```
root@kali:~/proxmark3/client# dmesg | grep tty
[ 7410.498640] cdc_acm 1-1:1.0: ttyACM0: USB ACM device
```

I have also tried to re-flash it but it does not work either:

```
root@kali:~/proxmark3/client# ./flasher -b ../bootrom/obj/bootrom.elf
Loading ELF file '../bootrom/obj/bootrom.elf'...
Loading usable ELF segments:
0: V 0x00100000 P 0x00100000 (0x00000200->0x00000200) [R X] @0x94
1: V 0x00200000 P 0x00100200 (0x00000cc8->0x00000cc8) [R X] @0x298
Attempted to write bootloader but bootloader writes are not enabled
Error while loading ../bootrom/obj/bootrom.elf
```

I tried to find something in the forum but without any luck...
Thanks in advance for any help

pwpiwi, 2017-08-16:
Please build the client again (`make clean && make client`) and post the compiler warnings/errors here.

84matte84, 2017-08-16:
Here you are the whole building log: https://pastebin.com/dqunQ10G
However, trying to execute it, I still have the same problem:

```
root@kali:~/proxmark3# cd client/
root@kali:~/proxmark3/client# ./proxmark3 /dev/ttyACM0
root@kali:~/proxmark3/client#
```

pwpiwi, 2017-08-16:
No errors, no warnings and compiles the correct files (no Intel specific SIMD cores).

If you run `./proxmark3 -h`, does it produce any output?

84matte84, 2017-08-16:
It correctly produces the help output.
I think it might be my proxmark3. I wanted to flash it but it did not work either.

Maybe I should try flahsing it from a PC and then trying to run it again from the Nexus?

pwpiwi, 2017-08-16:
Your flasher also behaves weird. It says `Attempted to write bootloader but bootloader writes are not enabled` although the `-b` option is there.

Do you eventually try to compile to 64Bit native code?

84matte84, 2017-08-16:
How?
Do you think it might also be a corrupted Nethunter installation? If so, I could also try reinstalling it...

pwpiwi, 2017-08-18:
`gcc -v` might give a hint. Or you can try to compile with option `-mabi=ilp32`. If you then get link errors, your environment is set up for 64 bit binaries.

pwpiwi, 2017-09-26:
Close due to no further interest?

pwpiwi, 2017-10-19:
@84matte84: still an issue for you?

pwpiwi, 2017-11-22:
Closed due to no feedback and lack of interest.

closed 2017-11-22 by pwpiwi

TomHarkness, 2018-08-29:
OS X is pretty stable now guys. As long as you don't install xquartz for the graph buffer gui. All .sh scripts are using OS detect and issuing correct commands for OS X.

pwpiwi, 2018-08-30:
Forgive me if this is a stupid question: what is the recommended setup to have a graph buffer gui on OS X then?

TomHarkness, 2018-08-30:
> https://github.com/Proxmark/proxmark3/issues/379#issuecomment-417183285

There is currently no way to run the client under OS X with the graph gui without serious performance issues. Hopefully someone can find a fix for this soon. I have been working on hardware so have had no time to look further in to it.