Proxmark/proxmark3 #383: PM3 as 14a tag doesn't work always

etmatrix, 2017-08-18:
I have a PM3 RDV2.0 from Electhouse with original antenna and custom antenna for emulate small tag.
I'm using PM3 with lastest version from github but I have an issue with tag emulation 14a.
I'm testing with many readers, some readers tag emulation is good but with one reader tag emulation is not working always, sometimes yes and sometimes no.
SCL3711 with libnfc is always OK.
PN532 breadboard with libnfc is always OK
HID OMNIKEY MultiISO not working very well.

I think the issue is on PM3 software. Maybe is timing but I have no idea how resolve it.
I develope a partial emulation of tag SRI4K 14b, this emulation with MultiISO is always working. I tried many position and I never seen a fail, almost when the distance is no more 5 centimers (1,9685 inches). Without move nothing when I acting 14a sim with hf 14a sim 1 11223344, MultiIso stop to work. There is a partial comunication, with hf list 14a I can see this log:

          Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |          
    ------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|          
              0 |       1056 | Rdr | 26                                                              |     | REQA          
           2228 |       4596 | Tag | 04  00                                                          |     |           
          22528 |      24992 | Rdr | 93  20                                                          |     | ANTICOLL          
          26164 |      31988 | Tag | 11  22  33  44  44                                              |     |           
          50168 |      60632 | Rdr | 93  70  11  22  33  44  44  51  9c                              |  ok | SELECT_UID          
          61804 |      65324 | Tag | 08  b6  dd                                                      |     |           
       10432422 |   10433478 | Rdr | 26                                                              |     | REQA          
       10434650 |   10437018 | Tag | 04  00                                                          |     |           
       10455078 |   10457542 | Rdr | 93  20                                                          |     | ANTICOLL          
       10458714 |   10464538 | Tag | 11  22  33  44  44                                              |     |           
       10482718 |   10493182 | Rdr | 93  70  11  22  33  44  44  51  9c                              |  ok | SELECT_UID          
       10494354 |   10497874 | Tag | 08  b6  dd                                                      |     |           
       17221156 |   17222212 | Rdr | 26                                                              |     | REQA          
       17223384 |   17225752 | Tag | 04  00                                                          |     |           
       17243684 |   17246148 | Rdr | 93  20                                                          |     | ANTICOLL          
       17247320 |   17253144 | Tag | 11  22  33  44  44                                              |     |           
       17271324 |   17281788 | Rdr | 93  70  11  22  33  44  44  51  9c                              |  ok | SELECT_UID          
       17282960 |   17286480 | Tag | 08  b6  dd                                                      |     |           

The reader should receive UID but doesn't show it on console, so it tell me not tag in field. Command used on MultiIso is S which try to select all type of tags (14a, 14b, 15963 etc...) in the field.
I tried also mifare simulate but without success with MultiIso. I tried an original mifare with MultiIso and I can select it without problem.
Only one time MultiIso selected my simulated mifare on about 30 tests, I don't understand what happened that time.

etmatrix, 2017-08-18:
I tried snoop comunication between MultiISO and a real mifare, PM3 has the same antenna of emulation tag, this is the result:

          Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |          
    ------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|          
              0 |       1056 | Rdr | 26                                                              |     | REQA          
           2244 |       4612 | Tag | 04  00                                                          |     |           
          22528 |      24992 | Rdr | 93  20                                                          |     | ANTICOLL          
          26180 |      32004 | Tag | 5e  fe  56  53  a5                                              |     |           
          50176 |      60704 | Rdr | 93  70  5e  fe  56  53  a5  2d  4a                              |  ok | SELECT_UID          
          61892 |      65412 | Tag | 08  b6  dd                                                      |     |           
       12946000 |   12947056 | Rdr | 26                                                              |     | REQA          
       12948244 |   12950612 | Tag | 04  00                                                          |     |           
       12968528 |   12970992 | Rdr | 93  20                                                          |     | ANTICOLL          
       12972180 |   12978004 | Tag | 5e  fe  56  53  a5                                              |     |           
       12996176 |   13006704 | Rdr | 93  70  5e  fe  56  53  a5  2d  4a                              |  ok | SELECT_UID          
       13007892 |   13011412 | Tag | 08  b6  dd                                                      |     |           

MultiIso show the UID in console without problem.
Time for trasmission is always 65412 carrier.

etmatrix, 2017-08-19:
MultiIso don't accept all UID, for it some UID is invalid, so for example 11223344, F1232322, F1232325  are not valid, but F43947DA, 87553b49 yes. So the problem is not PM3.

etmatrix, 2017-08-19:
I reopen the issue because I tried to set some UID on magic card and MultiIso select it without problem the same UID on PM3 is not working.
Example:

    hf 14a sim 1 1CD83E12

This is the trace:

          Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |          
    ------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|          
              0 |       1056 | Rdr | 26                                                              |     | REQA          
           2228 |       4596 | Tag | 04  00                                                          |     |           
          22528 |      24992 | Rdr | 93  20                                                          |     | ANTICOLL          
          26164 |      31988 | Tag | 1c  d8  3e  12  e8                                              |     |           
          50168 |      60632 | Rdr | 93  70  1c  d8  3e  12  e8  9e  5c                              |  ok | SELECT_UID          
          61804 |      65324 | Tag | 08  b6  dd                                                      |     |           
        3919050 |    3920106 | Rdr | 26                                                              |     | REQA          
        3921278 |    3923646 | Tag | 04  00                                                          |     |           
        3941578 |    3944042 | Rdr | 93  20                                                          |     | ANTICOLL          
        3945214 |    3951038 | Tag | 1c  d8  3e  12  e8                                              |     |           
        3969250 |    3979714 | Rdr | 93  70  1c  d8  3e  12  e8  9e  5c                              |  ok | SELECT_UID          
        3980886 |    3984406 | Tag | 08  b6  dd                                                      |     |           

and this is trace for snoop between MultiIso and magic card with the same UID

          Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |          
    ------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|          
              0 |       1056 | Rdr | 26                                                              |     | REQA          
           2260 |       4628 | Tag | 04  00                                                          |     |           
          22528 |      24992 | Rdr | 93  20                                                          |     | ANTICOLL          
          26196 |      32020 | Tag | 1c  d8  3e  12  e8                                              |     |           
          50176 |      60640 | Rdr | 93  70  1c  d8  3e  12  e8  9e  5c                              |  ok | SELECT_UID          
          61908 |      65428 | Tag | 08  b6  dd                                                      |     |           
        7003136 |    7004192 | Rdr | 26                                                              |     | REQA          
        7005380 |    7007748 | Tag | 04  00                                                          |     |           
        7025664 |    7028128 | Rdr | 93  20                                                          |     | ANTICOLL          
        7029316 |    7035140 | Tag | 1c  d8  3e  12  e8                                              |     |           
        7053312 |    7063776 | Rdr | 93  70  1c  d8  3e  12  e8  9e  5c                              |  ok | SELECT_UID          
        7065028 |    7068548 | Tag | 08  b6  dd                                                      |     |           

I see only a different time, magic card is more slow but MultiIso select it always never a fail.

pwpiwi, 2017-08-20:
The PM3 uses its internal 13,56MHz clock even when simulating a tag. A real tag would derive its clock from the reader's 13,56MHz carrier clock. I would assume that the clock of your MultiIso reader is off enough to cause issues to the PM3. At least this would explain why the MultiIso works fine with tags and why the PM3 simulation works with other readers.

etmatrix, 2017-08-20:
Thank you @pwpiwi for your explain. So the issue is not easy to solve. 
Why I have the issue only with some UID?

pwpiwi, 2017-08-21:
Answering your last question without being able to reproduce the issue would be wild guessing. Please note that the reader DOES recognize the ID (it is repeated in the select commands - 5th line in your traces). So part of the question would need to be answered by the reader manufacturer as well.

etmatrix, 2017-08-24:
I have found another reader which don't support all UID. I don't know which is the chip used in this reader but I have the same behavior of reader MultiISO. Stop comunication after select UID.
Maybe the question is for the reader manufacturer but why some UID is working and don't stop comunication after select uid?
I don't think there is a relationship between carrier clock and UID.

closed 2017-10-01 by pwpiwi