Proxmark/proxmark3 #386: Build fails with AVX512

uzlonewolf, 2017-08-26:
```
g++  obj/proxmark3.o obj/uart_posix.o obj/uart_win32.o obj/util.o obj/util_posix.o obj/crapto1/crapto1.o obj/crapto1/crypto1.o obj/polarssl/des.o obj/polarssl/aes.o obj/mfkey.o obj/loclass/cipher.o obj/loclass/cipherutils.o obj/loclass/ikeys.o obj/loclass/elite_crack.o obj/loclass/fileutils.o obj/whereami.o obj/mifarehost.o obj/parity.o obj/crc.o obj/crc16.o obj/crc64.o obj/iso14443crc.o obj/iso15693tools.o obj/data.o obj/graph.o obj/ui.o obj/cmddata.o obj/lfdemod.o obj/cmdhf.o obj/cmdhf14a.o obj/cmdhf14b.o obj/cmdhf15.o obj/cmdhfepa.o obj/cmdhflegic.o obj/cmdhficlass.o obj/cmdhfmf.o obj/cmdhfmfu.o obj/cmdhfmfhard.o obj/hardnested/hardnested_bruteforce.o obj/cmdhftopaz.o obj/cmdhw.o obj/cmdlf.o obj/cmdlfawid.o obj/cmdlfcotag.o obj/cmdlfem4x.o obj/cmdlffdx.o obj/cmdlfgproxii.o obj/cmdlfhid.o obj/cmdlfhitag.o obj/cmdlfio.o obj/cmdlfindala.o obj/cmdlfjablotron.o obj/cmdlfnexwatch.o obj/cmdlfnoralsy.o obj/cmdlfparadox.o obj/cmdlfpcf7931.o obj/cmdlfpresco.o obj/cmdlfpyramid.o obj/cmdlfsecurakey.o obj/cmdlft55xx.o obj/cmdlfti.o obj/cmdlfviking.o obj/cmdlfvisa2000.o obj/cmdlfpac.o obj/cmdparser.o obj/cmdmain.o obj/scripting.o obj/cmdscript.o obj/pm3_binlib.o obj/pm3_bitlib.o obj/protocols.o obj/sha1.o obj/cmdcrc.o obj/reveng/reveng.o obj/reveng/cli.o obj/reveng/bmpbit.o obj/reveng/model.o obj/reveng/poly.o obj/reveng/getopt.o obj/proxgui.o obj/proxguiqt.o obj/proxguiqt.moc.o obj/hardnested/hardnested_bf_core_NOSIMD.o obj/hardnested/hardnested_bitarray_core_NOSIMD.o obj/hardnested/hardnested_bf_core_MMX.o obj/hardnested/hardnested_bitarray_core_MMX.o obj/hardnested/hardnested_bf_core_SSE2.o obj/hardnested/hardnested_bitarray_core_SSE2.o obj/hardnested/hardnested_bf_core_AVX.o obj/hardnested/hardnested_bitarray_core_AVX.o obj/hardnested/hardnested_bf_core_AVX2.o obj/hardnested/hardnested_bitarray_core_AVX2.o obj/deflate.o obj/adler32.o obj/trees.o obj/zutil.o obj/inflate.o obj/inffast.o obj/inftrees.o -L/opt/local/lib -L/usr/local/lib -lreadline -lpthread -lm -ltermcap -lncurses ../liblua/liblua.a -ldl -lQt5Widgets -lQt5Gui -lQt5Core  -o proxmark3
obj/hardnested/hardnested_bf_core_NOSIMD.o: In function `crack_states_bitsliced_dispatch':
/home/lonewolf/proxmark/proxmark3/client/hardnested/hardnested_bf_core.c:556: undefined reference to `crack_states_bitsliced_AVX512'
obj/hardnested/hardnested_bf_core_NOSIMD.o: In function `crack_states_bitsliced_dispatch':
hardnested_bf_core.c:(.text+0x6c): undefined reference to `crack_states_bitsliced_AVX512'
obj/hardnested/hardnested_bf_core_NOSIMD.o: In function `bitslice_test_nonces_dispatch':
hardnested_bf_core.c:(.text+0x11f): undefined reference to `bitslice_test_nonces_AVX512'
hardnested_bf_core.c:(.text+0x124): undefined reference to `bitslice_test_nonces_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `malloc_bitarray_dispatch':
/home/lonewolf/proxmark/proxmark3/client/hardnested/hardnested_bitarray_core.c:324: undefined reference to `malloc_bitarray_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `malloc_bitarray_dispatch':
hardnested_bitarray_core.c:(.text+0x54): undefined reference to `malloc_bitarray_AVX512'
(snip)
hardnested_bitarray_core.c:(.text+0x7e4): undefined reference to `count_bitarray_AND3_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `count_bitarray_AND4_dispatch':
hardnested_bitarray_core.c:(.text+0x88f): undefined reference to `count_bitarray_AND4_AVX512'
hardnested_bitarray_core.c:(.text+0x894): undefined reference to `count_bitarray_AND4_AVX512'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:215: proxmark3] Error 1
make[1]: Leaving directory '/home/lonewolf/proxmark/proxmark3/client'
make: *** [Makefile:33: client/all] Error 2
[lonewolf@lonewolf proxmark3]$
```

Editing client/Makefile and adding `$(MULTIARCHSRCS:%.c=$(OBJDIR)/%_AVX512.o)` after `$(MULTIARCHSRCS:%.c=$(OBJDIR)/%_AVX2.o)` appears to fix, but I haven't tried running the result yet.

iceman1001, 2017-08-26:
OS and GCC ?
uname -a
gcc -v

pwpiwi, 2017-08-26:
Ah, finally we found someone who can test the AVX512 implementation. :grin:

Unfortunately your workaround would break compilation for those people with older gcc versions.

Could you please type `gcc -dumpversion` at a command prompt and post the result?

And I would be interested in the performance of the AVX512 core. You could run a test (Proxmark needs not to be connected) with `hf mf hardnested t 1 ffffffffffff`

uzlonewolf, 2017-08-26:
Fedora 26

Linux [hostname] 4.11.8-300.fc26.x86_64 #1 SMP Thu Jun 29 20:09:48 UTC 2017 x86_64 x86_64 x86_64 GNU/Linux

```
$ gcc -v
Using built-in specs.
COLLECT_GCC=gcc
COLLECT_LTO_WRAPPER=/usr/libexec/gcc/x86_64-redhat-linux/7/lto-wrapper
OFFLOAD_TARGET_NAMES=nvptx-none
OFFLOAD_TARGET_DEFAULT=1
Target: x86_64-redhat-linux
Configured with: ../configure --enable-bootstrap --enable-languages=c,c++,objc,obj-c++,fortran,ada,go,lto --prefix=/usr --mandir=/usr/share/man --infodir=/usr/share/info --with-bugurl=http://bugzilla.redhat.com/bugzilla --enable-shared --enable-threads=posix --enable-checking=release --enable-multilib --with-system-zlib --enable-__cxa_atexit --disable-libunwind-exceptions --enable-gnu-unique-object --enable-linker-build-id --with-gcc-major-version-only --with-linker-hash-style=gnu --enable-plugin --enable-initfini-array --with-isl --enable-libmpx --enable-offload-targets=nvptx-none --without-cuda-driver --enable-gnu-indirect-function --with-tune=generic --with-arch_32=i686 --build=x86_64-redhat-linux
Thread model: posix
gcc version 7.1.1 20170622 (Red Hat 7.1.1-3) (GCC) 
```
```
proxmark3> hf mf hardnested t 1 ffffffffffff
--target block no:  0, target key type:A, known target key: 0xffffffffffff, file action: none, Slow: No, Tests: 1           

          
 time    | #nonces | Activity                                                | expected to brute force          
         |         |                                                         | #states         | time           
------------------------------------------------------------------------------------------------------          
       0 |       0 | Start using 6 threads and SSE2 SIMD core                |                 |          
       0 |       0 | Brute force benchmark: 396 million (2^28.6) keys/s      | 140737488355328 |    4d          
       1 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    4d          
       4 |       0 | Simulating key ffffffffffff, cuid 335b4b8c ...          | 140737488355328 |    4d          
       5 |     113 | Apply bit flip properties                               |  14540378996736 |   10h          
       6 |     226 | Apply bit flip properties                               |  12005817712640 |    8h          
       7 |     339 | Apply bit flip properties                               |  10723405070336 |    8h          
       8 |     451 | Apply bit flip properties                               |  10723405070336 |    8h          
       9 |     564 | Apply bit flip properties                               |  10723405070336 |    8h          
      10 |     677 | Apply bit flip properties                               |  10688651067392 |    8h          
      11 |     789 | Apply bit flip properties                               |  10653897064448 |    7h          
      12 |     900 | Apply bit flip properties                               |   8528471261184 |    6h          
      13 |    1011 | Apply bit flip properties                               |   8404868792320 |    6h          
      14 |    1124 | Apply bit flip properties                               |   8378623459328 |    6h          
      15 |    1237 | Apply bit flip properties                               |   8378623459328 |    6h          
      17 |    1346 | Apply bit flip properties                               |   8378623459328 |    6h          
      17 |    1457 | Apply bit flip properties                               |   8378623459328 |    6h          
      20 |    1568 | Apply Sum property. Sum(a0) = 0                         |    130553528320 |  6min          
      20 |    1681 | Apply bit flip properties                               |    122364919808 |  5min          
      20 |    1792 | Apply bit flip properties                               |    122364919808 |  5min          
      20 |    1900 | Apply bit flip properties                               |    117800845312 |  5min          
      21 |    2009 | Apply bit flip properties                               |    117800845312 |  5min          
      21 |    2120 | Apply bit flip properties                               |    115630923776 |  5min          
      21 |    2230 | Apply bit flip properties                               |    114275516416 |  5min          
      21 |    2336 | Apply bit flip properties                               |    114275516416 |  5min          
      21 |    2448 | Apply bit flip properties                               |    113346232320 |  5min          
      22 |    2560 | Apply bit flip properties                               |    112386637824 |  5min          
      22 |    2670 | Apply bit flip properties                               |    112386637824 |  5min          
      22 |    2779 | Apply bit flip properties                               |    112378781696 |  5min          
      22 |    2779 | (1. guess: Sum(a8) = 0)                                 |    112378781696 |  5min          
      27 |    2779 | Apply Sum(a8) and all bytes bitflip properties          |     28755195904 |   73s          
      27 |    2779 | (Test: Key found)                                       |               0 |    0s          
     207 |    2779 | Brute force phase:  24.89%                              |     21903552512 |   55s          
     209 |    2779 | Brute force phase completed. Key found: ffffffffffff    |               0 |    0s          
```

pwpiwi, 2017-08-26:
Hmm, `Start using 6 threads and SSE2 SIMD core` seems to imply that your CPU supports SSE2 only, however the dispatcher seems to point to the AVX512 core. What is your CPU exact type?

pwpiwi, 2017-08-26:
Probably a compile time issue and you really have CPU supporting SSE2 only. Can you please post the output of `gcc -dumpversion `

Fl0-0, 2017-08-26:
Same problem for me !

```
obj/hardnested/hardnested_bf_core_NOSIMD.o: In function `crack_states_bitsliced_dispatch':
/home/flo/RFID/flo-proxmark3/client/hardnested/hardnested_bf_core.c:556: undefined reference to `crack_states_bitsliced_AVX512'
obj/hardnested/hardnested_bf_core_NOSIMD.o: In function `bitslice_test_nonces_dispatch':
/home/flo/RFID/flo-proxmark3/client/hardnested/hardnested_bf_core.c:577: undefined reference to `bitslice_test_nonces_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `malloc_bitarray_dispatch':
/home/flo/RFID/flo-proxmark3/client/hardnested/hardnested_bitarray_core.c:324: undefined reference to `malloc_bitarray_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `free_bitarray_dispatch':
/home/flo/RFID/flo-proxmark3/client/hardnested/hardnested_bitarray_core.c:345: undefined reference to `free_bitarray_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `bitcount_dispatch':
/home/flo/RFID/flo-proxmark3/client/hardnested/hardnested_bitarray_core.c:366: undefined reference to `bitcount_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `count_states_dispatch':
/home/flo/RFID/flo-proxmark3/client/hardnested/hardnested_bitarray_core.c:387: undefined reference to `count_states_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `bitarray_AND_dispatch':
/home/flo/RFID/flo-proxmark3/client/hardnested/hardnested_bitarray_core.c:408: undefined reference to `bitarray_AND_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `bitarray_low20_AND_dispatch':
/home/flo/RFID/flo-proxmark3/client/hardnested/hardnested_bitarray_core.c:429: undefined reference to `bitarray_low20_AND_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `count_bitarray_AND_dispatch':
/home/flo/RFID/flo-proxmark3/client/hardnested/hardnested_bitarray_core.c:450: undefined reference to `count_bitarray_AND_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `count_bitarray_low20_AND_dispatch':
/home/flo/RFID/flo-proxmark3/client/hardnested/hardnested_bitarray_core.c:471: undefined reference to `count_bitarray_low20_AND_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `bitarray_AND4_dispatch':
/home/flo/RFID/flo-proxmark3/client/hardnested/hardnested_bitarray_core.c:492: undefined reference to `bitarray_AND4_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `bitarray_OR_dispatch':
/home/flo/RFID/flo-proxmark3/client/hardnested/hardnested_bitarray_core.c:513: undefined reference to `bitarray_OR_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `count_bitarray_AND2_dispatch':
/home/flo/RFID/flo-proxmark3/client/hardnested/hardnested_bitarray_core.c:534: undefined reference to `count_bitarray_AND2_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `count_bitarray_AND3_dispatch':
/home/flo/RFID/flo-proxmark3/client/hardnested/hardnested_bitarray_core.c:555: undefined reference to `count_bitarray_AND3_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `count_bitarray_AND4_dispatch':
/home/flo/RFID/flo-proxmark3/client/hardnested/hardnested_bitarray_core.c:576: undefined reference to `count_bitarray_AND4_AVX512'
collect2: error: ld returned 1 exit status
Makefile:215: recipe for target 'proxmark3' failed
make[1]: *** [proxmark3] Error 1
```

OS: Kali Linux x86_64, gcc 7

```

flo@kali:~/RFID/proxmark3# gcc -dumpversion
7
flo@kali:~/RFID/proxmark3# uname -a
Linux kali 4.12.0-kali1-amd64 #1 SMP Debian 4.12.6-1kali1 (2017-08-18) x86_64 GNU/Linux
```
My CPU does not support AVX512 SIMD instructions:

```
Architecture:          x86_64
CPU op-mode(s):        32-bit, 64-bit
Byte Order:            Little Endian
CPU(s):                8
On-line CPU(s) list:   0-7
Thread(s) per core:    2
Core(s) per socket:    4
Socket(s):             1
NUMA node(s):          1
Vendor ID:             GenuineIntel
CPU family:            6
Model:                 94
Model name:            Intel(R) Core(TM) i7-6700 CPU @ 3.40GHz
Stepping:              3
CPU MHz:               799.987
CPU max MHz:           4000.0000
CPU min MHz:           800.0000
BogoMIPS:              6816.00
Virtualization:        VT-x
L1d cache:             32K
L1i cache:             32K
L2 cache:              256K
L3 cache:              8192K
NUMA node0 CPU(s):     0-7
Flags:                 fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx pdpe1gb rdtscp lm constant_tsc art arch_perfmon pebs bts rep_good nopl xtopology nonstop_tsc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx smx est tm2 ssse3 sdbg fma cx16 xtpr pdcm pcid sse4_1 sse4_2 x2apic movbe popcnt tsc_deadline_timer aes xsave avx f16c rdrand lahf_lm abm 3dnowprefetch cpuid_fault epb intel_pt tpr_shadow vnmi flexpriority ept vpid fsgsbase tsc_adjust bmi1 hle avx2 smep bmi2 erms invpcid rtm mpx rdseed adx smap clflushopt xsaveopt xsavec xgetbv1 xsaves dtherm ida arat pln pts hwp hwp_notify hwp_act_window hwp_epp
```

Tested an other PC: same issue and tested with gcc 6.4.0: fails too.

Fl0-0, 2017-08-26:
Like uzlonewolf adding `$(MULTIARCHSRCS:%.c=$(OBJDIR)/%_AVX512.o)` after `$(MULTIARCHSRCS:%.c=$(OBJDIR)/%_AVX2.o)` in client/Makefile fix the issue.

```
flo@kali:~/RFID/proxmark3# git diff
diff --git a/client/Makefile b/client/Makefile
index cadd733..92db3b2 100644
--- a/client/Makefile
+++ b/client/Makefile
@@ -182,7 +182,8 @@ MULTIARCHOBJS = $(MULTIARCHSRCS:%.c=$(OBJDIR)/%_NOSIMD.o) \
                        $(MULTIARCHSRCS:%.c=$(OBJDIR)/%_MMX.o) \
                        $(MULTIARCHSRCS:%.c=$(OBJDIR)/%_SSE2.o) \
                        $(MULTIARCHSRCS:%.c=$(OBJDIR)/%_AVX.o) \
-                       $(MULTIARCHSRCS:%.c=$(OBJDIR)/%_AVX2.o)
+                       $(MULTIARCHSRCS:%.c=$(OBJDIR)/%_AVX2.o) \
+                       $(MULTIARCHSRCS:%.c=$(OBJDIR)/%_AVX512.o)
 
 GCC_GTEQ_490 := $(shell expr `gcc -dumpversion | sed -e 's/\.\([0-9][0-9]\)/\1/g' -e 's/\.\([0-9]\)/0\1/g' -e 's/^[0-9]\{3,4\}$$/&00/'` \>= 40900)
 HARD_SWITCH_NOSIMD = -mno-mmx -mno-sse2 -mno-avx -mno-avx2
```

uzlonewolf, 2017-08-26:
gcc -dumpversion
7

```
$ cat /proc/cpuinfo 
processor	: 0
vendor_id	: AuthenticAMD
cpu family	: 16
model		: 10
model name	: AMD Phenom(tm) II X6 1100T Processor
stepping	: 0
microcode	: 0x10000dc
cpu MHz		: 800.000
cache size	: 512 KB
physical id	: 0
siblings	: 6
core id		: 0
cpu cores	: 6
apicid		: 0
initial apicid	: 0
fpu		: yes
fpu_exception	: yes
cpuid level	: 6
wp		: yes
flags		: fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush mmx fxsr sse sse2 ht syscall nx mmxext fxsr_opt pdpe1gb rdtscp lm 3dnowext 3dnow constant_tsc rep_good nopl nonstop_tsc cpuid extd_apicid aperfmperf pni monitor cx16 popcnt lahf_lm cmp_legacy svm extapic cr8_legacy abm sse4a misalignsse 3dnowprefetch osvw ibs skinit wdt cpb hw_pstate vmmcall npt lbrv svm_lock nrip_save pausefilter
bugs		: tlb_mmatch apic_c1e fxsave_leak sysret_ss_attrs null_seg amd_e400
bogomips	: 6599.89
TLB size	: 1024 4K pages
clflush size	: 64
cache_alignment	: 64
address sizes	: 48 bits physical, 48 bits virtual
power management: ts ttp tm stc 100mhzsteps hwpstate cpb

(repeat 5 more times)
```

pwpiwi, 2017-08-26:
You only get `7`? Not `7.1.1`? That*s unexpected, breaks the Makefile logic and would be the root cause. @Fl0-0 : Same result for you?

Fl0-0, 2017-08-26:
Yes i confirm, i have only `7` instead of 7.2.0 ...

uzlonewolf, 2017-08-26:
Yeah, just "7"

uzlonewolf, 2017-08-26:
https://gcc.gnu.org/ml/gcc-patches/2017-01/msg00593.html

iceman1001, 2017-08-26:
1220 million (2^30.2) keys/s ...

iceman1001, 2017-08-26:
,  there is a fix in iceman fork,   which deals with gcc version output and makefile.
https://github.com/iceman1001/proxmark3/commit/2c198f75f552febc21968be08211caed05d62ff4

Try it and see if this fixes your problem

Fl0-0, 2017-08-26:
"1220 million (2^30.2) keys/s" ? it works pretty fast :)

iceman1001, 2017-08-26:
@Fl0-0   bloody hell, thats fast.

uzlonewolf, 2017-08-26:
@iceman1001 Despite the output after the awk/sed being "70101-3)" it does appear to eval correctly.  Compile still fails though.

```
$ gcc --version | awk '/gcc/{print $NF;}' | sed -e 's/\.\([0-9][0-9]\)/\1/g' -e 's/\.\([0-9]\)/0\1/g' -e 's/^[0-9]\{3,4\}$$/&00/'
70101-3)
```
```
$ expr `gcc --version | awk '/gcc/{print $NF;}' | sed -e 's/\.\([0-9][0-9]\)/\1/g' -e 's/\.\([0-9]\)/0\1/g' -e 's/^[0-9]\{3,4\}$$/&00/'` \>= 40900
1
$ expr `gcc --version | awk '/gcc/{print $NF;}' | sed -e 's/\.\([0-9][0-9]\)/\1/g' -e 's/\.\([0-9]\)/0\1/g' -e 's/^[0-9]\{3,4\}$$/&00/'` \>= 70900
0
```
```
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `count_bitarray_AND2_dispatch':
hardnested_bitarray_core.c:(.text+0x72f): undefined reference to `count_bitarray_AND2_AVX512'
hardnested_bitarray_core.c:(.text+0x734): undefined reference to `count_bitarray_AND2_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `count_bitarray_AND3_dispatch':
hardnested_bitarray_core.c:(.text+0x7df): undefined reference to `count_bitarray_AND3_AVX512'
hardnested_bitarray_core.c:(.text+0x7e4): undefined reference to `count_bitarray_AND3_AVX512'
obj/hardnested/hardnested_bitarray_core_NOSIMD.o: In function `count_bitarray_AND4_dispatch':
hardnested_bitarray_core.c:(.text+0x88f): undefined reference to `count_bitarray_AND4_AVX512'
hardnested_bitarray_core.c:(.text+0x894): undefined reference to `count_bitarray_AND4_AVX512'
collect2: error: ld returned 1 exit status
make[1]: *** [Makefile:218: proxmark3] Error 1
make[1]: Leaving directory '/home/lonewolf/proxmark/proxmark3/client'
make: *** [Makefile:33: client/all] Error 2
$ 
```

Fl0-0, 2017-08-26:
I confirm that, make the same test just right now !

iceman1001, 2017-08-26:
dang,  at least its not a  "7" vs "70101" issue then.

Fl0-0, 2017-08-26:
awk/sed line just detect 4.9 version and after

iceman1001, 2017-08-26:
yes,  it would look for 40900,  not 7,  which was the output from @uzlonewolf  ,  don't know if it would still detect correct with 7 vs 40900 either.  But that is a different story.

uzlonewolf, 2017-08-26:
Other then the compile glitch it works rather well.  ~3 min from zero to keys
```
$ ./client/proxmark3 /dev/ttyACM0 

** (proxmark3:11885): WARNING **: Couldn't connect to accessibility bus: Failed to connect to socket /tmp/dbus-Pemk5wvoP9: Connection refused
Prox/RFID mark3 RFID instrument          
bootrom: master/v3.0.1-71-g5c814c3-dirty-suspect 2017-08-26 00:53:52
os: master/v3.0.1-71-g5c814c3-dirty-suspect 2017-08-26 00:53:55
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/07/13 at 08:44:13
          
uC: AT91SAM7S512 Rev A          
Embedded Processor: ARM7TDMI          
Nonvolatile Program Memory Size: 512K bytes. Used: 194677 bytes (37%). Free: 329611 bytes (63%).          
Second Nonvolatile Program Memory Size: None          
Internal SRAM Size: 64K bytes          
Architecture Identifier: AT91SAM7Sxx Series          
Nonvolatile Program Memory Type: Embedded Flash Memory          
proxmark3> hf mf mif
-------------------------------------------------------------------------
Executing command. Expected execution time: 25sec on average
Press button on the proxmark3 device to abort both proxmark3 and client.
-------------------------------------------------------------------------
..............#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
.#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
.#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
.#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
Parity is all zero. Most likely this card sends NACK on every failed authentication.          
Attack will take a few seconds longer because we need two consecutive successful runs.          
.#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
.#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
#db# Mifare: Can't select card          
[... 51 lines cut ...]
       6 |     112 | Apply bit flip properties                               |    398003732480 | 18min          
       7 |     224 | Apply bit flip properties                               |    117357346816 |  5min          
       8 |     336 | Apply bit flip properties                               |     72445050880 |  3min          
       9 |     448 | Apply bit flip properties                               |     66856456192 |  3min          
      10 |     558 | Apply bit flip properties                               |     56939921408 |  3min          
      11 |     670 | Apply bit flip properties                               |     56939921408 |  3min          
      12 |     781 | Apply bit flip properties                               |     53590183936 |  2min          
      12 |     890 | Apply bit flip properties                               |     47602106368 |  2min          
      13 |    1001 | Apply bit flip properties                               |     33931171840 |  2min          
      14 |    1084 | Apply bit flip properties                               |     23158603776 |   63s          
      15 |    1196 | Apply bit flip properties                               |     18168924160 |   50s          
      16 |    1308 | Apply bit flip properties                               |     18058575872 |   49s          
      23 |    1420 | Apply Sum property. Sum(a0) = 128                       |      3622854400 |   10s          
      28 |    1532 | Apply bit flip properties                               |      2976336640 |    8s          
      32 |    1564 | Apply bit flip properties                               |      9456549888 |   26s          
      37 |    1636 | Apply bit flip properties                               |      5833083392 |   16s          
      43 |    1748 | Apply bit flip properties                               |      6454953472 |   18s          
      49 |    1827 | Apply bit flip properties                               |      1459956224 |    4s          
      54 |    1937 | Apply bit flip properties                               |      1459956224 |    4s          
      58 |    1982 | Apply bit flip properties                               |      1459956224 |    4s          
      62 |    2044 | Apply bit flip properties                               |      1459956224 |    4s          
      65 |    2044 | Starting brute force...                                 |      1459956224 |    4s          
      74 |    2044 | Brute force phase completed. Key found: 5da7c3e4834f    |               0 |    0s          
proxmark3>
```

iceman1001, 2017-08-26:
no need to run hardnested against a tag which you can run darkside against.
use darkside + nested instead. Much faster.

Fl0-0, 2017-08-26:
OK finally  it builds with iceman1001/proxmark3@2c198f7

```
flo@kali:~/RFID/proxmark3# git diff
diff --git a/client/Makefile b/client/Makefile
index cadd733..589d6d4 100644
--- a/client/Makefile
+++ b/client/Makefile
@@ -184,7 +184,8 @@ MULTIARCHOBJS = $(MULTIARCHSRCS:%.c=$(OBJDIR)/%_NOSIMD.o) \
                        $(MULTIARCHSRCS:%.c=$(OBJDIR)/%_AVX.o) \
                        $(MULTIARCHSRCS:%.c=$(OBJDIR)/%_AVX2.o)
 
-GCC_GTEQ_490 := $(shell expr `gcc -dumpversion | sed -e 's/\.\([0-9][0-9]\)/\1/g' -e 's/\.\([0-9]\)/0\1/g' -e 's/^[0-9]\{3,4\}$$/&00/'` \>= 40900)
+GCC_GTEQ_490 := $(shell expr `gcc --version | awk '/gcc/{print $$NF;}' | sed -e 's/\.\([0-9][0-9]\)/\1/g' -e 's/\.\([0-9]\)/0\1/g' -e 's/^[0-9]\{3,4\}$$/&00/'` \>= 40900)
+
 HARD_SWITCH_NOSIMD = -mno-mmx -mno-sse2 -mno-avx -mno-avx2
 HARD_SWITCH_MMX = -mmmx -mno-sse2 -mno-avx -mno-avx2
 HARD_SWITCH_SSE2 = -mmmx -msse2 -mno-avx -mno-avx2
```

Fl0-0, 2017-08-26:
The problem was a 40900 vs 7, @iceman1001 was right !

uzlonewolf, 2017-08-26:
iceman1001/proxmark3@2c198f7 does not work for me, but iceman1001/proxmark3@dc202ee (the next one) does.

Fl0-0, 2017-08-26:
The latest code from @iceman1001 fork works, you are right it is iceman1001/proxmark3@dc202ee.

`GCC_GTEQ_490 := $(shell expr `gcc --version | awk '/gcc/{print $$NF;}' | sed -e 's/\.\([0-9][0-9]\)/\1/g' -e 's/\.\([0-9]\)/0\1/g' -e 's/^[0-9]\{3,4\}$$/&00/'` \>= 40900)`

iceman1001, 2017-08-26:
yeah,  there were two commits,  I just assumed you would have taken the latest source anyway when testing...   my bad,   but good that it solves your issues :)

uzlonewolf, 2017-08-27:
On further inspection, the updated expression is bad.  On an old CentOS 5 machine it results in:
```
$ gcc --version | awk '/gcc/{print $$NF;}'
20080704

$ gcc --version
gcc (GCC) 4.1.2 20080704 (Red Hat 4.1.2-46)

$ gcc --version | awk '/gcc/{print $$NF;}' 
20080704
```

In that `awk` line NF = 7, so $NF = $7 = "4.1.2-46)" and $$NF = $$7 = $4 = "20080704".  In other words the `$$NF` should just be `$NF`, assuming the "4.1.2-46)" is what we want.  It happens to work for 7.x as $$7 = $7 in that case.

iceman1001, 2017-08-27:
you need to conside that Makefile variable expansion happens.  (https://github.com/iceman1001/proxmark3/issues/123)

So you need to test that expression output from Makefile perspective.  
I added https://github.com/iceman1001/proxmark3/blob/master/client/Makefile#L265  this in able to look at Makefile variables.   It will your testing.

closed 2017-08-27 by iceman1001