Proxmark/proxmark3 #389: FPGA_MAJOR_MODE_HF_SNOOP

iceman1001, 2017-08-29:
I see two references to this define and FPGA mode,    FPGA_MAJOR_MODE_HF_SNOOP.
one in hfsnoop.c and one in fpgaloader.h    (https://github.com/Proxmark/proxmark3/search?utf8=%E2%9C%93&q=FPGA_MAJOR_MODE_HF_SNOOP&type=)

But I can't seem to find any reference of it in the FPGA code.  Why is that?

iceman1001, 2017-08-29:
This thread discusses the differences from Enio's hfsnoop and iZsh's split of Fpga images.

http://www.proxmark.org/forum/viewtopic.php?id=2101

And in  fpga_hf.v  (https://github.com/Proxmark/proxmark3/blob/fe087813f65b2a08ee814ad78d1164ae7eac6db0/fpga/fpga_hf.v) 
the most majormode seems to be wired but no snoop is wired.

iceman1001, 2017-08-29:
The original Enio's repo, with his commits,  
https://github.com/EnioArda/proxmark3/commit/1bae666bcfaeab3b86c5a64bb150559862e1089d

marshmellow42, 2017-08-29:
the fpga_hf.v line 128 contains the snoop/sniffer major mode module definition which refers to the hi_sniffer.v

iceman1001, 2017-08-30:
Aha, thanks.  So modules is all in parallell.

closed 2017-08-30 by iceman1001