Proxmark/proxmark3 #390: `lf em 410xsim` does not work without some other LF commands first

doegox, 2017-09-11:
Hi,

When running `lf em 410xsim` immediately after power-up, the command acts as normal but a reader doesn't see the emulated tag.
Apparently there is some missing LF initialization.
If running e.g. `lf search` or `lf em 410xread 1` first, then the `lf em 410xsim` works properly.

iceman1001, 2017-09-11:
that should work,  there was a fix some issue with that before. 
However,  if the FPGA was in HF mode before,   and you executed em sim,  the command FpgaDownloadAndGo(FPGA_BITSTREAM_LF)  will destroy your simulation data needed.
This is commented in the source code. 

If device was in LF mode,  the sim works on the first run.
in iceman fork there is a solution,  still pm3 offical might just need to do a single call to CmdSetDivisor("89") before CmdLFSim("0") in  cmdlfem4x.c

pwpiwi, 2017-09-11:
Fix was PR #369. Which Version do you use?

iceman1001, 2017-09-11:
...and I beg the diff.  The PR which piwi refers to has this problem I just mentioned.
The call to FpgaDownloadAndGo(FPGA_BITSTREAM_LF)
as seen here https://github.com/Proxmark/proxmark3/blob/master/armsrc/lfops.c#L390  is the cause to it.

marshmellow42, 2017-09-11:
I'm not certain #369 fixed this call.  It fixed others but the em sim cmds are build differently.

iceman1001, 2017-09-11:
luckily its quite easy to reproduce.  

1. make sure device is in HF mode.
2. lf em4x sim...
3. and the sim should fail..

vs.
1. lf search
2. lf em4x sim
3.. and the sim is ok (which is what @doegox points out)

doegox, 2017-09-11:
FTR I was using official/v3.0.1-75-g1dae9811-suspect

iceman1001, 2017-09-11:
Would you mind just verify it with

hf 14a read
lf em4x sim

doegox, 2017-09-11:
indeed @iceman1001  lf em sim fails when the hf command is issued before.

In one single session:

```
lf search
lf em 410xsim => ok
hf 14a read
lf em 410xsim => fail
```

iceman1001, 2017-09-11:
perfect. 
You can either patch in on the client side, in lfem4x.c ,  with a CmdSetDivisor("95") .
Or on deviceside..

iceman fork solves it on the deviceside.

marshmellow42, 2017-09-11:
see #391 for a patch.

marshmellow42, 2017-09-11:
this was on my list of things to do anyway, as i fixed the other sim commands but hadn't gotten to the raw lf sim command.  this just pushed it along.. ;)

doegox, 2017-09-11:
it works fine now, thanks a lot @marshmellow42 and @iceman1001 !

closed 2017-09-11 by doegox