Proxmark/proxmark3 #394: add work with gen2 "magic chineese" cards

merlokk, 2017-09-22:
it needs to think how to identify it in mfCIdentify()

merlokk, 2017-09-23:
it seems it here (writed by @iceman1001 )
https://github.com/iceman1001/proxmark3/blob/master/armsrc/mifarecmd.c#L1495

But it was commented...

iceman1001, 2017-09-25:
A Mifare Classic write command will need an authentication before.  Which is why the code is commented out,  since we don't have a key available.  Hence the test as it is is bad.   For gen2 detection other methods will be needed.

merlokk, 2017-09-25:
@iceman1001 do you have gen2 cards? how they write sector0?
they authenticates with "backdoor" keys in normal mode? or just send write command wo authentication?

iceman1001, 2017-09-25:
The idea with gen2 is that they do not have backdoor commands...
if they use backdoor commands, it is a gen1...

merlokk, 2017-09-25:
Tested. This not works. Or I have not a gen2 tag

```
	// gen2
	FpgaWriteConfWord(FPGA_MAJOR_MODE_OFF);
	SpinDelay(50);
	iso14443a_setup(FPGA_HF_ISO14443A_READER_LISTEN);
	
	if(!iso14443a_select_card(NULL, NULL, NULL, true, 0)) {
		if (MF_DBGLEVEL >= 1)	Dbprintf("Can't select card");
		};

	uint16_t len = mifare_sendcmd_short(NULL, 1, 0x30, 0, receivedAnswer, receivedAnswerPar, NULL);
	if ((len != 1) || (receivedAnswer[0] != 0x0A)) {   
		Dbprintf("gen2 test fail");
	};
		
	if(mifare_classic_halt(NULL, 0)) {
		if (MF_DBGLEVEL > 2)	Dbprintf("Halt error");
	};
	
```

merlokk, 2017-09-25:
Is it a gen2 tag? it's strange, but it answers to ATS tag
```
proxmark3> hf 14a reader
 UID : 00 00 00 00
ATQA : 00 02
 SAK : 18 [2]
TYPE : NXP MIFARE Classic 4k | Plus 4k SL1
SAK incorrectly claims that card doesn't support RATS
 ATS : 09 78 00 91 02 da bc 19 10 f0 05
       -  TL : length is 9 bytes
       -  T0 : TA1 is present, TB1 is present, TC1 is present, FSCI is 8 (FSC = 256)
       - TA1 : different divisors are supported, DR: [], DS: []
       - TB1 : SFGI = 1 (SFGT = 8192/fc), FWI = 9 (FWT = 2097152/fc)
       - TC1 : NAD is NOT supported, CID is supported
       -  HB : da bc 19 10
#db# gen2 test fail
Answers to chinese magic backdoor commands: NO
```

iceman1001, 2017-09-25:
I'm not sure how you think but to clarify,  you will need a normal write command (with correct key) to block0 on S0,  to see if the card is a gen2.  None of which you are doing.

merlokk, 2017-09-26:
I have gen2 card. and i have not seen a backdoor on it(  so it really needs to know a key. so identification impossible.

but.
I see that it have some iso14443-4 part. i look at it. but maybe here is a final point(

merlokk, 2017-09-26:
65b9f4f8966488db9ffb55538cef62e54d7c35b6

nidac1957, 2018-10-16:
Please inform me, if is possible, how can I send the backdoor commands, 7 bits commands(50 00 57 CD; 40; 43 )for mifare 1k chinese card, to MFRC522: by I2C interface or by SPI interface?

merlokk, 2018-10-16:
https://github.com/nfc-tools/libnfc

nidac1957, 2018-10-16:
Thanks merlokk, but I don't understand: by uart, i2c or spi interface? Thanks again.

nidac1957, 2018-10-16:
SPI interface works only 8 bits transfer.

merlokk, 2018-10-16:
here bugs/improvements for proxmark.
if you want you can ask here.
http://www.proxmark.org/forum/viewforum.php?id=21
p.s. you need to read the documentation for MFRC522

nidac1957, 2018-10-16:
I'm on line, but I can't post on proxmark.org/forum/.....!!!