Proxmark/proxmark3 #397: Witch iso proxmark support?

userDevop123, 2017-09-24:
Witch iso proxmark support? 
And how can i see on the code witch iso it recognize?

merlokk, 2017-09-24:
iso14443

userDevop123, 2017-09-24:
@merlokk 
14443 a or b or both?

merlokk, 2017-09-24:
both

userDevop123, 2017-09-24:
@merlokk 
can you please link me to the code the proxmark recognize type b and where can i emulate type b?
thank you!

merlokk, 2017-09-25:
https://github.com/Proxmark/proxmark3/blob/master/client/cmdhf14b.c#L87

and near.

as for simulation it might need some programming...

userDevop123, 2017-09-25:
@merlokk 
thank about your reply.
but i don't think you right.
there is no implementation on fpga to iso14443b, there is only to 14443a(you can look on [fpga folder](https://github.com/Proxmark/proxmark3/tree/master/fpga).
Additionally on [14443b armsrc](https://github.com/Proxmark/proxmark3/blob/master/armsrc/iso14443b.c) i saw on there is no [modulation to 14443b](https://github.com/Proxmark/proxmark3/blob/master/armsrc/iso14443b.c#L297)

marshmellow42, 2017-09-25:
The 14b works currently so you are missing something.  It does share a lot of 15693 fpga modes as they are similar at that level.

userDevop123, 2017-09-25:
@marshmellow42 are you sure?
can you show me when iso14443b load the fpga of 15693?
i not see it [here](https://github.com/Proxmark/proxmark3/blob/master/armsrc/iso14443b.c)
i see that wrote there `FPGA_HF_SIMULATOR_NO_MODULATION` that seem like there is no implement al all

can you show it on the code?
by the way 15693 mode is 10% or 100% 14443b is only 10% ,

merlokk, 2017-09-25:
https://github.com/Proxmark/proxmark3/blob/master/armsrc/iso14443b.c#L833
here transmit to the card

userDevop123, 2017-09-25:
@merlokk
What i see on your link?
I not see the implement to 14443b on fpga
I dont see where the arm of 14b call to his fpga

merlokk, 2017-09-25:
#define FPGA_MAJOR_MODE_HF_READER_TX				(0<<5)
goes here
https://github.com/Proxmark/proxmark3/blob/fe087813f65b2a08ee814ad78d1164ae7eac6db0/fpga/fpga_hf.v#L88

userDevop123, 2017-09-25:
@merlokk i so thank you about you help
can you please exmplain me more about this "transmit" ? this happend only on 14443b?
on 14b according to [doc.](http://www.atmel.com/images/doc2056.pdf) the modulation is 10% 
i don't see where arm load fpga that define 10% modulation.

merlokk, 2017-09-26:
ToSendStuffBit
https://github.com/Proxmark/proxmark3/blob/master/armsrc/appmain.c#L64

To check if it works - you just need to test 1 "b" card. Usually its a bank card. 
Just check - if it works - look how ...

merlokk, 2017-09-26:
[Here cool document](https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=3&ved=0ahUKEwiHhYGbhsPWAhUJG5oKHfXAANEQFgg1MAI&url=http%3A%2F%2Fe2e.ti.com%2Fcfs-file%2F__key%2Fcommunityserver-discussions-components-files%2F667%2F2072.ISO_5F00_NFC-Standards-and-Specifications-Overview_5F00_2014.pdf&usg=AFQjCNE_V3ZiPehs2zL4ZKDmvWa5wED8Yw)

As I said you need to write code to proxmark to work it as you need. In this area not a big support.

userDevop123, 2017-09-26:
where can i put a breakpoint on c code to see if the proxmark recognize my 14b tag?
i want to put the tag on the proxmark and see if t recognize
thank!

pwpiwi, 2017-10-01:
You don't need a breakpoint. Just run `hf 14b info` to check if a tag is recognized. Please use the forum to ask basik questions. And read some posts there before asking.

closed 2017-10-01 by pwpiwi