Proxmark/proxmark3 #399: How to compie and burn fpga+arm?

jordanmv1c, 2017-09-25:
I not understand from the compiling file.
I want to burn and run the proxmark on windows,so (i think) i need to compile all the project and burn to proxmark the fpga and arm.

I saw rather from download every sofward i can download proxspace that include all from  http://code.google.com/p/proxmark3/downloads/list

I download it.
What should i do now to burn all and i can read uid/tag on proxmark via windows

Thanks

merlokk, 2017-09-26:
proxspace here https://github.com/Gator96100/ProxSpace
compile instructions - into this repository

pwpiwi, 2017-10-01:
Google code repository has been abandoned years ago. Please use the forum to ask basic questions. Here you would report bugs and provide fixes.

closed 2017-10-01 by pwpiwi