Proxmark/proxmark3 #404: hf mf hardnested error

merlokk, 2017-09-27:
hf mf hardnested causes runtime error while compiled on windows10
proxmark3 just ended process by error.

merlokk, 2017-09-27:
Unhandled exception [13056] and try to execute debugger

merlokk, 2017-09-27:
hangs here:

brute_force_per_second = brute_force_benchmark();

merlokk, 2017-09-27:
in float brute_force_benchmark()
here:
	brute_force_bs(&bf_rate, test_candidates, 0, 0, maximum_states, NULL, 0);

merlokk, 2017-09-27:
in
bool brute_force_bs(...)

here:
	for(uint32_t i = 0; i < NUM_BRUTE_FORCE_THREADS; i++){
		thread_args[i].thread_ID = i;
		thread_args[i].silent = silent;
		thread_args[i].cuid = cuid;
		thread_args[i].num_acquired_nonces = num_acquired_nonces;
		thread_args[i].maximum_states = maximum_states;
		thread_args[i].nonces = nonces;
		thread_args[i].best_first_bytes = best_first_bytes;
		pthread_create(&threads[i], NULL, crack_states_thread, (void*)&thread_args[i]);
	}

merlokk, 2017-09-27:
as i see maybe in thread here
static void* crack_states_thread(void* x){

merlokk, 2017-09-27:
log like that
proxmark3>  hf mf hardnested 0 A A0A1A2A3A4A5 4 A FFFFFFFFFFFF
--target block no:  4, target key type:A, known target key: 0xffffffffffff, file action: none, Slow: No, Tests: 0
TThread 1 starts working on bucket 1
Thread 2 starts working on bucket 2
Thread 0 starts working on bucket 0
hread 3 starts working on bucket 3

merlokk, 2017-09-28:
breaks here            
 const uint64_t key = crack_states_bitsliced(thread_arg->cuid, thread_arg->best_first_bytes, bucket, &keys_found, &num_keys_tested, nonces_to_bruteforce, bf_test_nonce_2nd_byte, thread_arg->nonces);

marshmellow42, 2017-09-28:
I compile and run on Windows 10 and have no issues with hardnested.  Check that you have all the files required in the correct locations.  Check file permissions (admin elevated access).  Tell us your exact cpu

merlokk, 2017-09-28:
CPU-Z version			1.81.0.x64
Processors
-------------------------------------------------------------------------
Number of sockets		1
Number of threads		4
APICs
-------------------------------------------------------------------------
Socket 0	
	-- Core 0	
		-- Thread 0	0
	-- Core 1	
		-- Thread 0	1
	-- Core 3	
		-- Thread 0	3
	-- Core 2	
		-- Thread 0	2
Timers
-------------------------------------------------------------------------
	ACPI timer		3.580 MHz
	Perf timer		2.943 MHz
	Sys timer		1.000 KHz
Processors Information
-------------------------------------------------------------------------
Socket 1			ID = 0
	Number of cores		4 (max 4)
	Number of threads	4 (max 4)
	Name			AMD Athlon II X4 640
	Codename		Propus
	Specification		AMD Athlon(tm) II X4 640 Processor
	Package 		Socket AM3 (938)
	CPUID			F.5.3
	Extended CPUID		10.5
	Brand ID		9
	Core Stepping		BL-C3
	Technology		45 nm
	TDP Limit		95.0 Watts
	Core Speed		3013.3 MHz
	Multiplier x Bus Speed	15.0 x 200.9 MHz
	HT Link speed		2008.8 MHz
	Stock frequency		3000 MHz
	Instructions sets	MMX (+), 3DNow! (+), SSE, SSE2, SSE3, SSE4A, x86-64, AMD-V
	L1 Data cache		4 x 64 KBytes, 2-way set associative, 64-byte line size
	L1 Instruction cache	4 x 64 KBytes, 2-way set associative, 64-byte line size
	L2 cache		4 x 512 KBytes, 16-way set associative, 64-byte line size
	Max CPUID level		00000005h
	Max CPUID ext. level	8000001Bh
	Cache descriptor	Level 1, I, 64 KB, 1 thread(s)
	Cache descriptor	Level 1, D, 64 KB, 1 thread(s)
	Cache descriptor	Level 2, U, 512 KB, 1 thread(s)
	FID/VID Control		yes
	FID range		4.0x - 15.0x
	Max VID			1.400 V
	# of P-States		4
	P-State			FID 0xE - VID 0x0C - IDD 19 (15.00x - 1.400 V)
	P-State			FID 0x7 - VID 0x14 - IDD 14 (11.50x - 1.300 V)
	P-State			FID 0x2 - VID 0x1C - IDD 13 (9.00x - 1.200 V)
	P-State			FID 0x100 - VID 0x28 - IDD 8 (4.00x - 1.050 V)

	Package Type		0x1
	Model			40
	String 1		0xA
	String 2		0x6
	Page			0x0
	CmpCap			4
	ApicIdCoreSize		4
	TDC Limit		76 Amps
	Boosted P-States	0
	Max non-turbo ratio	15.00x
	Max turbo ratio		15.00x
	Max CPU COF		30
	Core Performance Boost	no
	P-State			0, FID 0xE - VID 0x0C (15.00x - 1.400 V)
	P-State			1, FID 0x7 - VID 0x14 (11.50x - 1.300 V)
	P-State			2, FID 0x2 - VID 0x1C (9.00x - 1.200 V)
	P-State			3, FID 0x100 - VID 0x28 (4.00x - 1.050 V)
	Attached device		PCI device at bus 0, device 24, function 0
	Attached device		PCI device at bus 0, device 24, function 1
	Attached device		PCI device at bus 0, device 24, function 2
	Attached device		PCI device at bus 0, device 24, function 3
	Attached device		PCI device at bus 0, device 24, function 4
	PERFEVT			122652032.0 MHz

	Temperature 0		36 degC (95 degF) (Core #0)
	Temperature 1		35 degC (95 degF) (Core #1)
	Temperature 2		35 degC (95 degF) (Core #3)
	Temperature 3		35 degC (95 degF) (Core #2)
	Power 00		106.40 W (Package)
	Clock Speed 0		3013.26 MHz (Core #0)
	Clock Speed 1		3013.26 MHz (Core #1)
	Clock Speed 2		3013.26 MHz (Core #3)
	Clock Speed 3		3013.26 MHz (Core #2)
	Core 0 max ratio	15.0 (effective 15.0)
	Core 1 max ratio	15.0 (effective 15.0)
	Core 2 max ratio	15.0 (effective 15.0)
	Core 3 max ratio	15.0 (effective 15.0)

merlokk, 2017-09-28:
file permissions is ok, files from repository.

pwpiwi, 2017-09-28:
Any compilation or link warnings? Does it print anything before exiting? Which compiler do you use?

No problems here with Windows 10.

merlokk, 2017-09-28:
There is no computer wahrnings. And I use proxspace from Gator96100

merlokk, 2017-09-29:
Maybe it not works on AMD Athlon II X4 640 ?

marshmellow42, 2017-09-29:
How recently did you get Gators proxspace?  He had an issue in his environment a while back that caused an issue with hardnested.  
Might need to re install it?

marshmellow42, 2017-09-29:
See http://www.proxmark.org/forum/viewtopic.php?id=5034

pwpiwi, 2017-09-29:
@merlokk: do you use precompiled binaries? Or did you compile yourself?

merlokk, 2017-09-29:
I compile myself with latest Gator's proxspace from github.  Its like my problem. Ill look.

merlokk, 2017-09-29:
There is no rights problem, UAC disabled and so on. Compiling ok, no problem with perl(

merlokk, 2017-09-29:
tried to put this to run bat
powershell -Command "Start-Process 'msys\msys.bat' -Verb runAs

no result.

merlokk, 2017-09-29:
how do you start it?  
cd client
proxmark3 com1 
?

pwpiwi, 2017-10-01:
yes (if your proxmark is on com1), Or `client/proxmark3 com1`

iceman1001, 2017-10-01:
very rarely is a device connected to com1 on a windows-based computer.  Normally com3 or upwards

merlokk, 2017-10-02:
mine com6. usually, i run to from client folder, i have tried to run id your way, but not succeed.

pwpiwi, 2017-10-02:
Does it print anything before exiting?

merlokk, 2017-10-02:
no, windows just kills the process

pwpiwi, 2017-10-02:
Before you can type any command?

merlokk, 2017-10-02:
before it works. no its not serial. its like the process need somewhat and dont found.

merlokk, 2017-10-20:
on appveyor works( strange....

iceman1001, 2017-12-25:
@merlokk   whats the status with this one?   Does hardnested work for you or not?

merlokk, 2017-12-25:
from my pc it not works(  maybe lets close it and i will open it when i have time to check

iceman1001, 2017-12-25:
well,  a non-working hardnested command is a showstopper for a new release.

merlokk, 2017-12-25:
ill look in it tomorrow

merlokk, 2017-12-27:
as i see it hangs here
https://github.com/Proxmark/proxmark3/blob/master/client/hardnested/hardnested_bf_core.c#L292

`bitsliced_even_feedback[bitsliced_blocks] = `
here `bitsliced_blocks == 0`

but it looks that memory is assigned....

merlokk, 2017-12-27:
MAX_BITSLICES == 128

merlokk, 2017-12-27:
OK, it works but with decrease MAX_BITSLICES for __SSE2__
```
#elif defined(__SSE2__)
#define MAX_BITSLICES 64
```

merlokk, 2017-12-27:
```
       0 |       0 | Start using 4 threads and SSE2 SIMD core                |                 |
       0 |       0 | Brute force benchmark: 24 million (2^24,5) keys/s       | 140737488355328 |   67d
       0 |       0 | Starting Test #1 ...                                    | 140737488355328 |   67d
       2 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |   67d

```

merlokk, 2017-12-27:
What we should do?
check why it happens? it looks it not compiled with SSE2 or SSE2 cant work with 128 bytes vectors...

or just change this value and thats all?

pwpiwi, 2017-12-27:
SSE2 can work with 128bit vectors. What is your CPU type?

merlokk, 2017-12-27:
AMD Athlon II X4 640

marshmellow42, 2017-12-27:
Sounds like a cpu bug.

merlokk, 2017-12-27:
maybe gcc have `__SSE2__` define bunt dont generate SSE2 code in mentioned place

pwpiwi, 2017-12-27:
Supports SSE2. What is the command used to compile the *_SSE2.o files?

merlokk, 2017-12-27:
```
gcc -MT obj/hardnested/hardnested_bf_core_SSE2.o -MMD -MP -MF obj/hardnested/hardnested_bf_core.Td  -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../common/polarssl -I../zlib -I../uart -I/opt/local/include -I../liblua -Wall -g -O3 -DHAVE_GUI -mmmx -msse2 -mno-avx -mno-avx2 -mno-avx512f -c -o obj/hardnested/hardnested_bf_core_SSE2.o hardnested/hardnested_bf_core.c
gcc -MT obj/hardnested/hardnested_bitarray_core_SSE2.o -MMD -MP -MF obj/hardnested/hardnested_bitarray_core.Td  -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../common/polarssl -I../zlib -I../uart -I/opt/local/include -I../liblua -Wall -g -O3 -DHAVE_GUI -mmmx -msse2 -mno-avx -mno-avx2 -mno-avx512f -c -o obj/hardnested/hardnested_bitarray_core_SSE2.o hardnested/hardnested_bitarray_core.c
```

merlokk, 2017-12-28:
ok, found it.
but.....
```
#define MAX_BITSLICES 64
#define VECTOR_SIZE (MAX_BITSLICES/8)
result:
      0 |       0 | Brute force benchmark: 25 million (2^24,6) keys/s       | 140737488355328 |   65d
```

and

```
#define MAX_BITSLICES 128
#define VECTOR_SIZE (MAX_BITSLICES/4)
result:
 0 |       0 | Brute force benchmark: 6 million (2^22,4) keys/s        | 140737488355328 |  290d
```

strange.....

but there is a bug....

```
#define VECTOR_SIZE (MAX_BITSLICES/8)
typedef unsigned int __attribute__((aligned(VECTOR_SIZE))) __attribute__((vector_size(VECTOR_SIZE))) bitslice_value_t;

```
length of int == 4
MAX_BITSLICES in bytes so....
correct one

```
#define MAX_BITSLICES 128
#define VECTOR_SIZE (MAX_BITSLICES/4)
```

pwpiwi, 2017-12-28:
`MAX_BITSLICES` is in Bits. `VECTOR_SIZE` is in Bytes.

pwpiwi, 2017-12-28:
Travis xcode8.3 and later uses SSE2. No issues.

merlokk, 2017-12-28:
sounds like compiler not compiled SSE2 on my platform or bug in CPU

merlokk, 2017-12-29:
It really launches SSE2 code and really goes to XOR's.

i looked into ASM. it looks good.

but it stops and show try to read from FFFFF... 
[image]

it looks like its CPU bug(

merlokk, 2017-12-29:
[image]

as i see there is no FFFFs. it lokks like cpu bug(

pwpiwi, 2017-12-29:
... or a compiler bug. Did you try another version?

merlokk, 2017-12-29:
i have tried several PC and looks like only my have this error

merlokk, 2017-12-29:
Ill try another version of gcc but maybe after NY

Gator96100, 2018-01-01:
I finally did update my Proxspace environment to gcc 5.3 in order to have support for AVX-512 optimizations. This might also solve your problem. [https://github.com/Gator96100/ProxSpace/tree/NewGCC](url)

merlokk, 2018-01-02:
Thanks! Ill try

merlokk, 2018-01-02:
it not helps (

and it have strange double prompts for me...  but it offtopic)
```
pm3 ~/client$ gcc --version
gcc.exe (i686-posix-dwarf-rev0, Built by MinGW-W64 project) 5.3.0
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

pm3 ~/client$                                                                                                                     
pm3 ~/client$
```

merlokk, 2018-01-02:
OK,
this works
`bitslice_value_t t = lstate_p[(47- 0)/2].value;`
this works
`bitslice_value_t t = lstate_p[(47- 0)/2].value ^ 0xFF;`
this works
`bitslice_value_t t = lstate_p[(47- 10)/2].value ^ 0xff;`
this works (maybe because of compiler optimization)
`bitslice_value_t t = lstate_p[(47- 0)/2].value ^ lstate_p[(47- 0)/2].value;`
this not(
`bitslice_value_t t = lstate_p[(47- 0)/2].value ^ lstate_p[(47- 10)/2].value;`

merlokk, 2018-01-02:
OK. im pass(
[image]

there are 2 mov instructions. program hangs on 2nd.
as i see all is OK with memory alignment and access..
this code successfully executes on another CPU and travis
so it looks that this bug in very low level. maybe in compiling SSE2 with gcc. 
I cant fix it.

Maybe it needs to add key to `hardnested` command not to use any of optimization. and thats all(

iceman1001, 2018-01-02:
the first one moves from eax register,   the second from ebp-0x07e8 memory location.
Where does those registers point to?  is the memoryallocation too small?

merlokk, 2018-01-02:
ebp-0x07e8 - memory is ok. there are 2 memory windows at top - with address of ptr, on the bottom - with memory

eax also contains address and it ok(

merlokk, 2018-01-03:
iceman fork:
[image]
the same (

[11 more replies]

commit 4ed45d44 (2018-01-06): fix #404
closed 2018-01-06 by iceman1001