Proxmark/proxmark3 #406: Proxmark3 don't read jcop card in emulation mifare classic.

maxben14, 2017-09-30:
My proxmark3 easy don't read jcop card, in acr122 i do fix patch add this instruction from libnfc:
nfc_device_set_property_bool(pnd, NP_AUTO_ISO14443_4,false);  
How and where in frimware from github iceman can add this instruction ?
I want add this instruction in firmware and do compile firmware.elf.

merlokk, 2017-09-30:
What Type of jcop chip do you have?
The only difference on the old ones - jcop dont like pauses between commands, but proxmark dont do pauses

maxben14, 2017-09-30:
proxmark3> hf 14a reader
#db# Emulator stopped. Tracing: 0  trace length: 35902
iso14443a card select failed
proxmark3> hf 14a reader
 UID : 67 67 b9 d1
ATQA : 00 04
 SAK : 28 [1]
TYPE : JCOP31 or JCOP41 v2.3.1
 ATS : 10 78 80 70 02 00 31 c0 64 08 04 61 00 00 90 00 47 0d
       -  TL : length is 16 bytes
       -  T0 : TA1 is present, TB1 is present, TC1 is present, FSCI is 8 (FSC = 256)
       - TA1 : different divisors are NOT supported, DR: [], DS: []
       - TB1 : SFGI = 0 (SFGT = (not needed) 0/fc), FWI = 7 (FWT = 524288/fc)
       - TC1 : NAD is NOT supported, CID is supported
       -  HB : 00 31 c0 64 08 04 61 00 00 90 00
Answers to chinese magic backdoor commands: NO

maxben14, 2017-09-30:
I think problem in NP_AUTO_ISO14443_4 
From libnfc:  This option can be used to enable or disable the auto-switching mode to ISO14443-4 is device is compliant. In initiator mode, it means that NFC chip will send RATS automatically when select and it will automatically poll for ISO14443-4 card when ISO14443A is requested. In target mode, with a NFC chip compliant (ie. PN532), the chip will emulate a 14443-4 PICC using hardware capability

How off send RATS ? If in proxmark off RATS then he will read jcop very good. Please write me how off RATS.

merlokk, 2017-09-30:
Proxmark dont switch to iso14443-4 in the reading procedure.
Can you Issue commands and pu result here
hf mf rdbl 0 a xxxxxxxxxxxx
Hf List 14a
?

maxben14, 2017-09-30:
proxmark3> hf mf rdbl 0 a a0a1a2a3a4a5
--block no:0, key type:A, key:a0 a1 a2 a3 a4 a5
#db# Auth error
#db# READ BLOCK FINISHED
isOk:00
proxmark3> hf list 14a
Recorded Activity (TraceLen = 132 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |        992 | Rdr | 52                                                              |     | WUPA
       2228 |       4596 | Tag | 04  00                                                          |     |
       7040 |       9504 | Rdr | 93  20                                                          |     | ANTICOLL       
      10692 |      16580 | Tag | 67  67  b9  d1  68                                              |     |
      18816 |      29280 | Rdr | 93  70  67  67  b9  d1  68  c7  39                              |  ok | SELECT_UID     
      30516 |      34036 | Tag | 28  b4  fc                                                      |     |
      35584 |      40352 | Rdr | e0  80  31  73                                                  |  ok | RATS
      42820 |      63684 | Tag | 10  78  80  70  02  00  31  c0  64  08  04  61  00  00  90  00  |     |
            |            |     | 47  0d                                                          |  ok |
      65280 |      69984 | Rdr | 60  00  f5  7b                                                  |  ok | AUTH-A(0)      
proxmark3>

merlokk, 2017-09-30:
It Issues rats. On monday ill look how to disable it.

maxben14, 2017-09-30:
I write comand from libnfc about how to disable rats. nfc_device_set_property_bool(pnd, NP_AUTO_ISO14443_4,false);
Can you say how this comand in proxmark frimware write.

merlokk, 2017-09-30:
It needs to look how to disable it into anticollision procedure. You can look into read procedure. I cant look - im from mobile phone

maxben14, 2017-09-30:
I correctly understood that you are proposing a command from the library libnfc to execute in the libnfc utility, and proxy to intercept the exchange and put it here?

merlokk, 2017-09-30:
No, I proposed to look into proxmark source and disable sending RATS command

maxben14, 2017-09-30:
I found in firmware rats in iso1443a.c https://github.com/Proxmark/proxmark3/blob/master/armsrc/iso14443a.c
in function int iso14443a_select_card(byte_t *uid_ptr, iso14a_card_select_t *p_hi14a_card, uint32_t *cuid_ptr, bool anticollision, uint8_t num_cascades)
I think command ReaderTransmit(rats, sizeof(rats), NULL); replace on return 1;
Do you agree ?

pwpiwi, 2017-09-30:
I  had observed the same issue and I am working on a fix.

maxben14, 2017-09-30:
pwpiwi, can send me patch fix or write how patch.

maxben14, 2017-09-30:
Also how to disable the white LED on proxmark. I want such a device to give the conductor in the bus, me need to turn off the white indicator LED, which is lit when the device is connected about usb.

merlokk, 2017-09-30:
https://github.com/Proxmark/proxmark3/blob/master/armsrc/iso14443a.c#L1873
here

merlokk, 2017-09-30:
and about led. )
mifare emulation on proxmark works too slow and almost all systems protected against that (it made special for that). 
It shows a mechanism..., but not optimized to work as emulator in real systems.

If you want...  optimized system you need to made by yourself)

merlokk, 2017-09-30:
p.s.) @pwpiwi have you allready made a fix? i can write it on monday) if you work on it - i just wait)
it needs to add default parameter to iso14443a_select_card and use it in mifare read/write block.

pwpiwi, 2017-10-01:
Working on it. Its not only read/write block.

iceman1001, 2017-10-01:
wasn't there someone who made "rats" optional and changed all calls to iso144_select_card ?
What happend with that PR or issue?   I belive the "hf 14a reader" might have too much stuff in it,  maybe a divide it into a "hf 14a info",  where all extra checks is done.   Then the rats call on armside could be removed. hm, is rats optional or mandatory?

maxben14, 2017-10-01:
@iceman1001 , i think that rats optional and can remove this comand.
@merlokk , i try proxmark in mode emulation 1k read with acr122 with good rezult and write successfully, but my smartphone with nfc bad reads it. I try it through the MCT application, UID see it, but MCT get lost tag on authorization in sector.

I want proxmark in mode sniffer using the command hf 14a shoop and put it in the cover with the card and give the conductor in the bus to intercept the key. But on it a white LED is on, how can I turn it off? The version easy proxmark has a red, yellow, green, blue LED left and right white. I correctly understood that white is a power supply of USB and can not be change from the firmware?

pwpiwi, 2017-10-01:
Keep to the subject please.

merlokk, 2017-10-01:
RATS is not optional for many cases. But RATS change protocol mode from iso14443-3 to iso14443-4.
And because of that it depends what we need to reach.
Generally, if we Issue RATS - All Mifare classic commands should not work coz they should work in iso1443-4. But chineese clones works.

merlokk, 2017-10-01:
Classic, Plus sl1, sl2(part), jcop mifare emulation  - 14443-3
Plus sl0, sl2, sl3, jcop javacard - 14443-4

iceman1001, 2017-10-06:
Can someone verfify that @pwpiwi  's fix solved this issue so we can close it?  @maxben14

merlokk, 2017-10-06:
all OK

```
proxmark3>  hf 14a reader
 UID : 60 e6 68 26
ATQA : 03 04
 SAK : 28 [1]
TYPE : JCOP31 or JCOP41 v2.3.1
 ATS : 0a 38 77 b1 4a 43 4f 50 33 30 da ae
       -  TL : length is 10 bytes
       -  T0 : TA1 is present, TB1 is present, TC1 is NOT present, FSCI is 8 (FSC = 256)
       - TA1 : different divisors are supported, DR: [2, 4, 8], DS: [2, 4, 8]
       - TB1 : SFGI = 1 (SFGT = 8192/fc), FWI = 11 (FWT = 8388608/fc)
       -  HB : 4a 43 4f 50 33 30
#db# halt error. response len: 3
Answers to chinese magic backdoor commands: NO
proxmark3> hf mf rdbl 0 a ffffffffffff
--block no:0, key type:A, key:ff ff ff ff ff ff
#db# READ BLOCK FINISHED
isOk:01 data:60 e6 68 26 c8 28 04 03 00 00 00 00 00 5c 92 29
proxmark3> hf list 14a
Recorded Activity (TraceLen = 188 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |     
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|     
          0 |        992 | Rdr | 52                                                              |     | WUPA
       2228 |       4596 | Tag | 04  03                                                          |     |
       7040 |       9504 | Rdr | 93  20                                                          |     | ANTICOLL
      10676 |      16564 | Tag | 60  e6  68  26  c8                                              |     |
      18816 |      29344 | Rdr | 93  70  60  e6  68  26  c8  17  79                              |  ok | SELECT_UID
      30516 |      34036 | Tag | 28  b4  fc                                                      |     |
      35584 |      40288 | Rdr | 60  00  f5  7b                                                  |  ok | AUTH-A(0)
      43316 |      47988 | Tag | e2  21  e7  4b                                                  |     |
      57472 |      66784 | Rdr | 9f 8b! 76!  7b  6f 0c! 53! d5!                                  | !crc| ?
      68148 |      72884 | Tag | de b9!  53 69!                                                  |     |
      78720 |      83424 | Rdr | 3e bf! b2!  f0                                                  | !crc| CHK_TEARING(191)         
      86196 |     107060 | Tag | d1 e4! 0a! 9c!  ac a5! 07!  f2 bd!  a1 51!  88  38 d9!  90 56!  |     |
            |            |     | 39  01                                                          | !crc|
     120064 |     124832 | Rdr | fc  ce  40 3f!                                                  | !crc| ?
```

maxben14, 2017-10-06:
@iceman1001 , i do fix and delete rats, problem fixed.

closed 2017-10-06 by iceman1001