Proxmark/proxmark3 #410: changes to hf mf  chk to speed up the testing

merlokk, 2017-10-05:
as discussed in https://github.com/Proxmark/proxmark3/pull/403

now it needs to have command on arm side to check keys on all sectors . This cycle https://github.com/merlokk/proxmark3/blob/nested-addfunct/client/cmdhfmf.c#L829 needs to be on arm side

as i see in
void MifareChkKeys(uint16_t arg0, uint8_t arg1, uint8_t arg2, uint8_t *datain) on mifarecmd.c - arm side.
there is no space to add cycle functionality. maybe high byte here?
bool clearTrace = arg1;
or it needs to add new command?

arg1 high byte will be sectors count.
and we may return 80 keys, 1 byte key index  ==  80 bytes (now it returns 6 bytes)

pwpiwi, 2017-10-05:
No, you could do everything on the client side. After the initial chk keys, create a new (empty) list of keys. For each sector: chk keys with the new list. If no key found, run nested on this sector. If nested finds a new key, add it to the list. Repeat with next sector.

merlokk, 2017-10-05:
for nested yes,
but i thought about speed up. nested found a key - we test this key on all the rest sectors. For this we need a check command with internal sector's cycle.
It will speed up nested a lot. and speeds up check command.

pwpiwi, 2017-10-05:
The major speedup is avoiding to run unecessary nested attacks by checking potential keys first. The difference between cycling through the sectors on client side compared to cycling on device side is negligible.

iceman1001, 2017-10-05:
I did an improved checkkeys six months ago,  migth aswell release it.   You will most likely benefit from the new methods.   Its about 10% faster than current implementation.

merlokk, 2017-10-05:
so what the algorithm?
we have sector/key for attack...
before attack we check keys, that already found and then run attack?

or after key is found - we check all the sectors if they have this key? - it seems this way will have more speed

merlokk, 2017-10-05:
@iceman1001 you have cool checkkeys ) ill try to import it) there are many changes that main proxmark repository dont have

iceman1001, 2017-10-06:
feel free to enjoy it :)  Thats the beauty of open source :)

merlokk, 2017-10-12:
here #414

closed 2017-10-16 by pwpiwi