Proxmark/proxmark3 #411: proxmark.exe crashes during second hardnested

gaucho1978, 2017-10-06:
there is a bug, i can call it "minor bug" in hardnested attack, even if it becomes important if we want to make a lua script for automatically get all the Keys of a tag.

the app crashes, this is the error reported by windows:

>  Name:	APPCRASH
  AppName:	proxmark3.exe
  Version:	0.0.0.0
  Timestamp :	02000002
  module name:	proxmark3.exe
  Version:	0.0.0.0
  Timestamp :	02000002
  exception Code:	c0000005
  exception Offset:	0007ef90
  SO Version:	6.1.7601.2.1.0.256.48
  ID local setting:	1040
  added information 1:	0a9e
  added information 2:	0a9e372d3b4ad19135b953a78882e789
  added information 3:	0a9e
  added information 4:	0a9e372d3b4ad19135b953a78882e789

the steps to reporoduce the error is the following:
ACTION 1) run "hf mf hardnested 0 A A0A1A2A3A4A5 19 B"
RESULT 1) the proxmark finds the key correctly
ACTION 2) run "hf mf hardnested 0 A A0A1A2A3A4A5 15 B"
RESULT 2) the proxmark app crashes

This is the log:

```
> proxmark3 com12
> Prox/RFID mark3 RFID instrument
> bootrom: master/v3.0.1-79-g3a05a1e-suspect 2017-09-22 19:29:20
> os: master/v3.0.1-79-g3a05a1e-suspect 2017-09-22 19:29:23
> LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
> HF FPGA image built for 2s30vq100 on 2017/07/13 at 08:44:13
> uC: AT91SAM7S512 Rev A
> Embedded Processor: ARM7TDMI
> Nonvolatile Program Memory Size: 512K bytes. Used: 198361 bytes (38%). Free: 325927 bytes (62%).
> Second Nonvolatile Program Memory Size: None
> Internal SRAM Size: 64K bytes
> Architecture Identifier: AT91SAM7Sxx Series
> Nonvolatile Program Memory Type: Embedded Flash Memory
> proxmark3> **hf mf hardnested 0 A A0A1A2A3A4A5 19 B**
> --target block no: 19, target key type:B, known target key: 0x000000000000 (not set), file action: none, Slow: No, Tests: 0
> time    | nonces | Activity                                                | expected to brute force
>         |         |                                                         | states         | time
> ------------------------------------------------------------------------------------------------------
>       0 |       0 | Start using 4 threads and AVX SIMD core                 |                 |
>       0 |       0 | Brute force benchmark: 241 million (2^27.8) keys/s      | 140737488355328 |    7d
>       7 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    7d
>      13 |     112 | Apply bit flip properties                               |    526890270720 | 36min
>      14 |     224 | Apply bit flip properties                               |    464051732480 | 32min
>      15 |     336 | Apply bit flip properties                               |    131341910016 |  9min
>      16 |     447 | Apply bit flip properties                               |     68624449536 |  5min
> db AcquireNonces: Can't select card (UID)
>      17 |     556 | Apply bit flip properties                               |     68624449536 |  5min
>      18 |     668 | Apply bit flip properties                               |     68624449536 |  5min
>      19 |     779 | Apply bit flip properties                               |     68624449536 |  5min
> db Authentication failed. Error card response.
> db AcquireNonces: Auth1 error
> db AcquireNonces: Can't select card (UID)
>       20 |     890 | Apply bit flip properties                               |     68624449536 |  5min
> db Authentication failed. Card timeout.
> db AcquireNonces: Auth1 error
> db Authentication failed. Error card response.
> db AcquireNonces: Auth1 error
> db AcquireNonces: Can't select card (UID)
>       21 |    1002 | Apply bit flip properties                               |     40817696768 |  3min
> db Authentication failed. Card timeout.
> db AcquireNonces: Auth1 error
>       22 |    1111 | Apply bit flip properties                               |     38436442112 |  3min
> db Authentication failed. Error card response.
> db AcquireNonces: Auth1 error
> db AcquireNonces: Can't select card (UID)
>       23 |    1218 | Apply bit flip properties                               |     38436442112 |  3min
>       24 |    1330 | Apply bit flip properties                               |     38436442112 |  3min
> db AcquireNonces: Can't select card (UID)
>       25 |    1440 | Apply bit flip properties                               |     38436442112 |  3min
>       29 |    1547 | Apply Sum property. Sum(a0) = 128                       |     22159554560 |  2min
> db AcquireNonces: Can't select card (UID)
>       33 |    1656 | Apply bit flip properties                               |     18453501952 |   77s
> db AcquireNonces: Can't select card (UID)
>       35 |    1767 | Apply bit flip properties                               |     17062355968 |   71s
>       38 |    1876 | Apply bit flip properties                               |     15487296512 |   64s
> db AcquireNonces: Auth1 error
> db AcquireNonces: Can't select card (UID)
>       39 |    1986 | Apply bit flip properties                               |     15487296512 |   64s
> db Authentication failed. Error card response.
> db AcquireNonces: Auth1 error
> db AcquireNonces: Can't select card (UID)
>       41 |    2096 | Apply bit flip properties                               |     15487296512 |   64s
>       42 |    2205 | Apply bit flip properties                               |     15487296512 |   64s
>       42 |    2205 | (Ignoring Sum(a8) properties)                           |     15487296512 |   64s
> db Authentication failed. Error card response.
> db AcquireNonces: Auth1 error
> db AcquireNonces: Can't select card (UID)
>       52 |    2205 | Starting brute force...                                 |     15487296512 |   64s
>      183 |    2205 | Brute force phase completed. Key found: 8dd3c1a937c3    |               0 |    0s
> proxmark3> **hf mf hardnested 0 A A0A1A2A3A4A5 15 B**
> --target block no: 15, target key type:B, known target key: 0x000000000000 (not set), file action: none, Slow: No, Tests: 0
>  time    | nonces | Activity                                                | expected to brute force
>          |         |                                                         | states         | time
> ------------------------------------------------------------------------------------------------------
>        0 |       0 | Start using 4 threads and AVX SIMD core                 |                 |
>        0 |       0 | Brute force benchmark: 250 million (2^27.9) keys/s      | 140737488355328 |    7d
>        6 |       0 | Using 235 precalculated bitflip state tables            | 140737488355328 |    7d
> db Authentication failed. Error card response.
> db AcquireNonces: Auth1 error
> db AcquireNonces: Can't select card (UID)
>       12 |     112 | Apply bit flip properties                               |  29933531299840 |   33h
> db AcquireNonces: Can't select card (UID)
>       13 |     224 | Apply bit flip properties                               |  24558023213056 |   27h
> db Authentication failed. Error card response.
> db AcquireNonces: Auth1 error
> db AcquireNonces: Can't select card (UID)
>       14 |     336 | Apply bit flip properties                               |  23888014606336 |   27h
>       15 |     446 | Apply bit flip properties                               |               0 |    0s
> db Authentication failed. Error card response.
> db AcquireNonces: Auth1 error
> db AcquireNonces: Can't select card (UID)
> db Authentication failed. Card timeout.
> db AcquireNonces: Auth1 error
> db Authentication failed. Card timeout.
> db AcquireNonces: Auth1 error
>       16 |     558 | Apply bit flip properties                               |               0 |    0s
> db AcquireNonces: Can't select card (UID)
>       17 |     669 | Apply bit flip properties                               |               0 |    0s
> db Authentication failed. Card timeout.
> db AcquireNonces: Auth1 error
> db AcquireNonces: Can't select card (UID)
>       19 |     779 | Apply bit flip properties                               |               0 |    0s
>       20 |     888 | Apply bit flip properties                               |               0 |    0s
> db Authentication failed. Error card response.
> db AcquireNonces: Auth1 error
> db AcquireNonces: Can't select card (UID)
>       21 |     998 | Apply bit flip properties                               |               0 |    0s
>       22 |    1106 | Apply bit flip properties                               |               0 |    0s
>       23 |    1217 | Apply bit flip properties                               |               0 |    0s
>       24 |    1326 | Apply bit flip properties                               |               0 |    0s
>       25 |    1438 | Apply bit flip properties                               |               0 |    0s
>       27 |    1548 | Apply bit flip properties                               |               0 |    0s
> 

```

iceman1001, 2017-10-06:
I already have a script which has none of your problems.  Same sourcecode (v3.0.1)
ref:  https://www.youtube.com/edit?o=U&video_id=tUZHtUJerJY

iceman1001, 2017-10-06:
given the nature of your problem, I would say a free memory based problem.  
How much ram do you have on the client machine?    When I used a virtual machine, with about 1gb,  the improved hardnested failed.  Once I increased it to 4gb, it worked just fine.  Also other on forum stated about 1.5 - 2gb was needed for them.

merlokk, 2017-10-06:
link to video https://www.youtube.com/watch?v=tUZHtUJerJY

gaucho1978, 2017-10-06:
This is really possible. My ram is always almos full due to some processes that I always run. I will retry with free ram, sorry.

commit 8bdb6043 (2017-10-25): identification fix for magic chinese cards (#444)

* fixed #411
closed 2017-10-25 by pwpiwi