Proxmark/proxmark3 #412: hf mf sim bug

maxben14, 2017-10-06:
I try to emulate the mifare classic and read through my smartphone  with android MCT apk.
proxmark3> hf mf sim n 0
mf 1k sim uid: N/A, numreads:0, flags:0 (0x00)
#db# 7B UID: 04 53 5d 42 a7 49 80
#db# Reader tried to operate (0x30) on out of range block: 222 (0xde), nacking
#db# Emulator stopped. Tracing: 1  trace length: 27908

proxmark3> hf list 14a
.......
188213078 |  188215446 | Tag | 44  00                                                          |     |
  188224958 |  188235486 | Rdr | 93  70  88  04  53  5d  82  17  d3                              |  ok | SELECT_UID     
  188236978 |  188240498 | Tag | 04  da  17                                                      |     |
  188243234 |  188245698 | Rdr | 95  20                                                          |     | ANTICOLL-2     
  188247382 |  188253270 | Tag | 42  a7  49  80  2c                                              |     |
  188257314 |  188267842 | Rdr | 95  70  42  a7  49  80  2c  2d  5e                              |  ok | ANTICOLL-2     
  188269398 |  188272918 | Tag | 08  b6  dd                                                      |     |
  188503146 |  188507914 | Rdr | 61  00  2d  62                                                  |  ok | AUTH-B(0)      
  188512414 |  188517150 | Tag | 01  02  03  04                                                  |     |
  188521244 |  188530620 | Rdr |95! cb!  40  8c 0f!  84  ec  75                                  | !crc| ANTICOLL-2     
  188539152 |  188543824 | Tag | e7  86  42  2d                                                  |     |
  188614336 |  188619104 | Rdr | 30  de  37  97                                                  | !crc| READBLOCK(222) 
  188622644 |  188623284 | Tag | 07                                                              |     |
  188826048 |  188827040 | Rdr | 52                                                              |     | WUPA
  188828596 |  188830964 | Tag | 44  00                                                          |     |
  188839586 |  188850114 | Rdr | 93  70  88  04  53  5d  82  17  d3                              |  ok | SELECT_UID     
  188851670 |  188855190 | Tag | 04  da  17                                                      |     |
  188857926 |  188860390 | Rdr | 95  20                                                          |     | ANTICOLL-2     
  188862074 |  188867962 | Tag | 42  a7  49  80  2c                                              |     |
  188872006 |  188882534 | Rdr | 95  70  42  a7  49  80  2c  2d  5e                              |  ok | ANTICOLL-2     
  188884090 |  188887610 | Tag | 08  b6  dd                                                      |     |
  188977072 |  188981840 | Rdr | 50  00  57  cd                                                  |  ok | HALT
................

I try decrypt this with mfkey32
C:\prox\ProxSpace\pm3\tools\mfkey>mfkey64 42a74980 01020304 95cb408c 0f84ec75 e786422d 30de3797
MIFARE Classic key recovery - based on 64 bits of keystream
Recover key from only one complete authentication!

Recovering key for:
   uid: 42a74980
    nt: 01020304
  {nr}: 95cb408c
  {ar}: 0f84ec75
  {at}: e786422d
{enc0}: 30de3797

LFSR successors of the tag challenge:
  nt' : 20f8ed56
  nt'': 3c2bcdad
Time spent in lfsr_recovery64(): 0.14 seconds

Keystream used to generate {ar} and {at}:
   ks2: 2f7c0123
   ks3: dbad8f80

Decrypted communication:
{dec0}: 500057cd

Found Key: [a0a1a2a3a4a5]

I found bug in this command. Why proxmark think that 30  de  37  97 this is decrypt communication ? After authorization, all communication is encrypted. And this is actually the halt command 50 00 57 cd.

And 2 problem with MCT: Error: None of the keys were valid for reading.
The MCT often returns an error when authorizing, when I attach a proxmark simulating a classic card to the phone. I tried it on acr122, everything works well there.

pwpiwi, 2017-10-07:
The `hf mf sim` command needs some work for the real simulation part (e.g. reading a block, writing a block - I don't think that this ever worked). However, WUPA, REQA, SELECT, HALT and the crypto1 authentication should work (i.e. those parts required to collect data for mfkey 🙁 ). 

Which version do you use? Please post the output of `hw ver`.

maxben14, 2017-10-07:
proxmark3> hw version
[[[ Cached information ]]]

Prox/RFID mark3 RFID instrument
bootrom: master/v3.0.1-94-g77aecdd2-dirty-suspect 2017-10-06 12:33:13
os: master/v3.0.1-94-g77aecdd2-dirty-suspect 2017-10-06 18:09:22
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/07/13 at 08:44:13

uC: AT91SAM7S256 Rev D
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 256K bytes. Used: 198800 bytes (76%). Free: 63344 bytes (24%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory

maxben14, 2017-10-07:
Before command hf mf sim n 0, i load dump 1k card in ff.eml file this comand hf mf eload ff.
proxmark3> hf mf eload ff
................................................................
Loaded 64 blocks from file: ff.eml
proxmark3> hf mf sim n 0
mf 1k sim uid: N/A, numreads:0, flags:0 (0x00)
#db# 4B UID: 0eb7d9b7

ff.eml file content:
0EB7D9B7D70804006263646566676869
00000000000000000000000000000000
00000000000000000000000000000000
A0A1A2A3A4A578778800A0A1A2A3A4A5
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF078069FFFFFFFFFFFF
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
FFFFFFFFFFFFFF0780BCFFFFFFFFFFFF

pwpiwi, 2017-10-07:
You indeed disovered two bugs. This is what happened:
* The reader did in fact send an unencrypted HALT command 50 00 57 cd
* Because there had been a successfull authentication before, the PM3 decrypted this command, resulting in 30 de 37 97 which it interpreted as a read block command.
* This command has a wrong CRC - but this is ignored.
* After a successfull authentication, the PM3 traces the decrypted reader commands.

So there are two bugs I am going to fix:
* CRC check. If CRC is wrong, card must switch to IDLE state.
* trace should always show the raw data

I stand corrected: the Mifare Classic commands simulation (read/write block, INC, DEC, etc) should work.

ToDo: Take Access Conditions into account when simulating. Currently they are ignored.

maxben14, 2017-10-07:
Okay, it's good.
But how to deal with the smartphone reader, why he sent a HALT. This is a timings problem, is it possible to somehow speed up the implementation of the algorithm crypto1, for example OpenMP?

pwpiwi, 2017-10-07:
I don't think that it is a timing problem. The only critical timings are during the anticollision phase when all cards in the reader field must respond at the same time. The responses during the authentication have to be send between 71us and 1ms after the reader command is received.

The AUTH-B command ended at 188507914 and tag started to respond at 188512414. This is a delay of 4500 * 1/13,56MHz = 332us which is well within the specified limits.

The reader challenge ended at 188530620 and the tag started to respond at 188539152, Again, the delay of 8532 * 1/13,56MHz = 629us is absolutely compliant with the specs.

You wrote that the simulation works against an ACR122. Wouldn't that indicate that the fault is with MCT? Did you compare the ACR122 and MCT traces?

maxben14, 2017-10-07:
Yes, i did compare with acr122 and MCT.
acr122

23858198 |   23860566 | Tag | 04  00                                                          |     |
   23873862 |   23884326 | Rdr | 93  70  0e  b7  d9  b7  d7  dc  48                              |  ok | SELECT_UID     
   23886010 |   23889530 | Tag | 08  b6  dd                                                      |     |
   26918380 |   26919372 | Rdr | 52                                                              |     | WUPA
   26920928 |   26923296 | Tag | 04  00                                                          |     |
   26936640 |   26947104 | Rdr | 93  70  0e  b7  d9  b7  d7  dc  48                              |  ok | SELECT_UID     
   26948596 |   26952116 | Tag | 08  b6  dd                                                      |     |
   29981000 |   29981992 | Rdr | 52                                                              |     | WUPA
   29983548 |   29985916 | Tag | 04  00                                                          |     |
   29999276 |   30009740 | Rdr | 93  70  0e  b7  d9  b7  d7  dc  48                              |  ok | SELECT_UID     
   30011360 |   30014880 | Tag | 08  b6  dd                                                      |     |
   33043746 |   33044738 | Rdr | 52                                                              |     | WUPA
   33046294 |   33048662 | Tag | 04  00                                                          |     |
   33061958 |   33072422 | Rdr | 93  70  0e  b7  d9  b7  d7  dc  48                              |  ok | SELECT_UID     
   33074106 |   33077626 | Tag | 08  b6  dd                                                      |     |
   35303750 |   35308454 | Rdr | 60  00  f5  7b                                                  |  ok | AUTH-A(0)      
   35312954 |   35317690 | Tag | 01  02  03  04                                                  |     |
   35319084 |   35328396 | Rdr | c9 95! 90! 5d! 71!  ec  f1 df!                                  | !crc| ?
   35337056 |   35341728 | Tag | 84  35  42  93                                                  |     |
   35546696 |   35551464 | Rdr |30! 00!  02  a8                                                  |  ok | READBLOCK(0)   
   35565884 |   35586684 | Tag |ce! bc!  f5 34! df!  ee 09!  78  bc  1f  b5 f3! 59! 88!  54 2f!  |     |
            |            |     | a6  36                                                          | !crc|
   46721152 |   46722144 | Rdr | 52                                                              |     | WUPA
   46723700 |   46726068 | Tag | 04  00                                                          |     |
   46739428 |   46749892 | Rdr | 93  70  0e  b7  d9  b7  d7  dc  48                              |  ok | SELECT_UID     
   46751384 |   46754904 | Tag | 08  b6  dd                                                      |     |
   49783722 |   49784714 | Rdr | 52                                                              |     | WUPA
   49786334 |   49788702 | Tag | 04  00                                                          |     |
   49801870 |   49812334 | Rdr | 93  70  0e  b7  d9  b7  d7  dc  48       
   
   
   nfc console, i try command 60 00 key uid & 30 00 and in console android apk this work.
   
   10816438 |   10818806 | Tag | 04  00                                                          |     |
   10827366 |   10837830 | Rdr | 93  70  0e  b7  d9  b7  d7  dc  48                              |  ok | SELECT_UID     
   10839514 |   10843034 | Tag | 08  b6  dd                                                      |     |
   11064064 |   11068832 | Rdr | 50  00  57  cd                                                  |  ok | HALT
   11111676 |   11112668 | Rdr | 52                                                              |     | WUPA
   11114224 |   11116592 | Tag | 04  00                                                          |     |
   11126056 |   11136520 | Rdr | 93  70  0e  b7  d9  b7  d7  dc  48                              |  ok | SELECT_UID     
   11138140 |   11141660 | Tag | 08  b6  dd                                                      |     |
   12974336 |   12979104 | Rdr | 50  00  57  cd                                                  |  ok | HALT
   13021288 |   13022280 | Rdr | 52                                                              |     | WUPA
   13023836 |   13026204 | Tag | 04  00                                                          |     |
   13035650 |   13046114 | Rdr | 93  70  0e  b7  d9  b7  d7  dc  48                              |  ok | SELECT_UID     
   13047734 |   13051254 | Tag | 08  b6  dd                                                      |     |
   14619024 |   14635248 | Rdr | 60  00  a0  a1  a2  a3  a4  a5  0e  b7  d9  b7  e3  a0          |  ok | AUTH-A(0)      
   19298428 |   19303196 | Rdr | 30  00  02  a8                                                  |  ok | READBLOCK(0)   
   19315568 |   19336368 | Tag | 0e  b7  d9  b7  d7  08  04  00  62  63  64  65  66  67  68  69  |     |
            |            |     | 9c  1e                                                          |  ok |
   19564136 |   19568904 | Rdr | 30  01  8b  b9                                                  |  ok | READBLOCK(1)   
   19581148 |   19602012 | Tag | 00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  00  |     |
[... cut ...]
   28007334 |   28009702 | Tag | 04  00                                                          |     |
   28019122 |   28029586 | Rdr | 93  70  0e  b7  d9  b7  d7  dc  48                              |  ok | SELECT_UID     
   28031270 |   28034790 | Tag | 08  b6  dd                                                      |     |
   29783968 |   29788736 | Rdr | 50  00  57  cd                                                  |  ok | HALT
   29832372 |   29833364 | Rdr | 52

I see in logs from MCT and android console apk many comand HALT and don't see HALT in log ACR122.

iceman1001, 2017-10-08:
**SIM**
Not sure if I agree about the idea of **sim**  trace should always show raw data. 

**HALT**
The concept that a mifare reader sends a un-encrypted HALT while been in the authenticated state,  seems more like a implementation bug in the reader software than a PM3 bug. 
It comes down to protocol,  iso14443a it should be un-encrypted,  but Mifare addition protocl above 14a protocol has its own quirks.  So to find the answer in how a HALT command should be implemented for mifare should be in their documents.  Not sure Mifare implementations is following protocol.

maxben14, 2017-10-08:
@iceman1001 , The main question is, why does not any android application want to read proxmark in the classic emulation mode?

pwpiwi, 2017-10-08:
@maxben14: your traces show two more bugs in hf mf sim:
* You made an error when using the AUTH-A command. Even crypto1 doesn't send the key in clear. Consequentially the authentication failed. But PM3 answered to the read block commands.
* The timing in the anticollision phase is wrong. As I said before, the timing needs to be very accurate to detect and distinguish several cards in the reader field. In this case there is one card only but the phone's NFC chip might be nitpicking about the correct timing.

@iceman1001: A trace should always be raw data because it can be interpreted in different ways. Interpreting traces is the task of the `hf list` commands. Decrypting encrypted data should be done there if necessary (e.g. with a new `hf list mf`)

iceman1001, 2017-10-08:
**hf list mf **  yeap,  that would be good, been thinking on it for some years, never got around to do it. The problem would be that the list command does have the key needed to decrypt while the sim has it.
and starting calling list with a key seems not optimal.   I do agree with "hf list 14a"  or  "hf 14a reader" does too much.    A  "hf mf info"  command would also be better.   A side note to the actual subject.

--
crc check to the decrypted command will solve the wrong answer to HALT.  That will ensure PM3 doesn't do wrong.  We can NOT garantue the Android or ACR122 implementations is correct as it looks now.

merlokk, 2017-10-08:
As i remembered there is no problem to decrypt first Authentication command, but a problem with decryption the next

pwpiwi, 2017-10-20:
@maxben14: could you please run your tests again with PR #419 and post the results here?

pwpiwi, 2017-10-20:
@maxben14: changes are committed. Looking forward to your results...

cjbrigato, 2017-10-23:
So far so good it works VERY well ... depending on the device.
Same device from same manufacturer same year... same firwmare... One is able to be detefcted near anywhere now, the other seems to be nothing more than a reader/sniffer/snooper.

Anyway, the fix from @pwpiwi  is absolutely crazy on my side... I'm able to chain succesffull Readers attacks in the wild in 1~3 try with less than 1 sec communication, ensuring a good 20% succes into finding at least a B-Key (depending on the UID chosen, because of vigik system, if it can't authenticate with A key, it will try RSA-signed services-codes/values from B block., whatever).

However, still it will _ALWAYS_ fail to face a standard reader beacause of a very simple thing : When authenticated, wheter it be from B  or A key, `hf mf sim `**ALWAYS** gives something like 

``` 
 a6 7f 8b f5 a7 88 04 00 46 8e 45 59 61 10 50 04
 01 07 4e 00 00 00 00 00 00 00 00 00 00 00 00 00
 55 55 00 00 00 00 00 00 00 00 00 00 00 00 00 00
 88 29 da 9d af 76 7f 07 88 00 88 29 da 9d af 76
``` 

So yes, @pwpiwi , Access bit condition are not followed... 
If i'm not mistaken, whatever would Acess bits be for any sector, the A key can **Never** be read.
At the very least, the device should **Never** send a sector trailer which doesn't start by "00 00 00 00 00 00 " ... 

This little fix should be enough to make the device work in more than 50% classic residential and standard implementations  (Well at least *here* in france). 

I've hoped that I could just blank the A key from emulator memory, however it makes the reader failing to authenticate with such a key  so...

Also timing are very very important here  too for first successu full authentification : Reader will immediately try an A key auth on sector 0 (for residentials), and upon fail will ONLY loop againts Bkey value block (for VIGIK Services codes/temporary"master keys" as mentioned above.) An A key auth will **not** be attempted again without the tag being out of the field for at least a few seconds (on vigik systems this state is indicated by a little red light instead of green, flashing with the same speed as the Auth are tryed).
Also  a note here it is quite easy on these systems to find suitable UID's for reader attacks : they are making the reader behave differently. As a master-key rsa-signe temporary masterkey could have any uid, an unknown uid will be challenged againts that behavior, creating flashing light. However a falining real candidate fro uid will trigger a "shutdown" or so of the filed , the red light will not flash, then shutdown and nothing will happen without Tag being out of the field + quick timeout.

Quite a bit of digression here, but anyway information worth the knowledge since it points towards specific needs on the Simulation side behavior.
Which, again, seems to be able to become like a bright sunday on some device  since your patch.

maxben14, 2017-10-27:
@pwpiwi , i try new version but comand "hf mf sim n 0" have again problems with auth on android smartphone.
9829638 |    9832006 | Tag | 44  00                                                          |     |
    9841036 |    9851564 | Rdr | 93  70  88  04  53  5d  82  17  d3                              |  ok | SELECT_UID     
    9852736 |    9856256 | Tag | 04  da  17                                                      |     |
    9858974 |    9861438 | Rdr | 95  20                                                          |     | ANTICOLL-2     
    9862610 |    9868498 | Tag | 42  a7  49  80  2c                                              |     |
    9872524 |    9883052 | Rdr | 95  70  42  a7  49  80  2c  2d  5e                              |  ok | ANTICOLL-2     
    9884224 |    9887744 | Tag | 08  b6  dd                                                      |     |
   10608680 |   10613384 | Rdr | 60  00  f5  7b                                                  |  ok | AUTH-A(0)      
   10618012 |   10622748 | Tag | 01  02  03  04                                                  |     |
   10626782 |   10636094 | Rdr | b7  76  2e 8e!  6e ae! 8a! fe!                                  | !crc| ?
   10644690 |   10649362 | Tag | b4  7e  61  24                                                  |     |
   10698634 |   10703402 | Rdr | 50  00  57  cd                                                  |  ok | HALT
   10707134 |   10707774 | Tag |03!                                                              |     |
   10890486 |   10891478 | Rdr | 52                                                              |     | WUPA
   10892714 |   10895082 | Tag | 44  00                                                          |     |
   10903288 |   10913816 | Rdr | 93  70  88  04  53  5d  82  17  d3                              |  ok | SELECT_UID     
   10914988 |   10918508 | Tag | 04  da  17                                                      |     |
   10921210 |   10923674 | Rdr | 95  20                                                          |     | ANTICOLL-2     

I see after good auth my android send HALT in tag. Is it problem timing android ?

iceman1001, 2017-10-27:
...some information seems missing.  In order to make use the "hf mf sim" command with authentication, you would need to load a proper dump into the device first.   With correct keys etc.. 

From your trace its not much to deduct.

maxben14, 2017-10-27:
Of course, i before write 
Before command hf mf sim n 0, i load dump 1k card in ff.eml file this comand hf mf eload ff.
proxmark3> hf mf eload ff

iceman1001, 2017-10-27:
and all correct keys are in it?

maxben14, 2017-10-27:
@iceman1001 , yes in 0 sector key a0a1a2a3a4a5 and in mct in key file this key, but acr122 good read when i do `hf mf sim n 0`

pwpiwi, 2017-10-27:
I think that there is still room for improvement regarding timing. Will try to get hold of a DSO to check.

cjbrigato, 2017-11-02:
I have a very strange addition but might be insightfull..
(if anyone could test this further I would be glad..)

I have three pm3 here (all are based on Elechouse V2 dev kit from different years and different quality) but they all act the same. Also this has been tested on several [french vigik residential] readers with consistent repeatable behavior.

* Take this dump : 
```
4285BA87FA880400C185149451703411
00008049000000000000000000000000
00000000000000000000000000000000
414C41524F4EFF078000424C41524F4E
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000127002793
414C41524F4EFF078000424C41524F4E
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
414C41524F4EFF078000424C41524F4E
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
414C41524F4EFF078000424C41524F4E
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
414C41524F4EFF078000424C41524F4E
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
414C41524F4EFF078000424C41524F4E
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
414C41524F4EFF078000424C41524F4E
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
414C41524F4EFF078000424C41524F4E
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
414C41524F4EFF078000424C41524F4E
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
414C41524F4EFF078000424C41524F4E
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
414C41524F4EFF078000424C41524F4E
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
414C41524F4EFF078000424C41524F4E
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
414C41524F4EFF078000424C41524F4E
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
414C41524F4EFF078000424C41524F4E
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
414C41524F4EFF078000424C41524F4E
00000000000000000000000000000000
00000000000000000000000000000000
00000000000000000000000000000000
414C41524F4EFF078000424C41524F4E
```
* And load it into the emulator memory.
* Start Mifare Classic simulation (Mifare1ksim())
* Challenge an "advanced reader" such as a relatively new nfc phone : It should work flawlessly
* Change the uid from 4285BA87 to 0A0A0A0A : it will work the very same
* This should also work with PN532 and alikes : even mfoc is working.
* Now challenge a "classic" standard french vigik residential reader with the very same dump 
  * With UID 0A0A0A0A, it will work near flawlessly everytime, i.e it will authenticate to A Key of sector 0. If the reader is looking for this key and uid, it will work. For a reader not looking for such UID or Key sets, it will fail with a "red" led, indicating a wrong tag.
 * With ORIGINAL uid (4285BA87), nothing seems to happen, and the logs shows that the reader never manage to go beyond anticollision, trying to do anticoll procedure again and again and again... !
* Now let's try something totally different : clear the whole emulator memory
* Start a simulation again, with UID 4285BA87 and blank memory
  * Reader will act the very same : not going beyond anticollision
* Now start a simulation with UID 0A0A0A0A : It will just "fail" (red light) immediately (because of blank dump)

**So the "quality" of simulation seems to depend on UID **Somehow****

* Now I tried something different to verify this behavior. 
* I challenge same class reader which I **KNOW** is not supposed to work for a tag with UID 0A0A0A0A but definitely should with 4285BA87.
* I Start simulation with flags for the reader attack
  * With UID 4285BA87 : never goes beyon anticol, nothing happens
  * With UID 0A0A0A0A: I happen to collect needed trace in near always less than ~2sec
     * Attack works providing a working B key ~3/5 of the time !

I made a lot of tests like that and whatever a UID is know to work with such Residential Readers or not, the behavior of the simulation drastically change with UID... two UID unknown to the Residential system can work very well or not at all, two Known UID can work very well or not at all... 
But this is all consistent : a "not working" UID like 4285BA87 will *never* work with any of such reader (but still work with e.g. nfc android phones) and UID 0A0A0A0A will *always* work with any of such reader (and also work with nfc android phones)...

@pwpiwi  what's your opinion on this ?

pwpiwi, 2017-11-02:
This is an interesting observation. I had noticed that 4 byte UID works, but 7 byte UID didn't. But I didn't try different 4Byte and 7Byte UIDs. Your observation might direct us to find the still existing (or yet another) `hf mf sim` bug.

merlokk, 2017-11-02:
It needs to sniff bad communication with 2nd proxmark. Maybe there are some timing Issues or anthenna quality issues

pwpiwi, 2017-11-02:
Good idey, but I am lacking the 2nd proxmark. Trying to get hold of a DSO.

merlokk, 2017-11-02:
@cjbrigato have)  @cjbrigato can you sniff and POST here the results?
P. S. I have only one too

iceman1001, 2017-11-02:
i do have several pm3,  i also has a vigik reader at home with tags,  so it would be possible to look at it,
however Im busy with fixing up the firmware for chameleon mini.  
Im certain @cjbrigato  will figure it out.

iceman1001, 2017-11-10:
now,  is problems solved?  
time to close?

maxben14, 2017-11-12:
@iceman1001, no, proxmark in mode emulator 1k work with android very bad, only acr122 good read,write emulator 1k proxmark.

pwpiwi, 2017-11-12:
"very bad" and "good read,write" is a bit vague :smile:. Can you please provide some traces again?

maxben14, 2017-11-26:
Sniff beetwen real tag and android:
proxmark3> hf list 14a
Recorded Activity (TraceLen = 2051 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

    ```
  Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |      38112 | Rdr | f0  1e  d4  00  66  20  62  03  aa  e7  65  9a  8b  59  00  00  |     |
            |            |     | 00  32  46  66  6d  01  01  11  03  02  00  13  04  01  96  25  |     |
            |            |     | c0                                                              |  ok | ?
     141744 |     142800 | Rdr | 26                                                              |     | REQA
     143988 |     146356 | Tag | 04  00                                                          |     |
     153968 |     156432 | Rdr | 93  20                                                          |     | ANTICOLL       
     157620 |     163508 | Tag | 01  02  03  04  04                                              |     |
     167504 |     178032 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
     179220 |     182740 | Tag | 08  b6  dd                                                      |     |
     790368 |     795136 | Rdr | 50  00  57  cd                                                  |  ok | HALT
     837888 |     838880 | Rdr | 52                                                              |     | WUPA
     840132 |     842500 | Tag | 04  00                                                          |     |
     851504 |     862032 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
     863220 |     866740 | Tag | 08  b6  dd                                                      |     |
    1065872 |    1070640 | Rdr | 50  00  57  cd                                                  |  ok | HALT
    1114096 |    1115088 | Rdr | 52                                                              |     | WUPA
    1116340 |    1118708 | Tag | 04  00                                                          |     |
    1127728 |    1138256 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
    1139444 |    1142964 | Tag | 08  b6  dd                                                      |     |
    1242528 |    1247296 | Rdr | 50  00  57  cd                                                  |  ok | HALT
    1290048 |    1291040 | Rdr | 52                                                              |     | WUPA
    1292292 |    1294660 | Tag | 04  00                                                          |     |
    1303680 |    1314208 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
    1315396 |    1318916 | Tag | 08  b6  dd                                                      |     |
    1414688 |    1419392 | Rdr | 60  01  7c  6a                                                  |  ok | AUTH-A(1)      
    1421412 |    1426084 | Tag | 15  80  12  e4                                                  |     |
    1430176 |    1439552 | Rdr | bb fa! ae! cf! ab!  9f 52!  32                                  | !crc| ?
    1440740 |    1445412 | Tag |bc! 85!  b1 c2!                                                  |     |
    1574480 |    1579184 | Rdr |25!  89  5c d0!                                                  | !crc| ?
    1580436 |    1601236 | Tag |0a!  17  ba a3! 93!  6f  13  98 e4!  a7 b5! 43! 6b!  06 58!  72  |     |
            |            |     | e1  3f                                                          | !crc|
    1684144 |    1688848 | Rdr | d6 34! 18! 76!                                                  | !crc| ?
    1690100 |    1710900 | Tag |df!  02  0b db!  e7  38 03!  2b c8! 8a! 4c!  a4  6d 69!  31 13!  |     |
            |            |     | 17  06                                                          | !crc|
    1819152 |    1823856 | Rdr | 3a da! 97!  9c                                                  | !crc| READ RANGE (218-151)
    1825108 |    1845972 | Tag | 9c 14!  2d  2e  64 df!  7c 43! d8!  6a  14  4e  92  1d 18! 5b!  |     |
            |            |     | 1d f5!                                                          | !crc|
    1949984 |    1954752 | Rdr |9f! 3b! 0f!  15                                                  | !crc| ?
    1997520 |    1998512 | Rdr | 52                                                              |     | WUPA
    1999764 |    2002132 | Tag | 04  00                                                          |     |
    2011152 |    2021680 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
    2022868 |    2026388 | Tag | 08  b6  dd                                                      |     |
    2148400 |    2153168 | Rdr | 50  00  57  cd                                                  |  ok | HALT
    2195936 |    2196928 | Rdr | 52                                                              |     | WUPA
    2198196 |    2200564 | Tag | 04  00                                                          |     |
[... cut ...]
   30946912 |   30950432 | Tag | 08  b6  dd                                                      |     |
   32760540 |   32765308 | Rdr | 50  00  57  cd                                                  |  ok | HALT
   32810432 |   32812800 | Tag | 04  00                                                          |     |
   32833568 |   32837088 | Tag | 08  b6  dd                                                      |     |
   34690064 |   34692432 | Tag | 04  00                                                          |     |
   34713216 |   34716736 | Tag | 08  b6  dd
```

pwpiwi, 2017-11-26:
Please have a look at the topic of this issue. I don't see a trace of `hf mf sim`?

maxben14, 2017-11-26:
@pwpiwi , proxmark3> hf mf eload 98
..#db# Emulator stopped. Tracing: 1  trace length: 0
..............................................................
Loaded 64 blocks from file: 98.eml
proxmark3> hf mf sim n 0
mf 1k sim uid: N/A, numreads:0, flags:0 (0x00)
#db# 4B UID: 01020304

iceman1001, 2017-11-26:
...this issue is starting to become very unclear.  I suggest we close it and when someone actually has problems with "hf mf sim",  with provided needed information, they can open a new issue.

pwpiwi, 2017-11-26:
@maxben14 : still no trace? You need to run `hf list 14a` after `hf mf sim`.

maxben14, 2017-11-26:
@pwpiwi, i see in log after correct AUTH i see android send HALT.
my log proxmark how emulator 1k.
proxmark3> hf list 14a
Recorded Activity (TraceLen = 1670 bytes)

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
iso14443a - All times are in carrier periods (1/13.56Mhz)
iClass    - Timings are not as accurate

      Start |        End | Src | Data (! denotes parity error)                                   | CRC | Annotation         |
------------|------------|-----|-----------------------------------------------------------------|-----|--------------------|
          0 |      38112 | Rdr | f0  1e  d4  00  9a  17  98  87  0d  3f  1c  88  68  19  00  00  |     |
            |            |     | 00  32  46  66  6d  01  01  11  03  02  00  13  04  01  96  6f  |     |
            |            |     | c6                                                              |  ok | ?
     141712 |     142768 | Rdr | 26                                                              |     | REQA
     143940 |     146308 | Tag | 04  00                                                          |     |
     154128 |     156592 | Rdr | 93  20                                                          |     | ANTICOLL       
     157764 |     163652 | Tag | 01  02  03  04  04                                              |     |
    6031184 |    6069296 | Rdr | f0  1e  d4  00  50  67  79  6d  b7  7e  9e  43  9f  17  00  00  |     |
            |            |     | 00  32  46  66  6d  01  01  11  03  02  00  13  04  01  96  71  |     |
            |            |     | 05                                                              |  ok | ?
    6172908 |    6173964 | Rdr | 26                                                              |     | REQA
    6175136 |    6177504 | Tag | 04  00                                                          |     |
    6185522 |    6187986 | Rdr | 93  20                                                          |     | ANTICOLL       
    6189158 |    6195046 | Tag | 01  02  03  04  04                                              |     |
    6199106 |    6209634 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
    6210806 |    6214326 | Tag | 08  b6  dd                                                      |     |
    6651568 |    6656336 | Rdr | 50  00  57  cd                                                  |  ok | HALT
    6698472 |    6699464 | Rdr | 52                                                              |     | WUPA
    6700700 |    6703068 | Tag | 04  00                                                          |     |
    6712132 |    6722660 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
    6723832 |    6727352 | Tag | 08  b6  dd                                                      |     |
    6861712 |    6866480 | Rdr | 50  00  57  cd                                                  |  ok | HALT
    6909342 |    6910334 | Rdr | 52                                                              |     | WUPA
    6911570 |    6913938 | Tag | 04  00                                                          |     |
    6922984 |    6933512 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
    6934684 |    6938204 | Tag | 08  b6  dd                                                      |     |
    6993904 |    6998672 | Rdr | 50  00  57  cd                                                  |  ok | HALT
    7041514 |    7042506 | Rdr | 52                                                              |     | WUPA
    7043742 |    7046110 | Tag | 04  00                                                          |     |
    7055206 |    7065734 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
    7066906 |    7070426 | Tag | 08  b6  dd                                                      |     |
    7213456 |    7218160 | Rdr | 60  01  7c  6a                                                  |  ok | AUTH-A(1)      
    7222724 |    7227396 | Tag | 00  00  00  00                                                  |     |
    7231512 |    7240824 | Rdr | 38 ea! ca!  56 09! 5f! cd!  b8                                  | !crc| ?
    7249548 |    7254220 | Tag | 48  79  15  09                                                  |     |
    7332986 |    7337754 | Rdr | 50  00  57  cd                                                  |  ok | HALT
    7341486 |    7342126 | Tag |03!                                                              |     |
    7456032 |    7457024 | Rdr | 52                                                              |     | WUPA
    7458260 |    7460628 | Tag | 04  00                                                          |     |
    7468868 |    7479396 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
    7480568 |    7484088 | Tag | 08  b6  dd                                                      |     |
    7682382 |    7687086 | Rdr | 60  01  7c  6a                                                  |  ok | AUTH-A(1)      
    7691586 |    7696258 | Tag | 00  00  00  00                                                  |     |
    7700366 |    7709742 | Rdr |b1!  e5  38 7a! 2c!  c4  bd 42!                                  | !crc| ?
[... cut ...]
   22959064 |   22963832 | Rdr | 60  08  bd  f7                                                  |  ok | AUTH-A(8)      
   22968204 |   22972876 | Tag | 00  00  00  00                                                  |     |
   22976976 |   22986288 | Rdr |82!  da  db  a8 9e! 8e! 1b!  0f                                  | !crc| ?
   22994884 |   22999556 | Tag | 56  94  5a  21                                                  |     |
   23099770 |   23104538 | Rdr | 50  00  57  cd                                                  |  ok | HALT
   23108270 |   23108846 | Tag |0a!

maxben14, 2017-11-26:
1. Why android send HALT after succesfull AUTH and second question why android send HALT decrypt 50 00, after correct AUTH communication need be encrypt but android decrypt command HALT send.

iceman1001, 2017-11-26:
so,  to make sure,  are you running the latest source code (flashed and compiled) from pm3 offical github?
It contains some fixes the last week.

merlokk, 2017-11-26:
`android decrypt command HALT send` - this is an android problem. send commands wo authentication.
cards just leaves field in this case. It have no sense)

as for here
```
7231512 | 7240824 | Rdr | 38 ea! ca! 56 09! 5f! cd! b8 | !crc| ?
7249548 | 7254220 | Tag | 48 79 15 09 | |
7332986 | 7337754 | Rdr | 50 00 57 cd | ok | HALT
```
1unit-74ns
7249548 - 7240824 = 8724 = 0.6ms (it seems that it good, but not so good as real card.)
 7332986 -7254220 = 78766 =  5.8ms 

so it may be just timeout.  and im sure that android dont see authentication. here also may be problems with field (and to be sure it needs to sniff communication with 2nd proxmark or another sniffer)

maxben14, 2017-11-26:
@iceman1001, my version is last.
proxmark3> hw version
[[[ Cached information ]]]

Prox/RFID mark3 RFID instrument
bootrom: master/v3.0.1-191-g1497150-suspect 2017-11-26 08:37:57
os: master/v3.0.1-191-g1497150-suspect 2017-11-26 08:38:01
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/10/27 at 08:30:59

uC: AT91SAM7S256 Rev D
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 256K bytes. Used: 199396 bytes (76%). Free: 62748 bytes (24%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory

maxben14, 2017-11-26:
@pwpiwi , i think basic problem in `hf mf sim` is  that the algorithm calculates the filter function not according to tables, but by formulas. I think the calculation on the tables will speed up the emulator.

merlokk, 2017-11-26:
will, or calculating in FPGA....
but there is no room for that.

pwpiwi, 2017-11-27:
@maxben14 : Can you please do the same with your ACR122 reader?

maxben14, 2017-11-27:
@pwpiwi ,
beetwen my emulator and acr122:
        0 |       2560 | Tag | d1  ff 00!                                                      |     |
   21435424 |   21437792 | Tag | 04  00                                                          |     |
   21511440 |   21513808 | Tag | 04  00                                                          |     |
   21587488 |   21589856 | Tag | 04  00                                                          |     |
   22174224 |   22176592 | Tag | 04  00                                                          |     |
   22249744 |   22252112 | Tag | 04  00                                                          |     |
   22325264 |   22327632 | Tag | 04  00                                                          |     |
   26007436 |   26008492 | Rdr | 26                                                              |     | REQA
   26009680 |   26012048 | Tag | 04  00                                                          |     |
   26020732 |   26023196 | Rdr | 93  20                                                          |     | ANTICOLL       
   26024400 |   26030288 | Tag | 01  02  03  04  04                                              |     |
   26051580 |   26062108 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
   26063312 |   26066832 | Tag | 08  b6  dd                                                      |     |
   30179244 |   30180236 | Rdr | 52                                                              |     | WUPA
   30181488 |   30183856 | Tag | 04  00                                                          |     |
   30197148 |   30207676 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
   30208880 |   30212400 | Tag | 08  b6  dd                                                      |     |
   33241196 |   33242188 | Rdr | 52                                                              |     | WUPA
   33243440 |   33245808 | Tag | 04  00                                                          |     |
   33259100 |   33269628 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
   33270816 |   33274336 | Tag | 08  b6  dd                                                      |     |
   36303132 |   36304124 | Rdr | 52                                                              |     | WUPA
   36305376 |   36307744 | Tag | 04  00                                                          |     |
   36320908 |   36331436 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
   36332624 |   36336144 | Tag | 08  b6  dd                                                      |     |
   39364940 |   39365932 | Rdr | 52                                                              |     | WUPA
   39367200 |   39369568 | Tag | 04  00                                                          |     |
   39382716 |   39393244 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
   39394448 |   39397968 | Tag | 08  b6  dd                                                      |     |
   42426748 |   42427740 | Rdr | 52                                                              |     | WUPA
   42429008 |   42431376 | Tag | 04  00                                                          |     |
   42444524 |   42455052 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
   42456240 |   42459760 | Tag | 08  b6  dd                                                      |     |
   45488556 |   45489548 | Rdr | 52                                                              |     | WUPA
   45490800 |   45493168 | Tag | 04  00                                                          |     |
   45506460 |   45516988 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
   45518176 |   45521696 | Tag | 08  b6  dd                                                      |     |
   48550508 |   48551500 | Rdr | 52                                                              |     | WUPA
   48552752 |   48555120 | Tag | 04  00                                                          |     |
   48568284 |   48578812 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
   48580000 |   48583520 | Tag | 08  b6  dd                                                      |     |
   51612316 |   51613308 | Rdr | 52                                                              |     | WUPA
   51614560 |   51616928 | Tag | 04  00                                                          |     |
   51630092 |   51640620 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
   51641824 |   51645344 | Tag | 08  b6  dd                                                      |     |
   54674124 |   54675116 | Rdr | 52                                                              |     | WUPA
   54676384 |   54678752 | Tag | 04  00                                                          |     |
   54691900 |   54702428 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
   54703648 |   54707168 | Tag | 08  b6  dd                                                      |     |
[... cut ...]
            |            |     |17! e0!                                                          | !crc|
   79813388 |   79818156 | Rdr |59!  af 73!  68                                                  | !crc| ?
   79844032 |   79864896 | Tag |58!  fd  f7 b9! a7! 55! 7c!  2b  48 90! 77!  ff 62!  9b c0!  cb  |     |
            |            |     | ff fa!                                                          | !crc|
  101814028 |  101815020 | Rdr | 52                                                              |     | WUPA
  101816272 |  101818640 | Tag | 04  00

pwpiwi, 2017-11-27:
@maxben14: sorry, I wasn't clear enough. Can you please run `hf mf sim` (after you have loaded a card content with `hf eload`) against your ACR122. Then run `hf list 14a` and provide the output here.

maxben14, 2017-11-27:
[1.txt](https://github.com/Proxmark/proxmark3/files/1507184/1.txt)

My log beetwen acr122 and proxmark as emulator 1k.

pwpiwi, 2017-11-28:
The initially identified bugs have been fixed. The trace with `hf mf sim`shows no incorrect tag responses. 

Both traces (with @maxben14 's simulator and with `hf mf sim`) show a consistent behaviour: it always takes three attempts to authenticate. The trace of the third attempt can be used to extract the key with mfkey64: a0a1a2a3a4a5. The first two attempts cannot be used to extract a key with mfkey32. Hence we can assume that the reader tries three different keys before succeeding. 

@maxben14: your simulator has timing issues. The response to a REQA must start after exactly 1172 carrier clock cycles, the response to a WUPA must start after exactly 1236 clock cycles.

closed 2017-11-28 by pwpiwi

cjbrigato, 2017-11-28:
Everything IS working as expected. Great work @pwiwi

maxben14, 2017-11-29:
@pwpiwi , "it always takes three attempts to authenticate." android system takes 2 attempts AUTH in 0 sector, my app takes only one attempt AUTH.

pwpiwi, 2017-11-29:
I am counting three:
```
76362588 | 76367292 | Rdr | 60 3f 81 b2 | ok | AUTH-A(63)
76760700 | 76765404 | Rdr | 60 3f 81 b2 | ok | AUTH-A(63)
77171052 | 77175756 | Rdr | 60 3f 81 b2 | ok | AUTH-A(63)
``|

maxben14, 2017-11-29:
@pwpiwi , you watch log beetwen acr122 and proxmark how emulator. I said that have problem with my device and android.

pwpiwi, 2017-11-29:
... and I continue to tell you that this is the proxmark github repository. Not the Android repository and not your device repository.

iceman1001, 2017-11-29:
Its hard with all these repositories,  which one to write to?

cjbrigato, 2017-11-29:
Starting a vote to rename this issue from "hf mf sim bug" to "maxben14 bug".

maxben14, 2017-11-29:
@pwpiwi, but proxmark in emulator mode with new firmware also have problem AUTH with android.
19694544 |   19695536 | Rdr | 52                                                              |     | WUPA
   19696772 |   19699140 | Tag | 04  00                                                          |     |
   19708184 |   19718712 | Rdr | 93  70  01  02  03  04  04  8e  25                              |  ok | SELECT_UID     
   19719884 |   19723404 | Tag | 08  b6  dd                                                      |     |
   20280162 |   20284866 | Rdr | 60  00  f5  7b                                                  |  ok | AUTH-A(0)      
   20289558 |   20294294 | Tag | 01  02  03  04                                                  |     |
   20298338 |   20307714 | Rdr | 6c 58!  c8 8f!  a4 ca! 91! 82!                                  | !crc| ?
   20316310 |   20321046 | Tag | 9b  0d  e1  07                                                  |     |
   20372708 |   20377476 | Rdr | 50  00  57  cd                                                  |  ok | HALT
   20381080 |   20381720 | Tag | 07

merlokk, 2017-11-29:
This problem on android side) Proxmark not so fast

maxben14, 2017-11-29:
@merlokk , problem in timing of proxmark. A good transport turnstile can easily distinguish a proxmark from a real card. Is it possible to speed up the performance of a `crypto1_bit` function so that the response time is close to that of real cards ?@merlokk , problem in timing of proxmark. A good transport turnstile can easily distinguish a proxmark from a real card. Is it possible to speed up the performance of a `crypto1_bit` function so that the response time is close to that of real cards ?

iceman1001, 2017-11-30:
This issue is closed.  discussion on enhancing performance of the current crypto1 impl is interesting but not suited for Github but on the forum.  If @maxben14 want fast impl,  feel free to do so.  No-one has "really" improved it since 4years when @pwpiwi  change sorting algo inside it.

pwpiwi, 2017-11-30:
The improved sorting speeded up the nested attack only. I would assume that @merlokk is right and the problem is on android side. The timing of `hf mf sim` is OK according to the NXP specs. And I would assume that commercial readers are obeying the NXP specs as well.

maxben14, 2017-11-30:
@iceman1001, can do calculate filter function with help tables and win many cycles of CPU time.

[1 more reply]