Proxmark/proxmark3 #413: proxdroid log empty

maxben14, 2017-10-08:
I try compile proxdroid 2.0.0. with this link http://www.proxmark.org/forum/viewtopic.php?id=1750&p=1
I copy on android proxmark3 and lib*.so in /system/lib. I connect by usb android with proxmark.
Device work, read card, but comand `hf 14a list` work very strange, i think this bug in this comand.

u0_a203@android:/ $ su
root@android:/ # cd sdcard
root@android:/sdcard # proxmark3 /dev/ttyACM0
proxmark3> hf mf rdbl 0 a a0a1a2a3a4a5
--block no:00 key type:00 key:a0 a1 a2 a3 a4 a5      
#db# READ BLOCK FINISHED
isOk:01 data:04 4a 93 6a 8a 44 80 08 44 00 12 01 11 00 23 16
proxmark3> hf 14a list
Recorded Activity

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
All times are in carrier periods (1/13.56Mhz)        

     Start |       End | Src | Data
-----------|-----------|-----|--------
         0 |  85370031 | Rdr | 00  05  24  00  00  40  09  02  80  44  00  c0  d1  36  00  00  a0  09  02  00  93  20  80  05  45  00  00  c0  16  05  80  88  04  4a  93  55  98  d1  64  00  00  20  29  09  00  93  70  88  04  4a  93  55  14  38  a7  00  95  92  00  00  c0  0d  03  80  04  da  17  20  51  a5  00  00  a0  09  02  00  95  20  80  95  b3  00     !crc     
 1397853395 |  83922882 | Tag | 00! 95! 70  6a! 8a  44! 80  24! 77!    !crc
 906519730 |    259855 | Tag | 00

After read card i get very strange log.
Then i try sniff and list is empty.

proxmark3> hf 14a snoop
#db# cancelled by button
#db# COMMAND FINISHED
#db# maxDataLen=4, Uart.state=0, Uart.len=0          
#db# traceLen=5611, Uart.output[0]=00000095          
proxmark3> hf 14a list
Recorded Activity

Start = Start of Start Bit, End = End of last modulation. Src = Source of Transfer
All times are in carrier periods (1/13.56Mhz)        

     Start |       End | Src | Data
-----------|-----------|-----|--------
         0 |  11386732 | Rdr | 00
proxmark3>

pwpiwi, 2017-10-09:
Proxdroid 2.0.0 is based on a pretty old source code. You may try the slightly newer one on https://github.com/FonkyCorp/proxdroid. Otherwise you may try to make the current source code compile on android. You are more than welcome to make Pull Requests to add your changes here.

maxben14, 2017-10-09:
@pwpiwi , i tried compile current source code with make tools and eclipse but i have problem with error compile. Can you give me useful link about how compile c++ in native library *.so or short manual about.

pwpiwi, 2017-10-10:
It is hard to help if all you tell us is "error compile".

micolous, 2017-10-10:
That project won't open in Eclipse -- you need to download the NDK (through Android Studio), put the build tools in your PATH, and build it with `ndk-build`.  Also, using Eclipse for Android development [was deprecated in 2015](https://android-developers.googleblog.com/2015/06/an-update-on-eclipse-android-developer.html).  You [should probably start here, it will tell you how to install and setup the NDK](https://developer.android.com/ndk/guides/index.html).

That `proxdroid` source builds fine on my (fairly recent) NDK with `~/Android/Sdk/ndk-bundle/ndk-build`, but you'll need to replace [this line](https://github.com/FonkyCorp/readline/blob/43afdda0196e27f53e8ad59822ff898b5049f60c/config.h#L40) with `#undef HAVE_GETPWENT`.  However, I don't have a rooted device which would verify that binary actually works.

The other thing is that different versions of the firmware running on the PM3 itself have protocol differences, so you'll need to build a firmware that matches the version of the PM3 client, ideally down to the same commit.  Whatever version you've either flashed on there or was flashed at the factory is probably something completely different.  You'll want to do this flashing from your PC, and build [that particular commit of PM3](https://github.com/FonkyCorp/proxmark3/tree/b324b02f8845417fc5c5ae7199146d7c998d1e16).

maxben14, 2017-10-13:
@pwpiwi , can tell me please what difference in https://github.com/angelsl/proxmark3-android/ and https://github.com/FonkyCorp/proxdroid ? I want try build ndk libs from fork @angelsl in eclipse, but i don't know how makelist convert in android.mk file which understand eclipse. Can me help with convert make file please ?

pwpiwi, 2017-10-13:
Sorry, I don't know any of these and I don't know how to build on Android. And didn't you read @micolous  comment on Eclipse?

maxben14, 2017-10-14:
@micolous , i try ndk-build by your instruction.
[x86] Compile        : proxmark3 <= crapto1.c
In file included from jni/../proxmark3/client/nonce2key/crapto1.c:20:
jni/../proxmark3/client/nonce2key/crapto1.h:73:9: warning: implicit declaration of function 'asm' is invalid in C99
      [-Wimplicit-function-declaration]
        asm(    "movl %1, %%eax\n"
        ^
jni/../proxmark3/client/nonce2key/crapto1.h:79:36: error: expected ')'
                "movzx %%al, %0\n" : "=r"(x) : "r"(x): "eax","ecx");
                                   ^
jni/../proxmark3/client/nonce2key/crapto1.h:73:12: note: to match this '('
        asm(    "movl %1, %%eax\n"
           ^
1 warning and 1 error generated.
make: *** [obj/local/x86/objs/proxmark3/client/nonce2key/crapto1.o] Error 1
Why show error in crapto1.c how fixed this ?
But in this moment i see ndk-build make in /obj files: proxmark3 and libs.
I put proxmark3 in /system/bin and libreadline.so and libtermcap.so and libusb.so in /system.lib put.
I try run client on android but command not answer.
u0_a203@android:/ $ su
root@android:/ # cd system
root@android:/system # cd bin
root@android:/system/bin # proxmark3 /dev/ttyACM0

Connected units:
        1. SN: ? [001/002]
proxmark3> hf mf rdbl 0 a a0a1a2a3a4a5
Can't open logfile, logging disabled!
--block no:00 key type:00 key:a0 a1 a2 a3 a4 a5      
Read only 63 instead of requested 64 bytes!
#db#
Read only 63 instead of requested 64 bytes!
Read only 63 instead of requested 64 bytes!

Can help me with this error ?

pwpiwi, 2017-10-17:
Obviously the compiler doesn't know about the `asm` function. I am wondering how the linker could create a working executable without crapto1.o ?

The "can't open logfile" error is probably a missing write permission.

The "Read only 63 instead of requested 64 bytes" originates from the USB read function.

iceman1001, 2017-10-17:
funny, I observed that 63 bytes read yesterday.  There is a (size-1) i usb_write..  which be protecting a potential overflow (zero-index remember), but it just becomes 63 bytes in the usb-transfer instead.

micolous, 2017-10-17:
> What is the difference between angelsl and fonkycorp's builds?

Fonkycorp is using an older version of Proxmark3 code, and is trying to build a Android Linux console binary.

By comparison, angelsl is trying to build a different version of proxmark3 (which seems to have the code missing / private -- something PM3 team may want to follow up as a GPL issue) and wrap it into an Android app, but still using Linux APIs rather than Android ones (so you need root too).

Angelsl has an APK up in the "releases" section you can try, but I have no idea which version of PM3 it is built against.

> ...build errors...

To explain what you're seeing, the `ndk-build` command will build for many architectures, and the specific one your build seems to be having trouble on is for the x86 target -- which despite being the same platform as most desktop PCs, is fairly uncommon in Android devices.

I'm also not sure why you got that warning, because I was able to build Fonkycorp's code without any errors.  I'm guessing that you still have an Android SDK folder hanging around from Eclipse on your system, and something is referencing that (possibly some environment variable like `ANDROID_HOME`).

As you've noticed, you still have other (probably ARM) binaries there, which seem to be good enough.  The specific code is skipped when running on non-x86 platforms, so I wouldn't worry too much about it.

The other step that you've missed though is you need to [build Fonkycorp's old version of Proxmark3 for your PC](https://github.com/FonkyCorp/proxmark3/tree/b324b02f8845417fc5c5ae7199146d7c998d1e16), including the firmware, and [flash that onto your PM3](https://github.com/Proxmark/proxmark3/wiki/flashing).  There are protocol changes between releases, which will cause you trouble.  This also means that some functionality won't be available anymore.

I'd also suggest trying `hw version` and `hw status` as these will give some extra debugging information and verify that you're actually talking with the hardware properly, and have the right firmware.

You may also find you're lacking the `usb-cdc` module (to test, try: `ls -l /dev/ttyACM0`).  This is specific to the kernel running on your Android device (and thus, the "build" or "ROM" you're using).  Most regular Android kernels do not ship with it, and you'll probably want to hit up your favourite Android development site for help with that one, or look at third-party Android distributions.

_Side note:_ I'm working on building a "proper" version of PM3 for Android, which would work without root and uses Android's APIs.  But this is also hard (and not ready).

iceman1001, 2017-11-30:
close?  Since its a question for the specific fork mentioned, not pm3 official.

closed 2017-11-30 by pwpiwi