Proxmark/proxmark3 #415: OSX 10.12.6 (16G29) - Impossible to compile proxmark3 client

masterix21, 2017-10-11:
I'm trying to compile an updated version of proxmark3 client but unsuccessfully, because the "make" command remains hung to:
`gcc -MT obj/hardnested/hardnested_bf_core_NOSIMD.o -MMD -MP -MF obj/hardnested/hardnested_bf_core.Td  -std=c99 -D_ISOC99_SOURCE -I. -I../include -I../common -I../zlib -I../uart -I/opt/local/include -I../liblua -Wall -g -O3 -DHAVE_GUI -mno-mmx -mno-sse2 -mno-avx -mno-avx2 -mno-avx512f -c -o obj/hardnested/hardnested_bf_core_NOSIMD.o hardnested/hardnested_bf_core.c`

After about 3 hours, i killed the process.

`luca$ gcc --version
gcc (Homebrew GCC 4.9.4) 4.9.4`

Same problem with "marshmellow42" fork.

Any idea?

merlokk, 2017-10-11:
iceman fork?

iceman1001, 2017-10-11:
... I doubt a mingw env is useful on OSX env.

pwpiwi, 2017-10-12:
@masterix21: Is there a reason why you use an old gcc instead of the standard clang compiler?

masterix21, 2017-10-12:
you're right :)

removed gcc@4.9 installed by brew and now compile with gcc OSX version.

Thanks and sorry.

closed 2017-10-12 by pwpiwi