Proxmark/proxmark3 #418: Where buy emulator 1k flat how card ?

maxben14, 2017-10-15:
I found this https://ikey.ru/sniffer-key/ emulator 1k flat how card. This emulator easy to put in a ticket purchase terminal but this device can only save the numbers Nr, Ar but I want a similar device that can be programmed as a proxmark. Can you tell where to buy this and whether there are such?

pwpiwi, 2017-10-15:
Please ask questions in the forum.

closed 2017-10-15 by pwpiwi

maxben14, 2017-10-15:
@pwpiwi ,i try ask on forum http://www.proxmark.org/forum/viewtopic.php?pid=29869#p29869 but there are rarely answered.

pwpiwi, 2017-10-15:
On github you are reaching the active Proxmark3 developers (I would say less than 10 people). On the forum there are hundreds of users (including the active developers). In general, the chances to get answers are better on the forum.

There are several reasons for not getting an answerr:
* The answer is obvious. Read the Wiki and the forum. Nobody wants to answer the same questions over and over again
* There is no answer or nobody knows it.
* The question is off topic.
In your case there may be no answer. And As @iceman1001 mentioned in  http://www.proxmark.org/forum/viewtopic.php?pid=29869#p29869 your questions is off topic.

maxben14, 2017-10-15:
Why my question is off topic ? The main problem of the proxmark that it is very large and does not fit into the payment terminal slot. Why not make device (proxmark or same)  in the form of a flat card as the Chinese do magic cards.

merlokk, 2017-10-15:
It can be done. but no one wants to to that)

maxben14, 2017-10-15:
@merlokk , why no one wants do this device ? Transport systems try to protect themselves from proxmark and similar devices. Therefore, I think the problem is urgent, we need a device that can enter into any payment terminal without consequences.

merlokk, 2017-10-15:
because in proxmark repository you see correctly working example. Tons of working code.
There is no problem to implement copy in another hardware. But
1. probably it's illegal. no one wants to make stealth emulator and put in to public access)
2. its costs... ones who can make it probably will not use it at all) because they dont use public transport (not eu and so on. its for 3rd world countries) Ones who will use it - cant buy it for 300-400$ and there is no money benefit
3. i now hundreds of engineers who can make it, but no one wants to pay like 20000-50000$ for development a device)

maxben14, 2017-10-15:
@merlokk , in my transport system there are JCOP,PLUS and this cards work in emulation mifare classic and very good crack tools hardnested)

merlokk, 2017-10-15:
I can guess a city in Russia))))
The protection will cost a money, The emulator costs a money also.
It just a race) You invest a money to emulator... They switch plus cards from sl1 to sl3.... you will lost)
there is just a science. there is no sense to fight with companies who writes a software for payment systems (i know what i say...). And also its illegal)))