Proxmark/proxmark3 #423: hf mf mifare reboot proxmark3

edwinhaq, 2017-10-16:
Hi,
when i run "hf mf mifare" the proxmark3 reboot after any seconds.
Sometimes shows lines like "#db# Mifare: Can't select card" and some others show dots.

Any ideas?

This is my "hw version"
Prox/RFID mark3 RFID instrument
bootrom: master/v3.0.1-98-g9d0a333-suspect 2017-10-16 23:00:53
os: master/v3.0.1-98-g9d0a333-suspect 2017-10-16 23:00:56
LF FPGA image built for 2s30vq100 on 2015/03/06 at 07:38:04
HF FPGA image built for 2s30vq100 on 2017/07/13 at 08:44:13

uC: AT91SAM7S512 Rev B
Embedded Processor: ARM7TDMI
Nonvolatile Program Memory Size: 512K bytes. Used: 200318 bytes (38%). Free: 323970 bytes (62%).
Second Nonvolatile Program Memory Size: None
Internal SRAM Size: 64K bytes
Architecture Identifier: AT91SAM7Sxx Series
Nonvolatile Program Memory Type: Embedded Flash Memory

pwpiwi, 2017-10-17:
I cannot confirm this issue. Any more hints? Which LEDs are on or blinking?

edwinhaq, 2017-10-17:
here's a video 
https://youtu.be/oUnyLzzN9B0

happens with any card, only once I have the key I expected,

there is a report a few years ago related http://www.proxmark.org/forum/viewtopic.php?id=2172

iceman1001, 2017-10-17:
the WDT timeout... 
it doesn't happen very often anymore.  On v3.0.1 you are the first to report it.

edwinhaq, 2017-10-17:
Could it be a hardware problem? 🤔

iceman1001, 2017-10-17:
nop.  not a hardware problem.  its a long execution loop which triggers the WDT inside **hf mf mifare**

pwpiwi, 2017-10-17:
Your video shows two issues: chkkey fails (probably again due to reduced timeout? @merlokk ?). And after some retries the watchdog kicks in. Probably fixing the first issue will eliminate the second.

iceman1001, 2017-10-17:
The "can't select tag" messages, 
... I would say you have the tag too close to the antenna.   1-2cm is usally a better distance.

merlokk, 2017-10-17:
I have not changed default timeout in iso14443a_setup()

	iso14a_set_timeout(1050); // 10ms default

pwpiwi, 2017-10-17:
@merlokk: you are correct. hf mf mifare calls mfCheckKeys here https://github.com/Proxmark/proxmark3/blob/9d0a333449d52dc0d9cfca217cf9f0cc0f792744/client/mifarehost.c#L207 without changing the default timeout. Sorry for blaming you.

@iceman1001: This seems to be a Fudan clone. For those tags I am getting the "can't select tag messages" as well when the green LED starts blinking.

@edwinhaq: even with a Fudan clone I cannot confirm the issue. Are you using matching client and firmware?

edwinhaq, 2017-11-18:
I apply fix #484 and it works very well, thanks!!!

closed 2017-11-18 by edwinhaq